Koenkk/zigbee2mqtt #19735: npm start error

MicMac11, 2023-11-16:
### What happened?

Hello everyone, 

$ npm --version
8.19.4
$ node --version
v20.9.0

I have an error during installation see log

### What did you expect to happen?

npm start

### How to reproduce it (minimal and precise)

reboot

### Zigbee2MQTT version

1.33.2

### Adapter firmware version

dont have

### Adapter

SonOFF Zigbee 3.0

### Debug log

info  2023-11-16 11:25:57: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2023-11-16.11-25-57' filename: log.txt
info  2023-11-16 11:25:57: Starting Zigbee2MQTT version 1.33.2 (commit #311ea07)
info  2023-11-16 11:25:57: Starting zigbee-herdsman (0.21.0)
error 2023-11-16 11:26:19: Error while starting zigbee-herdsman
error 2023-11-16 11:26:19: Failed to start zigbee
error 2023-11-16 11:26:19: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
error 2023-11-16 11:26:19: Exiting...
error 2023-11-16 11:26:19: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.start (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:103:27)
    at Controller.start (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/controller.ts:132:29)
    at Zigbee.start (/opt/zigbee2mqtt/lib/zigbee.ts:60:27)
    at Controller.start (/opt/zigbee2mqtt/lib/controller.ts:101:27)
    at start (/opt/zigbee2mqtt/index.js:107:5)

MicMac11, 2023-11-16:
Config : 
homeassistant: false
frontend:
  port: 8089
mqtt:
  base_topic: zigbee2mqtt
  server: mqtt://localhost:1883
serial:
  port: /dev/ttyUSB0

MicMac11, 2023-11-16:
I also tried this configuration for the :
/dev/serial/by-id/usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_464026bf9d9ded11ae75cfa5a7669f5d-if00-port0

francisp2, 2023-11-17:
Don't use a USB3 port, and use a USB extension cable of at least 1 meter.

MicMac11, 2023-11-19:
I don't have a usb 3.0 port. 3.0 is the dongle version.

Uhmoja, 2023-11-22:
Same problem here since I wanted to update Z2M.
I'm completely stuck.
Adding "adapter: ezsp" as suggested in other posts with the same problem has not changed anything for me.

closed 2024-06-20 by github-actions[bot]