Koenkk/zigbee2mqtt #19742: OTA update fails with error "Attempt to access memory outside buffer bounds"

jeremysherriff, 2023-11-17:
### What happened?

Am trying to push a firmware update for an experimental device, using an OTA index override. I believe that should not matter.
"Check for new updates" has been successful with new firmware detected.
Clicking the OTA button "Update device firmware" results in an error "Attempt to access memory outside buffer bounds" which I think is generated by node.js

### How to reproduce it (minimal and precise)

I expected the OTA firmware update process to begin.

### Zigbee2MQTT version

1.33.2-dev commit: ad4bed8

### Adapter firmware version

6.10.3.0 build 297

### Adapter

EZSP v8 (Sonoff ZB Dongle-E)

### Debug log

```
debug 2023-11-17 20:28:48: Received MQTT message on 'zigbee2mqtt/bridge/request/device/ota_update/update' with data '{"id":"0xa4c138aac2809fe3","transaction":"prdao-2"}'
info  2023-11-17 20:28:48: Updating '0xa4c138aac2809fe3' to latest firmware
debug 2023-11-17 20:28:54: Received Zigbee message from '0xa4c138aac2809fe3', type 'readResponse', cluster 'genBasic', data '{"dateCode":"20231116","swBuildId":"3001-0105"}' from endpoint 1 with groupID 0
info  2023-11-17 20:28:54: MQTT publish: topic 'zigbee2mqtt/0xa4c138aac2809fe3', payload '{"battery":69,"humidity":70,"last_seen":"2023-11-17T07:28:54.453Z","linkquality":208,"temperature":22,"temperature_display_mode":null,"update":{"installed_version":17117185,"latest_version":17117187,"state":"available"},"update_available":null,"voltage":2700}'
debug 2023-11-17 20:28:54: Updating to latest '0xa4c138aac2809fe3' (LYWSD03MMC)
debug 2023-11-17 20:28:54: Using endpoint '1'
debug 2023-11-17 20:28:54: Received Zigbee message from '0xa4c138aac2809fe3', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":17117185,"imageType":522,"manufacturerCode":4417}' from endpoint 1 with groupID 0
info  2023-11-17 20:28:54: MQTT publish: topic 'zigbee2mqtt/0xa4c138aac2809fe3', payload '{"battery":69,"humidity":70,"last_seen":"2023-11-17T07:28:54.979Z","linkquality":212,"temperature":22,"temperature_display_mode":null,"update":{"installed_version":17117185,"latest_version":17117187,"state":"available"},"update_available":null,"voltage":2700}'
debug 2023-11-17 20:28:54: Got OTA request '{"fieldControl":0,"manufacturerCode":4417,"imageType":522,"fileVersion":17117185}'
debug 2023-11-17 20:28:55: ZigbeeOTA: downloaded main index
debug 2023-11-17 20:28:55: ZigbeeOTA: Loading override index /app/data/index.json
debug 2023-11-17 20:28:55: getNewImage for '0xa4c138aac2809fe3', meta {"fileVersion":17117187,"url":"/app/data/1141-020a-99993001-ATC_v46.zigbee","sha512":"12619430b9504fb30f6485a44e456ce7eb22c0bf75bf6b234f8d06e6a859c69c25fd99678cef4997df06a6966e9b57978fe2ed34e0b4302bb416bb10565f3adb"}
debug 2023-11-17 20:28:55: ZigbeeOTA: getting local firmware file /app/data/1141-020a-99993001-ATC_v46.zigbee
debug 2023-11-17 20:28:55: OTA update checksum validation succeeded for '0xa4c138aac2809fe3'
debug 2023-11-17 20:28:55: Update of '0xa4c138aac2809fe3' failed (RangeError [ERR_BUFFER_OUT_OF_BOUNDS]: Attempt to access memory outside buffer bounds)
info  2023-11-17 20:28:55: MQTT publish: topic 'zigbee2mqtt/0xa4c138aac2809fe3', payload '{"battery":69,"humidity":70,"last_seen":"2023-11-17T07:28:54.979Z","linkquality":212,"temperature":22,"temperature_display_mode":null,"update":{"installed_version":17117185,"latest_version":17117187,"state":"available"},"update_available":null,"voltage":2700}'
info  2023-11-17 20:28:55: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/update', payload '{"data":{"id":"0xa4c138aac2809fe3"},"error":"Update of '0xa4c138aac2809fe3' failed (Attempt to access memory outside buffer bounds)","status":"error","transaction":"prdao-2"}'
error 2023-11-17 20:28:55: Update of '0xa4c138aac2809fe3' failed (Attempt to access memory outside buffer bounds)
debug 2023-11-17 20:28:55: RangeError: Attempt to access memory outside buffer bounds
    at new NodeError (node:internal/errors:405:5)
    at boundsError (node:internal/buffer:86:11)
    at Buffer.readUInt16LE (node:internal/buffer:245:5)
    at parseImage (/app/node_modules/zigbee-herdsman-converters/src/lib/ota/common.ts:55:34)
    at getNewImage (/app/node_modules/zigbee-herdsman-converters/src/lib/ota/common.ts:473:19)
    at processTicksAndRejections (node:internal/process/task_queues:95:5)
    at Object.updateToLatest (/app/node_modules/zigbee-herdsman-converters/src/lib/ota/common.ts:317:19)
    at OTAUpdate.onMQTTMessage (/app/lib/extension/otaUpdate.ts:268:41)
```

jeremysherriff, 2023-11-18:
I suspect that this "OTA" isn't formatted internally in the way an OTA file needs to be, as this specific file is the BLE-oriented firmware intended to return the device to its original state (see https://github.com/pvvx/ZigbeeTLc for project info if interested).

I believe the only way "back" for me is to use a USB UART to flash the device.
Closing.

closed 2023-11-18 by jeremysherriff

mr-aleks, 2024-02-28:
It is still impossible to return the device to its original state through an update?? I have the same error...

mcerveny, 2024-05-24:
@mr-aleks Yes possible. Use other non-corrupted ATC firmware - https://github.com/devbis/z03mmc/blob/master/assets/db15-0203-99993001-ATC_v46.zigbee and update inside identification manually (hexedit).

```
$ hexedit ota/modified.zigbee
$ xxd ota/modified.zigbee > /tmp/n
$ xxd Downloads/db15-0203-99993001-ATC_v46.zigbee > /tmp/o
$ diff /tmp/o /tmp/n
1c1
< 00000000: 1ef1 ee0b 0001 3800 0000 15db 0302 0130  ......8........0
---
> 00000000: 1ef1 ee0b 0001 3800 0000 4111 0a02 0130  ......8...A....0
$ tail -2 configuration.yaml
ota:
  zigbee_ota_override_index_location: my_index.json
$ cat my_index.json
[
    {
        "url": "ota/modified.zigbee",
        "force": true
    }
]
```

michi2005, 2024-09-27:
> @mr-aleks Yes possible. Use other non-corrupted ATC firmware - https://github.com/devbis/z03mmc/blob/master/assets/db15-0203-99993001-ATC_v46.zigbee and update inside identification manually [...]

Can you please explain more what to do. I opened it in a hex editor but I found not the mentioned string.

mcerveny, 2024-09-27:
copy "Downloads/db15-0203-99993001-ATC_v46.zigbee" to  "ota/modified.zigbee" and then change "15 db 03 02" to "41 11 0a 02" replacement starts at offset 0xA in "ota/modified.zigbee" (output of diff should be as displayed).

stedju499, 2024-10-04:
@michi2005 Using the latest version at the moment [db15-0203-99993001-ATC_v48.zigbee](https://github.com/pvvx/ATC_MiThermometer/blob/master/zigbee_ota/db15-0203-99993001-ATC_v48.zigbee) worked for me without editing the file.
More info at https://github.com/pvvx/ATC_MiThermometer/tree/master/zigbee_ota