Koenkk/zigbee2mqtt #19745: AQSZB-110

josujs, 2023-11-17:
### Link

https://www.zigbee2mqtt.io/devices/AQSZB-110.html

### Database entry

Failed to check if update available for '0x0015bc001e011c82' (Device didn't respond to OTA request)

### Comments

Good morning, everyone. I have a problem with 3 Zigbee devices that I've purchased. On the Zigbee2MQTT website, it shows that they are supported, but I'm unable to include them. The model numbers of the devices are AQSZB-110. I would appreciate any help on how to include them or to request that they be included, please. I'm not sure if this is the right channel to make such requests? Thank you very much for your time.

Best regards

Krzysztonek, 2023-11-18:
I have the same device successfully discovered and installed. OTA works fine. However, the `battery_level` entity always reports `0%`, even with brand new set of batteries just applied.

josujs, 2023-11-19:
> He descubierto e instalado con éxito el mismo dispositivo. La OTA funciona bien. sin embargo, el `battery_level`la entidad siempre informa `0%`, incluso con un juego de baterías nuevo recién aplicado.

Thanks for commenting, could you please tell me how you were able to install it?
Thank you so much

Krzysztonek, 2023-11-19:
> could you please tell me how you were able to install it?

I just switched the device to pairing mode by pressing for over 20s a black button next to the batteries. (A red LED blinks quickly)
After second or third attempt the device was discovered and then it joined the network gladly.

closed 2024-06-17 by github-actions[bot]