Koenkk/zigbee2mqtt #19747: Devices stop communicating randomly

Alfy1080, 2023-11-17:
### What happened?

Hello. I am not a zigbee expert so apologies if i provide incomplete information. I will try my best to mention everything.

Since 2 days ago my z2m instance in home assistant started acting up randomly.

At random times, one or multiple zigbee devices stop executing commands. 
ex1: Aqara TRV does not change the target temperature when i try to change it from either home assistant or z2m interface, the last seen status keeps increasing as if there is no communication between the TRV and the coordinator.
ex2: Philips Hue Lightstrip does not switch on or off whenever i attempt to toggle it either from home assistant or z2m. 

I have set the logging level to debug and when i try to send a command to the stuck device, the command shows up in the logs without any error whatsoever. 

Power cycling the stuck device or pressing the pairing button (where applicable) does nothing.
The only thing that seems to get my zigbee network back up and running temporarily is restarting z2m which makes me think that this is caused by something in z2m and not the devices that fail. Right after the z2m restart, all stuck devices start communicating again for a while until at some point either the same devices or others present the same behaviour as before.

My setup: 
Zigbee dongle: Home Assistant SkyConnect flashed with the latest firmware available through the web flasher here https://skyconnect.home-assistant.io/firmware-update/
Zigbee2Mqtt: Latest addon version available for Home Assistant (1.33.2-1)
Home Assistant Core version: 2023.11.2 
Home Assistant Supervisor version: 2023.11.3
OS: Debian 12
Server: Dell OptiPlex 9020 Micro, Core i7-4790t 3.90GHz, 16GB DDR3, SSD

### What did you expect to happen?

I expected that no device will get in a frozen state where i can not issue commands to it or receive state changes from it. At least not as often as once every few minutes/hours

### How to reproduce it (minimal and precise)

There is no replication steps that i can imagine. This issue can happen even when no zigbee device receives any command at all. I even left home for a few hours and after i left i restarted zigbee2mqtt to make sure it's all in working order and does not get any commands from anyone since nobody was home. When i returned home one of the Aqara TRVs was stuck.

### Zigbee2MQTT version

1.33.2

### Adapter firmware version

7.2.2.0 build 190

### Adapter

Home Assistant SkyConnect

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/13394273/log.txt)

Alfy1080, 2023-11-17:
Here you can see i have updated the temperature on 3 Aqara TRVs:

Living_Room_TRV: 21.5
Kids_Room_TRV: 17.5
Bedroom_TRV: 22.5

`debug 17-11-2023 18:24:29: Received MQTT message on 'zigbee2mqtt/Living_Room_TRV/set/occupied_heating_setpoint' with data '21.5'
debug 17-11-2023 18:24:29: Publishing 'set' 'occupied_heating_setpoint' to 'Living_Room_TRV'
debug 17-11-2023 18:24:29: Received MQTT message on 'zigbee2mqtt/Kids_Room_TRV/set/occupied_heating_setpoint' with data '17.5'
debug 17-11-2023 18:24:29: Publishing 'set' 'occupied_heating_setpoint' to 'Kids_Room_TRV'
debug 17-11-2023 18:24:29: Received MQTT message on 'zigbee2mqtt/Bedroom_TRV/set/occupied_heating_setpoint' with data '22.5'
debug 17-11-2023 18:24:29: Publishing 'set' 'occupied_heating_setpoint' to 'Bedroom_TRV'
info  17-11-2023 18:24:29: MQTT publish: topic 'zigbee2mqtt/Bedroom_TRV', payload '{"away_preset_temperature":null,"battery":100,"calibrate":null,"calibrated":null,"child_lock":"UNLOCK","device_temperature":27,"internal_heating_setpoint":30,"last_seen":"2023-11-17T18:24:29+02:00","linkquality":164,"local_temperature":23.5,"occupied_heating_setpoint":23,"power_outage_count":0,"preset":"manual","schedule":null,"schedule_settings":null,"sensor":"external","setup":false,"system_mode":"heat","update":{"installed_version":2590,"latest_version":2590,"state":"idle"},"update_available":null,"valve_alarm":false,"valve_detection":"ON","voltage":3300,"window_detection":"OFF","window_open":null}'
info  17-11-2023 18:24:29: MQTT publish: topic 'zigbee2mqtt/Bedroom_TRV', payload '{"away_preset_temperature":null,"battery":100,"calibrate":null,"calibrated":null,"child_lock":"UNLOCK","device_temperature":27,"internal_heating_setpoint":30,"last_seen":"2023-11-17T18:24:29+02:00","linkquality":164,"local_temperature":23.5,"occupied_heating_setpoint":22.5,"power_outage_count":0,"preset":"manual","schedule":null,"schedule_settings":null,"sensor":"external","setup":false,"system_mode":"heat","update":{"installed_version":2590,"latest_version":2590,"state":"idle"},"update_available":null,"valve_alarm":false,"valve_detection":"ON","voltage":3300,"window_detection":"OFF","window_open":null}'
debug 17-11-2023 18:24:30: Received Zigbee message from 'Bedroom_TRV', type 'attributeReport', cluster 'hvacThermostat', data '{"occupiedHeatingSetpoint":2250}' from endpoint 1 with groupID 0
info  17-11-2023 18:24:30: MQTT publish: topic 'zigbee2mqtt/Bedroom_TRV', payload '{"away_preset_temperature":null,"battery":100,"calibrate":null,"calibrated":null,"child_lock":"UNLOCK","device_temperature":27,"internal_heating_setpoint":30,"last_seen":"2023-11-17T18:24:30+02:00","linkquality":168,"local_temperature":23.5,"occupied_heating_setpoint":22.5,"power_outage_count":0,"preset":"manual","schedule":null,"schedule_settings":null,"sensor":"external","setup":false,"system_mode":"heat","update":{"installed_version":2590,"latest_version":2590,"state":"idle"},"update_available":null,"valve_alarm":false,"valve_detection":"ON","voltage":3300,"window_detection":"OFF","window_open":null}'`

Out of these 3 TRVs only the Bedroom_TRV actually executed the command. The other two completely ignored it and their last seen status did not update when i changed the target temperature:

[image]
Attached another full log after changing the target temperatures:
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/13394370/log.txt)

Ra72xx, 2023-11-18:
I have a similar problem with the Aqara thermostats (#19342). Sometimes the communication seems to fail, however Z2M does not notice that and pretends everything is right.

Alfy1080, 2023-11-19:
Forgot to mention that I have fully erased zigbee2mqtt from my system, reinstalled and reconfigured z2m from scratch, re-paired all my devices again to my coordinator and the exact same issue is still happening randomly. As a workaround I have set up an automation in home assistant to restart the z2m addon every 30 minutes just to make sure my TRVs don't stay stuck for too long and causes my heating to run forever. I have also ordered a Sonoff Dongle-P to rule out the possibility that the SkyConnect is broken, but that will arrive in one or two days so I'm still waiting to test that. 
Also replaced my USB extension cable to make sure i'm not using a faulty cable but that didn't improve the situation either.

Alfy1080, 2023-11-20:
I have reflashed my skyconnect dongle and re-paired everything from scratch, just to rule out the possibility of the last firmware update flashing improperly and corrupting my dongle, but the issue is still there. I have attached the latest debug log here.
[z2mlog.txt](https://github.com/Koenkk/zigbee2mqtt/files/13412943/z2mlog.txt)

PeterKawa, 2023-11-20:
**What happened?**
Similar issue here, but with "Sonoff Zigbee 3.0 USB Dongle Plus ZBDongle-E" 
It has run just fine without issue, for about six months.

It started when the add-on was updated to v.1.33.2, but it was not right after.;I took a while before I realised the update can have bugs for my setup.
I checked my wifi & zigbee channels, wifi 1 and zigbee 20 should be fine.
No warnings or errors in the logs!
I updated to Edge, but it results in exactly the same behaviour.

Sometimes restarting z2m resolves the unresponsiveness, sometimes one device keeps unresponsive.
Often 5 to 8 devices suddenly have a Offline status, but when I turn each of them on/off in the frontend, the stautus is online suddenly. I can't operate them as entity in the normal HA interface, only using the switches in z2m frontend until the status is online again.

I hope my zigbee-herdsman log will reveal anything.

**What did you expect to happen?**
I expected my zigbee devices to respond as they should

**How to reproduce it (minimal and precise)**
Install update v1.33.2 should do the trick

**Zigbee2MQTT version**
v.1.33.2 
and since nov. 18th: 1.33.2-dev commit: [ad4bed8](https://github.com/Koenkk/zigbee2mqtt/commit/ad4bed8)

**Adapter firmware version**
6.10.3.0 build 297

**Adapter**
Sonoff Zigbee 3.0 USB Dongle Plus ZBDongle-E (with 50cm extension cable, USB2)

**Debug log**
```
2023-11-20T19:20:50.432Z zigbee-herdsman:adapter:ezsp:ezsp <== 0x59: {"_cls_":"incomingRouteRecordHandler","_id_":89,"_isRequest_":false,"source":1027,"longId":{"type":"Buffer","data":[180,227,249,255,254,7,178,55]},"lastHopLqi":244,"lastHopRssi":-39,"relay":{"type":"Buffer","data":[35]}}
2023-11-20T19:20:50.433Z zigbee-herdsman:adapter:ezsp:driv handleRouteRecord: nwk=1027, ieee=������7, lqi=244, rssi=-39, relays=#
2023-11-20T19:20:50.481Z zigbee-herdsman:adapter:ezsp:uart <-- [01eab1a97d312a15b658944e24ab5593499c6fd372eeca9874f5ce3b83fc7d5e1e64e0c97d33087e]
2023-11-20T19:20:50.481Z zigbee-herdsman:adapter:ezsp:uart <-- DATA (0,1,0): 01eab1a9112a15b658944e24ab5593499c6fd372eeca9874f5ce3b83fc7e1e64e0c913087e
2023-11-20T19:20:50.481Z zigbee-herdsman:adapter:ezsp:uart --> ACK  (1)
2023-11-20T19:20:50.481Z zigbee-herdsman:adapter:ezsp:uart --> [8160597e]
2023-11-20T19:20:50.482Z zigbee-herdsman:adapter:ezsp:ezsp <== Frame: a890014500000401000401010001000021f4d90304ffff0808580a000021c30b04
2023-11-20T19:20:50.486Z zigbee-herdsman:adapter:ezsp:ezsp <== 0x45: {"_cls_":"incomingMessageHandler","_id_":69,"_isRequest_":false,"type":0,"apsFrame":{"profileId":260,"clusterId":1024,"sourceEndpoint":1,"destinationEndpoint":1,"options":256,"groupId":0,"sequence":33},"lastHopLqi":244,"lastHopRssi":-39,"sender":1027,"bindingIndex":255,"addressIndex":255,"message":{"type":"Buffer","data":[8,88,10,0,0,33,195,11]}}
2023-11-20T19:20:50.487Z zigbee-herdsman:adapter:ezsp:debg processMessage: {"messageType":0,"apsFrame":{"profileId":260,"clusterId":1024,"sourceEndpoint":1,"destinationEndpoint":1,"options":256,"groupId":0,"sequence":33},"lqi":244,"rssi":-39,"sender":1027,"bindingIndex":255,"addressIndex":255,"message":{"type":"Buffer","data":[8,88,10,0,0,33,195,11]}}
2023-11-20T19:20:50.515Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":88,"manufacturerCode":null,"commandIdentifier":10},"Payload":[{"attrId":0,"dataType":33,"attrData":3011}],"Command":{"ID":10,"name":"report","parameters":[{"name":"attrId","type":33},{"name":"dataType","type":32},{"name":"attrData","type":1000}]}},"address":1027,"endpoint":1,"linkquality":244,"groupID":0,"wasBroadcast":false,"destinationEndpoint":1}'
2023-11-20T19:20:50.558Z zigbee-herdsman:controller:endpoint DefaultResponse 0xb4e3f9fffe07b237/1 1024(10, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false})
2023-11-20T19:20:51.440Z zigbee-herdsman:adapter:ezsp:uart <-- [7d31eab1a90d2a16b66e264ddb55ac71fd68972688a708cd7e]
2023-11-20T19:20:51.443Z zigbee-herdsman:adapter:ezsp:uart <-- DATA (1,1,0): 11eab1a90d2a16b66e264ddb55ac71fd68972688a708cd7e
2023-11-20T19:20:51.446Z zigbee-herdsman:adapter:ezsp:uart --> ACK  (2)
2023-11-20T19:20:51.449Z zigbee-herdsman:adapter:ezsp:uart --> [82503a7e]
2023-11-20T19:20:51.452Z zigbee-herdsman:adapter:ezsp:ezsp <== Frame: a890015900030437b207fefff9e3b4f4d901234a
2023-11-20T19:20:51.458Z zigbee-herdsman:adapter:ezsp:ezsp <== 0x59: {"_cls_":"incomingRouteRecordHandler","_id_":89,"_isRequest_":false,"source":1027,"longId":{"type":"Buffer","data":[180,227,249,255,254,7,178,55]},"lastHopLqi":244,"lastHopRssi":-39,"relay":{"type":"Buffer","data":[35]}}
2023-11-20T19:20:51.459Z zigbee-herdsman:adapter:ezsp:driv handleRouteRecord: nwk=1027, ieee=������7, lqi=244, rssi=-39, relays=#
2023-11-20T19:20:51.489Z zigbee-herdsman:adapter:ezsp:uart <-- [21eab1a97d312a15b658944e24ab5593499c6cdf71eeca9874f5ce3b83fc7d5e1e64e0c9f6ae7e]
2023-11-20T19:20:51.491Z zigbee-herdsman:adapter:ezsp:uart <-- DATA (2,1,0): 21eab1a9112a15b658944e24ab5593499c6cdf71eeca9874f5ce3b83fc7e1e64e0c9f6ae7e
2023-11-20T19:20:51.492Z zigbee-herdsman:adapter:ezsp:uart --> ACK  (3)
2023-11-20T19:20:51.493Z zigbee-herdsman:adapter:ezsp:uart --> [83401b7e]
[... cut ...]
2023-11-20T19:21:00.592Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":89,"manufacturerCode":null,"commandIdentifier":10},"Payload":[{"attrId":0,"dataType":33,"attrData":3011}],"Command":{"ID":10,"name":"report","parameters":[{"name":"attrId","type":33},{"name":"dataType","type":32},{"name":"attrData","type":1000}]}},"address":1027,"endpoint":1,"linkquality":248,"groupID":0,"wasBroadcast":false,"destinationEndpoint":1}'
```

Ra72xx, 2023-11-21:
I only recently switched from Conbee II / Deconz to Skyconnect / Z2M and so I can't judge if the problem appeared only with recent versions of Z2M. The setup was more stable with Deconz (which I didn't expect)!
I also have improved my Zigbee situation by putting the stick on an 2.0 USB hub and a 2m cable away from the rest of the setup. Nevertheless, I have quite a lot of such occurrences (devices getting "out of sync" with what state Z2M thinks they are, devices getting offline or dropping silently out of the network)..
Some types of devices seem to be prone to this problem, e.g. the Aqara thermostats (or the problem is more obvious with a heating schedule in autumn than with random lights or sensors).

Alfy1080, 2023-11-21:
UPDATE: My new Sonoff Zigbee Dongle-P arrived yesterday and I have replaced my Skyconnect with it and re-paired everything to it. So far since yesterday I had no devices that got stuck the same way as they did on the Skyconnect dongle. I only had an issue with some roller shade motors which was fixed by flashing the latest firmware on the sonoff dongle and re-pairing the motors in z2m. 

So it seems to me that at least in my case, the issue is caused by a combination of Skyconnect and zigbee2mqtt version since as i've said previously, this issue started happening recently even though i have been running the same setup in terms of dongle, routers and client devices for over a year now.

Ra72xx, 2023-11-21:
That would be extremely annoying as I switched 70 devices from Deconz to Z2M/Skyconnect only recently to get rid of intermittent instability and seemingly now I traded that in for persistent instability. (Yes, I know, Skyconnect is only not yet officially supported...)

Aleborg, 2023-11-21:
I have the same issue, using Sonoff Zigbee Dongle-P.

Devices affected:
TS110E_1gang_1
TS110E_2gang_1
TS0001_power
TS0001_switch_module

Randomly stops responding to commands but seems to be reporting state. 
A restart of zigbee2mqtt temporarily resolves it

Ra72xx, 2023-11-22:
I tried to change QoS, but this doesn't help (however, as QoS is only between MQTT and Z2M, I admit I did not really expect anything from this setting).

Ra72xx, 2023-11-22:
Maybe related discussion: https://github.com/Koenkk/zigbee2mqtt/discussions/19763

JonasSL, 2023-11-23:
I experience the exact same problem after updating to the same Z2M version. I also use a Skyconnect as coordinator, and I am also having troubles with my Aqara TRV. 

A restart of Z2M fixes the issue.

helgek, 2023-11-23:
Having a similar issue, randomly one of my devices would not respond to commands coming from Z2M, results in a timeout error:https://github.com/Koenkk/zigbee2mqtt/issues/13993#issuecomment-1824743866 - coordinator is the highly praised SLZB-0 (based on stable chipset CC2652P  recommended by Z2M, not the experimental EFR32MG21 chipset).

lennartgrunau, 2023-11-26:
Hey everyone! I have the same issue. The devices (in my case only IKEA LED2005R5 (3x)) randomly disconnect and show up as unavailable - and offline in z2m. 
**However** those devices still react to commands to a zigbee group that they are placed in. So, while I cannot control the devices directly, I can control them via a zigbee group, which is super weird to me. 

Can anyone recreate this? 

Thanks & Cheers

Zigbee2MQTT version
[1.33.2](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.33.2) commit: [unknown](https://github.com/Koenkk/zigbee2mqtt/commit/unknown)
Coordinator type
EZSP v9
Coordinator revision
7.1.1.0 build 273
Frontend version
0.6.142
Zigbee-herdsman-converters version
15.106.0
Zigbee-herdsman version
0.21.0

Home Assistant Sky Connect as a coordinator.

Ra72xx, 2023-11-27:
Is there any way to push issues like this? The random communication loss is something which is a complete deal breaker for me, as I can no longer rely on my Zigbee network for lights, radiators and switches of all kind. Seemingly every day I have to count my devices and check the thermostats if they are really operating etc.

However this and quite a few other bug reports concerning similar problems I've been monitoring on this list for the last few days don't seem to get any developer attention and are probably simply getting lost because of newer issues.

IMHO a network problem which results loosing communication to network members without even noticing the user but pretending everything is fine is one of the worst things which can happen.

digitalkaoz, 2023-11-27:
i experienced the same with my setup (Aquara Thermostats & Sonoff Dongle Plus) running on 1.33.2

So a downgrade to 1.33.0 fixes the random unresponsiveness to commands? im mostly a "fail forward person", so is there anything i can do except downgrading? looking into the logs, i see some warnings, i dont know if they are related:

```
warn  2023-11-27 08:56:15: zigbee-herdsman-converters:aqara_trv: Unknown key 641 =4��44
warn  2023-11-27 08:56:15: zigbee-herdsman-converters:aqara_trv: Unknown key 643 = 0
warn  2023-11-27 08:56:15: zigbee-herdsman-converters:aqara_trv: Unknown key 644 = 0
```

Ra72xx, 2023-11-27:
I also see those messages with the strange keys some times, but not regularly.

trackhacs, 2023-11-30:
i am in on this too. just moved from seperate docker HA / Z2M to HAOS with Z2M addon under proxmox. 

Also switched from ConbeeII to ZBDongle (EZSP v12)  with 7.3.1.0 build 176.

Ikea lamp which worked fine for at least one year now gets randomly unavailable. Toogling state in Z2M guide revokes it though. sometimes it also comes back alone.

Ra72xx, 2023-12-01:
Unfortunately, nobody's interested in the problem :-(

PeterKawa, 2023-12-01:
@digitalkaoz wrote:
> So a downgrade to 1.33.0 fixes the random unresponsiveness to commands? 
> ```

In my case it did, Robert, it works as before, no issues at all, running v1.33.0.

@Ra72xx Hmmm... The update's log v1.34.0-1 of today shows nothing about this issue or a fix for it (yet). I'm **not** updating for sure.

trackhacs, 2023-12-01:
i just updated - i give feedback if behavior improved.

Aleborg, 2023-12-01:
I updated earlier today, no difference, would say that it got a little worse...

xEcho1, 2023-12-02:
Had the same problem when i updated to v1.34.0-1 from 1.33.0-1. My Aqara TRV's would not respond to anything, while my Ikea Tradfri lights would still work. Im using the Sonoff Dongle Plus V2. Restoring my backup of v1.33.0-1 makes them respond again.

Ra72xx, 2023-12-04:
Has anybody tried simply downgrading to the Docker container 1.33.0 and keeping the database? I have no backup to restore...

digitalkaoz, 2023-12-04:
@Ra72xx that didnt work for me, i had to repair all devices

PeterKawa, 2023-12-07:
@Ra72xx Like I wrote in my original post, z2m runs rock solid here on v1.33.0 and older versions. It's the first crap update I encountered.
While the add-on is maintained by volunteers, we can't demand anything, but I agree it is pretty odd there's no response whatsoever yet.
Cheers.

lennartgrunau, 2023-12-07:
JFYI: I was able to downgrade zigbee2mqtt from 1.34.0-1 to 1.33.0-1 by

1. Stop the add-on, disable autostart, auto-update & watchdog
2. In terminal, delete the current zigbee2mqtt container
3. Remove the current 1.34.* image
4. Pull the 1.33.0-1 image
5. Retag the 1.33 image to 1.34
6. Start the addon via home assistant

Z2M reports 1.33 in its about tab. 

With my Nabu Casa Sky Connect, it even reconnected all devices and I didn't need to re-pair anything. 

Of course: If you attempt this, please make a backup and try at your own risk. All complaints about a failed downgrade may be sent to dev/null :)

Ra72xx, 2023-12-08:
Has anybody in this thread also lost the ability to do OTA updates? Looking back, when I started using Z2M, probably with 1.33.0, I could do OTAs, but now that is no longer possible - nothing, not even an error, happens. The update check neither performs updates if newer versions are available nor gives feedback if the version is current (I think I had this in previous versions).

mundschenk-at, 2023-12-08:
@Ra72xx Can you provide logs of an incident? The last few days should still be in the `logs` folder insider your data directory. @Koenkk might also need the `zigbee-herdsman` logs which need to be [manually collected](https://www.zigbee2mqtt.io/guide/usage/debug.html#enabling-logging).

lhodgkins1977, 2023-12-10:
Hi All, 

I am having exactly the same experience with the Sky Connect, moved to it a couple of weeks ago on 2 separate deployments from Hue Hub, one on Pi4 and another on Pi3. Hue IR sensors were dropping every couple of days and had to be re-paired so replaced them with another brand. All devices are now randomly not responding at times then start working again, others need to be power cycled to get them back. All seemed good up to about 20 Zigbee devices then everything was sluggish. Suspected it could be the docker container causing the issue so rebuilt both systems to HAOS bit this as not helped. Updating from 2023.11.x to 2023.12.x seems to have made things worse. 

Often get Network busy or unable to deliver message errors. 

Really hoping this gets fixed to I don't have to go back to the Hue Hub. May try another Zigbee stick.

Alfy1080, 2023-12-10:
@lhodgkins1977 I can't guarantee by any means that this will work for you, but all my issues that i've mentioned in this thread were solved when moving from skyconnect to a Sonoff Dongle-P flashed with the latest firmware. I did have to re-pair all my devices which was no easy task considering the fact that i have around 60 devices but it's been about 3 weeks (without any host reboot/z2m restart) now since i made the move and everything has been running smoothly.

xEcho1, 2023-12-10:
> @lhodgkins1977 I can't guarantee by any means that this will work for you, but all my issues that i've mentioned in this thread were solved when moving from skyconnect to a Sonoff Dongle-P flashed [...]

Are you running on 1.34.0?

Alfy1080, 2023-12-10:
> Are you running on 1.34.0?

Yes

albsch, 2023-12-11:
I experienced the same issues others have encountered since I updated two months ago. My danfoss thermostats stopped communicating randomly. Only a full reboot helped. I downgraded the docker image from 1.33 to 1.30 and since a few days ago everything is stable again. Before, communication would stop after a few hours with at least one device (out of 5).

xEcho1, 2023-12-11:
> Yes

Okay good to know, seems to only be an issue for the Dongle-E.

Ra72xx, 2023-12-11:
Any hope that the SkyConnect will be fixed? (To be sure, there are also users with other devices having problems, but not so often.)

Alfy1080, 2023-12-11:
> > > > @lhodgkins1977 I can't guarantee by any means that this will work for you, but all my issues that i've mentioned in this thread were solved when moving from skyconnect to a Sonoff Dongle-P flashed with the latest firmware. I did have to re-pair all my devices which was no easy task considering the fact that i have around 60 devices but it's been about 3 weeks (without any host reboot/z2m restart) now since i made the move and everything has been running smoothly.
> > > 
> > > 
> > > Are you running on 1.34.0?
> > 
> > 
> > Yes
> 
> Just a quick update, I have swapped out the Sky Connect for the Sonoff Dongle-P flashed with the latest firmware. Migration was simple with the home assistant migrate radio option as I did not need to reconfigure devices and all 63 lights and sensors are working.
> 
> Very early days as only swapped over an hour ago but everything seems better, all sensors and light are responding much faster than before. I will report back in a few days on how its going.
> 
> Many thanks for the comments, motivated me to make the change and it was way simpler than I thought it would be.

Glad this worked out for you so far. But may i ask how did you use the Migrate radio feature of home assistant in zigbee2mqtt? I thought this is an option only for ZHA, not for Z2M as well.

> any docs about "how to flash newest firmware on the Sonoff dongle" ?

There's a home assistant addon for automated firmware flashing. You can watch this youtube guide on how to use it: https://www.youtube.com/watch?v=E39ECyCXJaA&list=LL&index=7

I didn't test this myself as I only found this after i upgraded my dongle to the latest firmware manually, but one of my friends tested this on his Sonoff Dongle-P and all went smoothly.

przemoc86, 2023-12-13:
I'm facing the same issue during last few feeks. I've downgraded to 1.32 and the problem is still there, so maybe damage is already done. I'm adding missing devices once again but after few hours or days they are vanished. Also a lot of devices are being offline during a day, then sometimes being back, sometimes not. It was working for more than one year - no changes with Wifi, Zigbee setup etc. 
I'm using Conbee2

Any ideas? Somebody is looking at this topic or not?

[zigbee.txt](https://github.com/Koenkk/zigbee2mqtt/files/13665361/zigbee.txt)
[image]

Alfy1080, 2023-12-13:
> I'm facing the same issue during last few feeks. I've downgraded to 1.32 and the problem is still there, so maybe damage is already done. I'm adding missing devices once again but after few hours or days they are vanished. Also a lot of devices are being offline during a day, then sometimes being back, sometimes not. It was working for more than one year - no changes with Wifi, Zigbee setup etc. I'm using Conbee2
> 
> Any ideas? Somebody is looking at this topic or not?
> 
> [zigbee.txt](https://github.com/Koenkk/zigbee2mqtt/files/13665361/zigbee.txt) ![2023-12-13 21_15_32-Zigbee2MQTT – Home Assistant – Brave](https://private-user-images.githubusercontent.com/62384501/290319490-1e52b16a-2af3-416b-9e70-aeb0b9726c60.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDI0OTk5MDcsIm5iZiI6MTcwMjQ5OTYwNywicGF0aCI6Ii82MjM4NDUwMS8yOTAzMTk0OTAtMWU1MmIxNmEtMmFmMy00MTZiLTllNzAtYWViMGI5NzI2YzYwLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFJV05KWUFYNENTVkVINTNBJTJGMjAyMzEyMTMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjMxMjEzVDIwMzMyN1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWZkNWE4ZDU5NTNjNDEwMmFlZjg5MzQ2MjYyYjFmOGU0ZWQwYmUwNGNjOTM4MWU5Nzg0NTM1ZDM2OWQ4ODJiOTImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.0U19D9E1GgpReI1-KCHEMEa4nVkZkNbjJGcE0YZL0Pg)

I see on the z2m recommended adapters list that there is this warning for the conbee 2 stick:

Warning: Conbee 2 firmware versions newer than 0x26580700 will result in an unstable network with devices dropping randomly, see [Issue 9554](https://github.com/Koenkk/zigbee2mqtt/issues/9554)

Check the firmware version that you are running, maybe that's the cause of the issue in this case.

przemoc86, 2023-12-13:
@Alfy1080 yes, I saw it - I really spent a lot of time for investigation, so firmware version was also checked by me
[image]

przemoc86, 2023-12-13:
And the problem is not for sure with signal range as I have 3 Hue Bulbs next to each other and 2 are working and 1 is out...
[image]

Alfy1080, 2023-12-13:
> And the problem is not for sure with signal range as I have 3 Hue Bulbs next to each other and 2 are working and 1 is out... ![2023-12-13 21_44_46-Zigbee2MQTT – Home Assistant – Brave](https://private-user-images.githubusercontent.com/62384501/290324074-23fbef27-fdeb-4a93-b571-d67ae6c60802.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDI1MDA2MDIsIm5iZiI6MTcwMjUwMDMwMiwicGF0aCI6Ii82MjM4NDUwMS8yOTAzMjQwNzQtMjNmYmVmMjctZmRlYi00YTkzLWI1NzEtZDY3YWU2YzYwODAyLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFJV05KWUFYNENTVkVINTNBJTJGMjAyMzEyMTMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjMxMjEzVDIwNDUwMlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTRkNTFjYzNlNDE0MmZkOTliNTFmYmNjMGMxYjVhYTliNGNkMWNjM2E5ZWY0ZGUzNzRkOGE5ODA4ODI5NzgwN2UmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.4h-mxNNWDoPlDQlHxLxy2yecFEhu-mEjxYt778_0_3w)

Yeah, signal range is definitely not the root cause. I have around 30 routers spread out in a 86 square meters apartment and i still had this issue with the skyconnect adapter. 

In my troubleshooting i wiped zigbee2mqtt and reconfigured and re-paired everything from scratch and i still had the issue. Until i switched to Sonoff Dongle-P, nothing i've tried worked. Also the issue seemed to get worse as time passed by, as when i first noticed this issue it used to happen once or twice every day but in time that got up to once every half-hour. Since the z2m restart fixed this issue temporarily i can only assume that it was caused by a combination of z2m version and dongle firmware. But since i did not have a backup of a previous z2m version, i can't confirm if reverting to an older version fixes this or not.

All I can say is that if you can/want you could give a CC2652 based dongle a try and see if that fixes the issue or not. If not, just return it and wait for this to be fixed at the Z2M level. At least that was my plan when i ordered my sonoff dongle :)

przemoc86, 2023-12-13:
To be honest probably if I will jump for something new, what will be based on those newer chipsets, the same used by NabuCasa. I know that it's experimental right now, but it should be something more promising for future.
The question is why it was working and stopped - even after downgrading to old Z2M version.

trackhacs, 2023-12-13:
there a two options:

a) wait till it is fixed, which will never happen
b) buy the sonoff-E (currently P here) - and it will be fixed as soon as i switched to the P - make the buy useless 

however - I went for b) - cause I am sick of it.

przemoc86, 2023-12-13:
I would like to change something with LAN - so You propose use something with CC2652P @trackhacs  ?

foothook, 2023-12-13:
Hi,
I have the same problem, driving me nuts for weeks now. I suspected RF interference dropping my zigbee devices connection.

But now I have found out that the devices are still reporting the update of the new state in the z2mqtt Log but are not visualized in the details tab of the device.
also i cannot turn on or off from there, so also not from Homeassistant.

I have a homeassistant yellow with latest versions of homeassistant, os, core and z2mqtt.

see the logs screenshot the green boxed log entries are reported from a working sonoff relais (where i did Press the physical button to turn of and on)

the red boxed is an identical sonoff - that stopped working.. pressing the physical button it appears in the logs but the device detail don't gets the on off change (but does still upate the link quality field).

[image]

a reboot  of z2mqtt addon does temporarly fix the problems but sfter some hours or days another few devices have the problem

trackhacs, 2023-12-13:
> I would like to change something with LAN - so You propose use something with CC2652P @trackhacs ?

further up the thread it has been documented that the P seems to work more stable than the E. 

but what do mean with "LAN" - is still all wireless ...

przemoc86, 2023-12-13:
@trackhacs  not USB stick ;) but connected to Z2M via LAN so I can move terminal in different place ;) so that's why Sonoff P or E is not for me ;) Something like SLZB-05

trackhacs, 2023-12-13:
> @trackhacs not USB stick ;) but connected to Z2M via LAN so I can move terminal in different place ;) so that's why Sonoff P or E is not for me ;) Something like SLZB-05

ah - didnt know this one - interesting indeed. 

however - i am pretty sure its not the chipset which causes this issues, but the z2m in combination of the concrete firmware. looks like your LAN stick also uses the same zigbee base firmware  - and is therefore also affected.

its certainly a bug in z2m - since its been reported to work with older version. its a pitty that ha and z2m breaks very often with working basic stuff trough updates.

Ra72xx, 2023-12-14:
Do I see it right that not only the "experimental" Zigbee coordinators are affected by this problem, but sometimes also "stable" ones? There are now quite a few different people and hardware configurations in this thread to consider this bug not only some rare edge case.

przemoc86, 2023-12-14:
I think so. Conbee2 is described as recommended adapter and it was working since I bough it in 2021 or beginning of 2022

verticalsoft, 2023-12-14:
I use a Tubes CC2652P2 Based Zigbee to PoE Coordinator and am experiancing the issue with the Aqara TRV's becoking unresponsive but update from the devce back to z2m.

Issuing commands to the TRV's does not work, I've over 15 of these and it's now very frustrating.

verticalsoft, 2023-12-14:
I have no backup of the container so I have no idea how to downgrade the z2m so any pointers how to downgrade would be appreciated.

Ra72xx, 2023-12-14:
Yesterday the power had to be turned off for about half an hour in my flat, and once again the Zigbee network, which had stabilized a bit before, is a complete mayhem. Especially almost half of the window sensors and temperature sensors had dropped off (i.e. disappeared, not even marked as "offline" in the UI), and most of the Aqara thermostats were in that strange mode when they are seemingly online, but do no longer interact with Zigbee2MQTT. Unnervingly, some connections which had still worked after the power loss also dropped out hours later - and things still keep dropping out.

All devices can be simply rejoined by putting them and Z2M in pairing mode, then everything is back, including custom names and settings, which means that they only had disappeared from the UI, but were still somehow in the database.

Assuming that the error occured somewhere between 1.33.0 (working) and 1.33.2 (no longer working), the changelog does not immediately offer culprits, maybe something which has nothing to do with specific end devices, but with core or coordinator, e.g.:

https://github.com/Koenkk/zigbee-herdsman/pull/742
https://github.com/Koenkk/zigbee-herdsman/pull/749
https://github.com/Koenkk/zigbee-herdsman/pull/756
https://github.com/Koenkk/zigbee-herdsman/pull/763
or, this explictely for EZSP, but only for 1.33.2:
https://github.com/Koenkk/zigbee-herdsman/pull/771

I'm not a developper, but IMHO it should be doable for somebody who understands the inner workings of Z2M to find a possible reason if you know which version was the last working version.

jmcollin78, 2023-12-20:
I have notice that after a pairing it is working as expected. On the next restart of HA I loose the informations (Aqara T1 which doesn't send the power in my case).

klibro, 2023-12-20:
I have the same issues. Z2M is effectively unusable because of the fact that random devices randomly disappear. This happens pretty much every day. I don't think I've gone much longer than a day without some issue. Re-pairing usually works, and previous settings get picked up, so the information is still in the system somewhere, even when it claims the device doesn't exist. I have a sonoff ZBBridge flashed with Tasmota.

foothook, 2023-12-20:
Update:
i downgraded addon to z2mqtt 1.22.2-1 (had a backup with this version) a week ago and disconnection problems did not come back, so i am almost sure and can confirm there must be a bug introduced with the newer versions mentioned above in the other comments

Ra72xx, 2023-12-21:
Developer feedback 😀in discussion: https://github.com/Koenkk/zigbee2mqtt/discussions/20323

trackhacs, 2023-12-21:
as i said - asap my new stick come - it get fixed and works with the old one ...

thx for putting info here.

PeterKawa, 2024-01-04:
Update v1.35.0 is presented today, but I don't see any reference to either this topic 19747, or the other one https://github.com/Koenkk/zigbee2mqtt/discussions/20323
So, Peter is not updating

[132 more replies]

closed 2025-05-08 by github-actions[bot]