Koenkk/zigbee2mqtt #19750: SmartThings Plugs GP-WOU019BBDWG need to be periodically reconfigured to keep reporting

Ulrar, 2023-11-17:
### What happened?

I have a bunch of SmartThings [GP-WOU019BBDWG](https://www.zigbee2mqtt.io/devices/GP-WOU019BBDWG.html#smartthings-gp-wou019bbdwg) plugs, with energy monitoring.
They work fine and the on/off never fails, however the energy monitoring periodically stops working until I go into Z2M manually and click reconfigure, at which point all the missing data shows up at once and HA thinks the plug just used hundreds of kWh at once. Maybe I should setup an automation to use `zigbee2mqtt/bridge/configure` every day or so ?

I used ZHA for a while (although I've been back on Z2M for well over a year) and I remember noticing this once or twice with it, so I don't think this is caused by Z2M. Maybe there's something that Z2M could do, just reconfigure once or day at midnight automatically for example ?

### What did you expect to happen?

For reporting to keep working without action

### How to reproduce it (minimal and precise)

- Use a [GP-WOU019BBDWG](https://www.zigbee2mqtt.io/devices/GP-WOU019BBDWG.html#smartthings-gp-wou019bbdwg) for a few months
- Notice it stops reporting seemingly randomly every few weeks / months until manually reconfigured

### Zigbee2MQTT version

1.33.2 commit: 9996c93

### Adapter firmware version

20230507

### Adapter

TubesZB poe (older model with cc chip, not the newer one)

hugalafutro, 2023-11-29:
I have the same issue with the same plug, I just wasn't aware about the workaround. For what it's worth this is something that I think changed either in HA or Z2M as I had the plug for years and only noticed this behaviour in last few months as it gotten colder and I had to put the heater on more.

jkpe, 2023-12-09:
FWIW I've seen this behaviour with [GP-WOU019BBDWG](https://www.zigbee2mqtt.io/devices/GP-WOU019BBDWG.html#smartthings-gp-wou019bbdwg) across many Z2M versions and many different Zigbee adapters.

@Ulrar wow, didn't know about using reconfigure to bring it back to life. All this time I have been re-pairing them every 3-4 months.

Ulrar, 2023-12-11:
I'd take the credit but that was suggested to me by the maintainer of zha-toolkit back when I was using ZHA. When the issue happened with Z2M I figured it was worth a shot, and it does indeed work with it too.

It's clearly a firmware issue with those plugs, but I do wonder if Z2M could notice this is happening and trigger a re-configure itself. Come to think of it, maybe this could be caught by an HA automation which would trigger a reconfigure by sending a message on `zigbee2mqtt/bridge/request/device/configure` .. I should try that

jkpe, 2023-12-11:
This is what I've got for now.. stolen from https://github.com/Koenkk/zigbee2mqtt/issues/18315#issuecomment-1686092894

```yaml
alias: Reconfigure power plug if updates stop
description: ""
trigger:
  - platform: template
    value_template: >-
      {{ as_timestamp(now()) -
      as_timestamp(states.sensor['washing_machine_smart_plug_power'].last_updated)|int(0)
      > (3600*24) }}
    id: washing
  - platform: template
    value_template: >-
      {{ as_timestamp(now()) -
      as_timestamp(states.sensor['tumble_dryer_smart_plug_power'].last_updated)|int(0)
      > (3600*24) }}
    id: tumble
condition: []
action:
  - choose:
      - conditions:
          - condition: trigger
            id:
              - washing
        sequence:
          - service: mqtt.publish
            data:
              qos: "2"
              topic: zigbee2mqtt/bridge/request/device/configure
              payload: "{\"id\": \"0xec1bbdfffe41d9fe\"}"
          - service: system_log.write
            data:
              level: warning
              message: >-
                Had to reconfigure washing power plug as it stopped sending
                updates
      - conditions:
          - condition: trigger
            id:
              - tumble
        sequence:
          - service: mqtt.publish
            data:
              qos: "2"
              topic: zigbee2mqtt/bridge/request/device/configure
              payload: "{\"id\": \"0xccccccfffec6d804\"}"
          - service: system_log.write
            data:
              level: warning
              message: >-
                Had to reconfigure tumble power plug as it stopped sending
                updates
mode: single
```

closed 2024-07-10 by github-actions[bot]