Koenkk/zigbee2mqtt #19762: [New device support]: Moes Curtain Switch ZRS-SR-EUC-GY

morpheus302, 2023-11-18:
### Link

https://moeshouse.com/products/zigbee-star-ring-curtain-switch

### Database entry

{"id":26,"type":"Router","ieeeAddr":"0xa4c138c928bd3b8f","nwkAddr":49879,"manufId":4417,"manufName":"_TZE204_srmahpwl","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"�[�,\u0013�[�,e�[�,\u0012","65506":56,"65508":0,"65534":0,"stackVersion":0,"dateCode":"","appVersion":74,"manufacturerName":"_TZE204_srmahpwl","zclVersion":3,"modelId":"TS0601","powerSource":1}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-1440156621},"lastSeen":1700339863157,"defaultSendRequestWhen":"immediate"}

### Comments

Very similar to Moes ZTS-EUR-C
[image]

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const legacy = require('zigbee-herdsman-converters/lib/legacy');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    // Since a lot of TuYa devices use the same modelID, but use different datapoints
    // it's necessary to provide a fingerprint instead of a zigbeeModel
    fingerprint: [
        {
            // The model ID from: Device with modelID 'TS0601' is not supported
            // You may need to add \u0000 at the end of the name in some cases
            modelID: 'TS0601',
            // The manufacturer name from: Device with modelID 'TS0601' is not supported.
            manufacturerName: '_TZE204_srmahpwl',
        },
    ],
    model: 'ZRS-SR-EUC-GY',
    vendor: 'Moes',
	icon: 'ZRS-SR-EUC-GY.jpg',
    description: 'Zigbee + RF curtain switch',
    fromZigbee: [legacy.fz.moes_cover, fz.ignore_basic_report],
    toZigbee: [legacy.tz.moes_cover],
    onEvent: tuya.onEventSetLocalTime,
    configure: tuya.configureMagicPacket,
    exposes: [e.cover_position().setAccess('position', ea.STATE_SET), e.enum('backlight', ea.STATE_SET, ['OFF', 'ON']),
            e.enum('calibration', ea.STATE_SET, ['OFF', 'ON']), e.enum('motor_reversal', ea.STATE_SET, ['OFF', 'ON'])],
    
};

module.exports = definition;
```

tdelsupexhe, 2024-01-19:
`// Add the lines below
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const {} = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: [{modelID: 'TS0601', manufacturerName: '_TZE204_srmahpwl'}],
    model: 'ZS-SR-EUC',
    vendor: 'Moes',
    description: 'Star Ring - Smart Curtain Switch',
    options: [exposes.options.invert_cover()],
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    //onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
    //configure: tuya.configureMagicPacket,
    exposes: [
        // Here you should put all functionality that your device exposes
        //e.cover_position(),
        e.cover_position().setAccess('position', ea.STATE_SET),
        e.enum('calibration', ea.STATE_SET, ['START', 'END']).withDescription('Calibration'),
        //e.binary('backlight_mode', ea.STATE_SET, 'ON', 'OFF').withDescription('Backlight'),
        e.enum('indicator_mode', ea.STATE_SET, ['off', 'on_off', 'off_on']).withDescription('Relay LED indicator mode'),
        e.enum('motor_steering', ea.STATE_SET, ['FORWARD', 'BACKWARD']).withDescription('Motor Steering')
    ],
    meta: {
        // All datapoints go in here
        tuyaDatapoints: [
            [1, 'state', tuya.valueConverterBasic.lookup({'CLOSE': tuya.enum(2), 'STOP': tuya.enum(1), 'OPEN': tuya.enum(0)})], //Curtain Switch 1 
            [2, 'position', tuya.valueConverter.coverPosition], // Percentage 1
            [3, 'calibration', tuya.valueConverterBasic.lookup({'START': tuya.enum(0), 'END': tuya.enum(1)})],, // Accurate Calibration
            [8, 'motor_steering', tuya.valueConverterBasic.lookup({'FORWARD': tuya.enum(0), 'BACKWARD': tuya.enum(1)})], // Motor steering
            // [14, 'backlight_mode', tuya.valueConverter.backlightModeOffNormalInverted], // Indicator Status
            [14, 'indicator_mode', tuya.valueConverterBasic.lookup({'off': tuya.enum(0), 'on_off': tuya.enum(1), 'off_on': tuya.enum(2)})],
        ],
    },
};
    
    module.exports = definition;`

This works to open/close the curtain but the other fonctions not

erkin-o, 2024-08-27:
Is this a working solution, or has a better version already been merged?

closed 2025-01-10 by github-actions[bot]