Koenkk/zigbee2mqtt #19769: OSRAM Smartplug+ doesn't update the state

LeoeLeoeL, 2023-11-19:
### What happened?

Device: https://www.zigbee2mqtt.io/devices/AB3257001NJ.html
If I switch ON or OFF the device using the physical button Z2M doesn't update the state

### What did you expect to happen?

I expect to see the state updated in Z2M/HA when I press the button on the smartplug

### How to reproduce it (minimal and precise)

Pressing the button on the smartplug

### Zigbee2MQTT version

1.33.2

### Adapter firmware version

20230507

### Adapter

Sonoff Zigbee Dongle Plus P

Ra72xx, 2023-11-21:
Maybe related to #19747

LeoeLeoeL, 2023-11-21:
It's different. The smartplug is linked.
If I switch on or off the device from HA or Z2M it works.
The issue is when I switch on or off by physical button Z2M doesn't update the state; I can have the plug ON but Z2M reports OFF.
It's very strange to see this kind of issue in this device because is very old. How is possible nobody reported that yet.

Ra72xx, 2023-11-21:
The problem seems to be similar, however, only the other way round: State updates from the devices do not reach Z2M.
I am new to Z2M, so I can't judge it, but when scrolling the issues list, I see lots of problems which come down to missing state reporting between Z2M and devices and/or missing endpoints of some kind.

LeoeLeoeL, 2023-11-27:
It's very strange lots of devices have this issue.
If they are implemented they should work.
I never had this kind of issue in deconz; many more :-) but not this one.

Ra72xx, 2023-11-28:
The same for me: Not satisfied with deconz because of lots of smaller shortcomings, but after the switch to Z2M a few weeks ago now I have an partially unreliable network.

LeoeLeoeL, 2023-12-01:
I didn't find anything about this device among the fixes in release 1.34.0 but know works and is fast! :-)
Thank you

closed 2023-12-01 by LeoeLeoeL

LeoeLeoeL, 2023-12-10:
The issue stroke only 2 of my 4 osram smartplugs.
Same firmware but (obviously) different positions.
Maybe some router doesn't like to send its state.