Koenkk/zigbee2mqtt #19771: Trying to set up SONOFF SNZB-06P timeout - what's the format?

fribse, 2023-11-19:
### What happened?

I've set up the SONOFF SNZB-06P in Z2M, it is recognized nicely.
There is only one topic in 'Exposes', and that is occupancy

There is no Sensitivity Setting and there is no Timeout setting exposed.
Both of these are available in Sonoff's own Zigbee bridge.
There is some explanation in 'Settings (specific)' that explains
Sends a message after the last time no occupancy (occupancy: false) was detected.
So it looks like you can have it send multiple messages.
It also explains that it should be set to an object like [10,60] and it will then send a message after 10s and after 60s.
But the field shown does not accept objects, only numbers, and setting it to a number doesn't change anything.

### What did you expect to happen?

That I could enter [10] and it would timeout after 10s, or [60] and it would time out after 60s.
When I saw the device was supported, I actually expted it to support the parameters exposed to eWeLink though Sonoff ZBBridge.

### How to reproduce it (minimal and precise)

Just add it to Z2M

### Zigbee2MQTT version

1.33.2

### Adapter firmware version

20230507

### Adapter

TubesZB CC2652P2 POE adapter

Busta999, 2023-11-20:
Same issue.

Tested 2 so far, it reports occupancy indefinitely even when placed in an empty box.

occupancy is permanently on

tjeffree, 2023-11-20:
@Busta999 I got a couple delivered today. They do work ok at reporting occupancy - it's annoying that I can't adjust the timeout yet so they don't "clear" for 60 seconds.

I did have to reset one a couple of times closer to a coordinator to get it to report something other than "N/A". Press and hold the button for ~5 seconds.

nadi1971, 2023-11-20:
Same here.
I tried to configure it on both Z2M and ZHA, the status is always occupancy detected

ramyaladdin, 2023-11-21:
same thing here ,
Its always showing occupancy detected, I did reset it for a couple of times and it gets configured but then shows offline all the time.

Busta999, 2023-11-21:
I have ordered a Sonoff Hub to test the sensors on. 

Identify if I have bad sensors or an issue in zigbee2MQTT.

should have an answer by Wednesday this week.

yanhao666, 2023-11-21:
The minimum timeout time can be set to 30 seconds, both ZHA and Z2M can configure it.
[image]
[image]
And it will be optimized to 15s in the V1.0.5 firmware, sonoff is working on it. OTA may be available in the coming weeks.

yanhao666, 2023-11-21:
> Same issue.
> 
> Tested 2 so far, it reports occupancy indefinitely even when placed in an empty box.
> 
> occupancy is permanently on

In fact, microwave radar can penetrate most non-metallic materials, including walls. Moreover, it has reflection capabilities, meaning that objects in motion behind the sensor or outside of the room may trigger it.

Busta999, 2023-11-21:
> In fact, microwave radar can penetrate most non-metallic materials, including walls. Moreover, it has reflection capabilities, meaning that objects in motion behind the sensor or outside of the room [...]

Good point, I tested the device in a closed cardboard box with sensor pointing away from towards a wall and it continued to report occupancy.

Overnight the sensor reset to clear, when the room was unoccupied.

I have just returned to the room and the sensor did take 60+ seconds to report occupied.

It went to clear very quickly after leaving the room,  No occupancy is set to 1 second.

I’ll have the sonoff hub hopefully today and will test further.

But your point is valid and I suspect correct, the device was continuing to detect presence, even while sat in cardboard box.

If it continues to function as this morning’s testing indicates then this device will perform as desired.

Use a PIR sensor to detect entrance into the space to turn on lights, then use the Sonoff to detect ongoing presence to stop lights from going off.

My main use case is in rooms where people may be still for longer periods than a PIR sensor time out period, for me Lounge, Kitchen, Office.

At this stage I am not so sure either sensor of zigbee2MQTT is malfunctioning, I suspect the sensor is just slow identifying and reporting occupancy, which for its price, is ok and workable.

Will perform further testing on zigbee2MQTT and sonoff hub.

fribse, 2023-11-21:
> I have ordered a Sonoff Hub to test the sensors on.
> 
> Identify if I have bad sensors or an issue in zigbee2MQTT.
> 
> should have an answer by Wednesday this week.

I tested with one of these, and in that I can set sensitivity and duration, so it would be nice to be able to set both of these through "Exposes" or at least in the "settings (specific)"

fribse, 2023-11-21:
> I have ordered a Sonoff Hub to test the sensors on.
> 
> Identify if I have bad sensors or an issue in zigbee2MQTT.
> 
> should have an answer by Wednesday this week.

As I wrote, it was set up with that already, and it works, I can also see it showing clear now, so I guess it's working. And thankyou to all you other guys about how to interpret the explanation on the 'Time' in Settings Specific, seems like the text is simply wrong.

Pedro-vk, 2023-11-22:
Same issue here, the documentation is not very clear about how to configure the timeout format and, apart from that, it always emits a `{"occupancy": true}`

evelant, 2023-11-22:
What's needed to allow zigbee2mqtt to configure the sensitivity of the sensor? The product page lists "Three levels of sensitivity (2.5m, 3.5m, 4m)" but there's no way to configure it yet in z2m

nanderson97651, 2023-11-22:
> What's needed to allow zigbee2mqtt to configure the sensitivity of the sensor? The product page lists "Three levels of sensitivity (2.5m, 3.5m, 4m)" but there's no way to configure it yet in z2m

In addition the marketing materials mention some sort of lux sensor but I don't see it exposed either.

Busta999, 2023-11-22:
In the Sonoff Hub you can automate against Luminance High/Low with no configuration of thresholds.

I’ll check the exact field names.

nanderson97651, 2023-11-22:
> In the Sonoff Hub you can automate against Luminance High/Low with no configuration of thresholds.

I'm using Z2M with SkyConnect and Home Assistant so no Sonoff Hub anywhere in my system.  I was hoping to still be able to automate based on ambient brightness.

Busta999, 2023-11-22:
The hope is that the basic luminance along with presence sensitivity will be added to zigbee2MQTT, I am pretty new to zigbee2MQTT so not yet sure how that happens

nanderson97651, 2023-11-22:
I'm also new to Z2M, switching from ZHA.  Happy to help where I can to get these added but it's pretty new territory for me.

yanhao666, 2023-11-23:
> What's needed to allow zigbee2mqtt to configure the sensitivity of the sensor? The product page lists "Three levels of sensitivity (2.5m, 3.5m, 4m)" but there's no way to configure it yet in z2m

The sensitivity can be configured here.    "1" means 2.5m, "2" means 3.5m, "3" means 4m
[image]

evelant, 2023-11-23:
@yanhao666 thanks! How did you find this information? Or rather, where do you find the information about which clusters/attributes can be used to configure any particular device?

fribse, 2023-11-23:
> 
> The sensitivity can be configured here. "1" means 2.5m, "2" means 3.5m, "3" means 4m [image]

Gold information, thankyou @yanhao666 !

evelant, 2023-11-25:
Hmm it also seems like binding isn't working properly with this sensor. When I try to bind it to a group of lights it doesn't turn them on/off. There's seemingly no way to bind to the on/off cluster of the target. Trying to bind to an individual bulb (ikea tradfri in this case) only shows the "Identify" cluster.

[image]

KmanOz, 2023-11-29:
> The sensitivity can be configured here. "1" means 2.5m, "2" means 3.5m, "3" means 4m [...]

Yes this worked for me. Thank you.

zaberxxx, 2023-11-29:
Mine showing occupied permanently.
I tried chaging the ultrasonicOToUDelay to 30, and ultrasonicUToOThreshold to 1, still no luck.
Will repairing the device to Z2M help?
The firmware is still 1.0.3 and I don't know how to upgrade to 1.0.5 (I don't own a sonoff hub), no upgrade option in Z2M OTA section.

Bod1985, 2023-11-29:
> Mine showing occupied permanently.
>
> I tried chaging the ultrasonicOToUDelay to 30, and ultrasonicUToOThreshold to 1, still no luck.
>
> Will repairing the device to Z2M help?
>
> The firmware is still [...]

I had to re-pair mine a few times to make them work. Then when they started working they were fine until I turned off power and rebooted them.

I don't believe 1.0.5 is available yet. 1.0.3 is the current firmware I think

linkedupbits, 2023-12-01:
> Hmm it also seems like binding isn't working properly with this sensor. When I try to bind it to a group of lights it doesn't turn them on/off. There's seemingly no way to bind to the on/off cluster [...]

I raised a ticket with Sonoff, got this reply:
Sorry, now we could not support this function.

Regards,
Frankie in SONOFF

KmanOz, 2023-12-01:
> Mine showing occupied permanently. I tried chaging the ultrasonicOToUDelay to 30, and ultrasonicUToOThreshold to 1, still no luck. Will repairing the device to Z2M help? The firmware is still 1.0.3 [...]

Mine did that as well. It's very sensitive. I moved the position of mine on the wall and has worked well since.

evelant, 2023-12-01:
@linkedupbits lol thanks for trying, the product page says `【Local Smart Scene】Scene linkage between Zigbee devices can be executed normally even if there is no internet connection.` which I took to mean it supports bindings. Not sure what else it could mean.

Bod1985, 2023-12-02:
Zigbee2MQTT update has enabled OTA for these. 

I've just updated mine to 1.0.5, hopefully there's some improvement. 

Does anyone know if/where Sonoff publish firmware changelogs?

Bod1985, 2023-12-03:
> > Zigbee2MQTT update has enabled OTA for these.
> > I've just updated mine to 1.0.5, hopefully there's some improvement.
> > Does anyone know if/where Sonoff publish firmware changelogs?
> 
> How did you update it? When I try to check for an update it says "Failed to check if update available for 'Office presence sensor' (Unexpected token � in JSON at position 0)".

Strange, all three of my sensors updated without error. Sorry I don't have any suggestions as I haven't done many Z2M OTA upgrades

Bod1985, 2023-12-03:
> What does
> 
> > > > Zigbee2MQTT update has enabled OTA for these.
> > > > I've just updated mine to 1.0.5, hopefully there's some improvement.
> > > > Does anyone know if/where Sonoff publish firmware changelogs?
> > > 
> > > 
> > 
> > Strange, all three of my sensors updated without error. Sorry I don't have any suggestions as I haven't done many Z2M OTA upgrades
> 
> What does the OTA settings page look like in your Z2M?

Like this, I’m running 1.34.0

[image]

hjpfourie, 2023-12-04:
When I read or write the settings, I get a visible indictaor that it has either read, or set the values as per screenshot.

[image]

evelant, 2023-12-05:
@zaberxxx @hjpfourie There's a silly bug in the UI. If you have filtered the logs tab then messages don't show up in the dev console. Go to the logs tab and make sure log level is set to "info" or "debug" and then you will see readouts in the dev console.

yanhao666, 2023-12-08:
> Still not able to make mine sensor working properly. Tried numerous times of repairing, off power, change position, upgrade to v1.0.5, reboot zigbee2mqtt service, still not able to get it work... I also tried change settings is dev console but is it normal that when I write the new number into ultrasonicUToOThreshold, there's no indicator on page showed that I successfully wrote in?
> 
> EDIT: I later tried setting the sensor in a blank space with no moving parts to interfere, and it works properly. This sensor is too sensitive and even the bathroom ventilator/dryer will trigger it.

@zaberxxx After configuring the sensitivity, did it work for you? I mean, in the original place.

KmanOz, 2023-12-08:
I found that after I upgraded to 1.0.5 I have to re-write the register as their firmware seems to have defaulted it back to 3

barcelona92, 2023-12-09:
> What does
> 
> > > > Zigbee2MQTT update has enabled OTA for these.
> > > > I've just updated mine to 1.0.5, hopefully there's some improvement.
> > > > Does anyone know if/where Sonoff publish firmware changelogs?
> > > 
> > > 
> > > How did you update it? When I try to check for an update it says "Failed to check if update available for 'Office presence sensor' (Unexpected token � in JSON at position 0)".
> > 
> > 
> > Strange, all three of my sensors updated without error. Sorry I don't have any suggestions as I haven't done many Z2M OTA upgrades
> 
> What does the OTA settings page look like in your Z2M?
> 
> ![image](https://private-user-images.githubusercontent.com/22861530/287555352-3945430e-149b-4108-af7b-75af4e82fa08.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDIxMjIzMDgsIm5iZiI6MTcwMjEyMjAwOCwicGF0aCI6Ii8yMjg2MTUzMC8yODc1NTUzNTItMzk0NTQzMGUtMTQ5Yi00MTA4LWFmN2ItNzVhZjRlODJmYTA4LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFJV05KWUFYNENTVkVINTNBJTJGMjAyMzEyMDklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjMxMjA5VDExNDAwOFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWM0MTIyOGZkMTY5NDU3NzE2ODNlNmRmYTBkMzVkZWFkYzg4NjY2YWUzOTk0YzJlZWYwZjIyY2EzYTg0MTlmZTImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.XfKEwz4draQ7Kbl6cn1g8i92s-8N9ujV-fWrY8Z0zT8)

Hi,

I have the same problem appear erro with check for update in SNZB-06  the error:
Failed to check if update available for 'Pres entrance' (localIndex.concat(...).map is not a function)

[image]

Typ1er, 2023-12-12:
I can neither read nor write the sensor. what am I doing wrong?
[image]

hjpfourie, 2023-12-12:
> I can neither read nor write the sensor. what am I doing wrong? ![Bildschirmfoto 2023-12-12]

You haven't selected the attribute, only the cluster. Select the attribute you want to read or write.

zaberxxx, 2023-12-16:
> @zaberxxx After configuring the sensitivity, did it work for you? I mean, in the original place.

It works 95% of the time, but sometime false recognition still occurs.

metec22, 2023-12-21:
This device is really disappointing. Is anyone else's sensor behaving like a motion sensor? It's about 2-3 ft away from me and if I sit still, it will stop showing presence (if it's not stuck showing presence for hours when no one is present). I have two of them, one on firmware 	1.0.5 and one on 1.0.3. Both behave the same way.

yanhao666, 2023-12-22:
> This device is really disappointing. Is anyone else's sensor behaving like a motion sensor? It's about 2-3 ft away from me and if I sit still, it will stop showing presence (if it's not stuck [...]

@metec22 It seems possible that the radar microwave has been interfered with. Are there metal objects blocking the middle? Or using another radar nearby at the same time?

evelant, 2023-12-22:
HA 2023.12.x seems to have issues with zigbee, devices go "unavailable" in HA even though z2m is still communicating with them. Other than when the network is working smoothly I don't have any of those issues with these sensors. They respond fast and state stays "occupied" when people are still. I set msOccupancySensing -> ultrasonicUToOThreshold to 3 on all of mine. If you have two or more of the sensors too close to each other or pointed toward each other they can interfere. Any other radar based devices might interfere as well (I think some security systems?).

metec22, 2023-12-22:
Weird. I don't have any metal objects anywhere near the sensor. I only used one sensor at a time in the same spot. It's also much slower than a motion sensor I was hoping to replace. When I walk into the room, it takes about 2-3 seconds for the light to come on. On the updated sensor, I set the time out to 30s and tried all three settings for threshold. On the one with the older firmware, I haven't changed anything. Maybe there is some interference in the room but nothing obvious.

fribse, 2023-12-24:
@metec22 , for me it sounds defective. I have three mw sensors on top of each other while testing, they all work perfectly, and with the released notes on how to adjust sensitivity duration, and the firmware at 1.0.3 it works quite nicely.

Alfy1080, 2024-01-05:
> This device is really disappointing. Is anyone else's sensor behaving like a motion sensor? It's about 2-3 ft away from me and if I sit still, it will stop showing presence (if it's not stuck [...]

I received 6 of these sensors yesterday. I've set up one per room in z2m and upgraded the firmware to v1.0.5. One of them is in a 5 meter long hallway with no interference source whatsoever, but the detection time sometimes is instantaneous and sometimes it takes about 3 seconds to detect presence.
Another one of these sensors is in my bedroom and while awake and present in the room i see that it properly stays as "Detected" all the time until i leave the room. However, overnight when both me and my wife were sleeping in that bedroom, the sensor switched between "Detected" and "Clear" every few minutes. This has happened with detection sensitivity set to high. If i set it to medium or low it would switch to "Clear" even when I am awake on my laptop in bed. Seems an awful lot like a motion sensor at this point. 
I've now realised that the nest hub that i also have in the bedroom might interfere with the presence sensor as it uses radar to detect gestures and monitor sleep so i've turned off these features, but to me it sounds like that kind of interference should keep the presence sensor in "Detected" state instead of making it think the room is clear. Anyway I'll get back with new info after testing it overnight.

Alfy1080, 2024-01-16:
I have another update. The bedroom still behaves pretty much the same, but i realised that the behaviour is very much related to the positioning of the sensor. I got much better results when placing the sensors about 1.5 m high on the wall. The bedroom one still stops detecting me from time to time but the bathroom one is now working flawlessly since i repositioned it. I will keep playing with it and report back, but as far as i can tell this is a good sign not to lose faith in this sensor. ATM 5 out of 6 sensors work exactly as expected.

rjmcfadd, 2024-01-16:
Mine will work amazingly for a little while then go back to detected for hours at a time.

johnrobertcobbold, 2024-01-16:
I also had a few issues when I installed mine but everything has been flawless since I moved it to a slightly different place. It looks like objects / walls were initially bouncing back the waves. I now wish that the lux could be updated at regular intervals.. It now detects my presence but won't update the lux while I am there, effectively preventing any automation from switching on/off the lights if and when it later gets dark (sunset) / brighter (sunrise)

Alfy1080, 2024-01-16:
> Mine will work amazingly for a little while then go back to detected for hours at a time.

This happened to me with sensors that were placed very high on the wall and in the corners of the rooms or with the sensor pointing downwards. I found that the sweet spot for me is around 1.5m height, as far from the corners of the wall as possible and with the sensor pointing straight ahead.

For the light intensity issues, i'm just using the data from other light sensors around my home until the ones in the presence sensor get better. The fact that the light intensity only updates upon presence detection is quite stupid especially for a mains powered device that could send updates to z2m as often as it needed without worrying about battery life.

evelant, 2024-01-16:
[I ripped one open](https://www.reddit.com/r/homeassistant/comments/184svmt/sonoff_snzb06p_teardown/) (for science!) and the photodiode is on the radar board not the mainboard so it may be a firmware thing that it doesn't get illuminance updates until the radar controller sends it to the main board.

nanderson97651, 2024-01-16:
> so it may be a firmware thing that it doesn't get illuminance updates until the radar controller sends it to the main board

This is why I opened a support ticket with Sonoff directly asking if it can be changed in a future firmware update.  It may be a long shot but they've already updated fw once to reduce the cooldown.  I recommend more people submit a request [here](https://sonoff.tech/support/) to decouple photodetector updates from radar updates.

yanhao666, 2024-01-17:
> I have another update. The bedroom still behaves pretty much the same, but i realised that the behaviour is very much related to the positioning of the sensor. I got much better results when placing [...]

Maybe, when the sensor is close to the chest of a person, the detection of presence/breathing will be more effective.

charlidog, 2024-01-30:
There is a useful update [here on the sonoff website](https://sonoff.tech/product-review/tutorial/snzb-06p-firmware-upgrade-and-home-assistant-operation-guide/#2) - the latest firmware 1.0.6 - seems to work well for me

Alfy1080, 2024-01-30:
> There is a useful update [here on the sonoff website](https://sonoff.tech/product-review/tutorial/snzb-06p-firmware-upgrade-and-home-assistant-operation-guide/#2) - the latest firmware 1.0.6 - seems to work well for me

I can't see any mention of v1.0.6 on that page. Also how did you update? I have v1.0.5 in Z2M and checking for new versions doesn't give me anything.

charlidog, 2024-01-30:
I did the update manually as per the OTA update ZHA section
>Waiting for the firmware to update automatically may take a considerable amount of time. We recommend proceeding with a manual update. To begin, you’ll need to copy the device’s IEEE address.

evelant, 2024-01-30:
There is no version 1.0.6 as far as I can tell? 1.0.5 is the latest

hjpfourie, 2024-01-30:
No option to manually update through Z2M from what I can tell. Z2M also doesn't detect a new firmware update (yet).

kennethpangky, 2024-02-05:
> There is no version 1.0.6 as far as I can tell? 1.0.5 is the latest

Mine came ootb with 1.0.6. Also having the same issue of being stuck on motion detected. Also, it doesn't even expose the occupany entity.

Alfy1080, 2024-02-06:
> > There is no version 1.0.6 as far as I can tell? 1.0.5 is the latest
> 
> Mine came ootb with 1.0.6. Also having the same issue of being stuck on motion detected. Also, it doesn't even expose the occupany entity.

The motion entity is the occupancy entity. You just have to change the entity type in HA.

kennethpangky, 2024-02-08:
> > > There is no version 1.0.6 as far as I can tell? 1.0.5 is the latest
> > 
> > 
> > Mine came ootb with 1.0.6. Also having the same issue of being stuck on motion detected. Also, it doesn't even expose the occupany entity.
> 
> The motion entity is the occupancy entity. You just have to change the entity type in HA.

I see. Because for those who have it connected to ZHA it shows both motion and occupancy. And apparently the occupancy entity works but the motion one doesn't, which is why I was looking for that entity.

If it's a matter of just changing the entity type then it wouldn't work anyway. It's always stuck in detected state.

Sr-psycho, 2024-02-19:
Anyone whose sensor is stuck in the “detected” state should go to Settings (specific) and select the required time in minutes after which the sensor will report that the object is missing. The sensor with this parameter has been working for the second week without errors. HA rebooted many times during this time. All OK.

[7 more replies]

closed 2024-11-10 by github-actions[bot]