Koenkk/zigbee2mqtt #19780: ZNCLDJ12LM (Aqara B1 Curtain Motor) missing state attributes battery, motor_state, running and voltage

allistermaguire, 2023-11-20:
### What happened?

I recently added another [ZNCLDJ12LM](https://www.zigbee2mqtt.io/devices/ZNCLDJ12LM.html#xiaomi-zncldj12lm) device and it is missing some state attributes: `battery`, `motor_state`, `running` and `voltage`. This is what it has:

```
{
    "linkquality": 144,
    "options": {
        "hand_open": true,
        "reverse_direction": false
    },
    "position": 94,
    "update": {
        "installed_version": 24,
        "latest_version": 24,
        "state": "idle"
    }
}
```

### What did you expect to happen?

I have six other B1's and they are all returning all the expected state attributes.
```
{
    "battery": 91,
    "linkquality": 144,
    "motor_state": "stop",
    "options": {
        "hand_open": true,
        "reverse_direction": false
    },
    "position": 0,
    "power_outage_count": 37,
    "running": false,
    "state": "OFF",
    "update": {
        "installed_version": 24,
        "latest_version": 24,
        "state": "idle"
    },
    "voltage": 3000
}
```

### How to reproduce it (minimal and precise)

The last of the working B1 was paired when I was running version 1.28.4.

I have tried pairing it multiple times, and it always has the same results.

### Zigbee2MQTT version

1.33.2

### Adapter firmware version

6.10.3.0 build 297

### Adapter

ITead Sonoff Zigbee 3.0 USB Dongle Plus V2 model "ZBDongle-E"

### Debug log

Debug for B1 with missing state attributes:
```
Debug 2023-11-20 15:43:31Received MQTT message on 'zigbee2mqtt/0x04cf8cdf3c8141e0/set' with data '{ "position": 89 }'
Debug 2023-11-20 15:43:31Publishing 'set' 'position' to '0x04cf8cdf3c8141e0'
Debug 2023-11-20 15:43:32Received MQTT message on 'zigbee2mqtt/0x04cf8cdf3c8141e0/set' with data '{ "position": 85 }'
Debug 2023-11-20 15:43:32Publishing 'set' 'position' to '0x04cf8cdf3c8141e0'
Debug 2023-11-20 15:43:33Received Zigbee message from '0x04cf8cdf3c8141e0', type 'attributeReport', cluster 'genAnalogOutput', data '{"61440":131072,"presentValue":0}' from endpoint 1 with groupID 0
Debug 2023-11-20 15:43:33Received Zigbee message from '0x04cf8cdf3c8141e0', type 'readResponse', cluster 'genAnalogOutput', data '{"presentValue":94}' from endpoint 1 with groupID 0
Info 2023-11-20 15:43:33MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c8141e0', payload '{"battery":null,"linkquality":148,"motor_state":null,"options":{"hand_open":true,"reverse_direction":false},"position":94,"power_outage_count":null,"running":null,"state":null,"update":{"installed_version":24,"latest_version":24,"state":"idle"},"update_available":null}'
Debug 2023-11-20 15:43:35Received Zigbee message from '0x00158d000267522f', type 'read', cluster 'genTime', data '["time"]' from endpoint 1 with groupID 0
Debug 2023-11-20 15:43:35Received Zigbee message from '0x04cf8cdf3c8141e0', type 'attributeReport', cluster 'genAnalogOutput', data '{"61440":131072,"presentValue":2}' from endpoint 1 with groupID 0
Debug 2023-11-20 15:43:35Received Zigbee message from '0x04cf8cdf3c8141e0', type 'attributeReport', cluster 'genAnalogOutput', data '{"61440":131072,"presentValue":0}' from endpoint 1 with groupID 0
Debug 2023-11-20 15:43:35Received Zigbee message from '0x04cf8cdf3c8141e0', type 'readResponse', cluster 'genAnalogOutput', data '{"presentValue":85}' from endpoint 1 with groupID 0
Info 2023-11-20 15:43:35MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c8141e0', payload '{"battery":null,"linkquality":148,"motor_state":null,"options":{"hand_open":true,"reverse_direction":false},"position":85,"power_outage_count":null,"running":null,"state":null,"update":{"installed_version":24,"latest_version":24,"state":"idle"},"update_available":null}'
Debug 2023-11-20 15:43:35Received Zigbee message from '0x04cf8cdf3c8141e0', type 'readResponse', cluster 'genAnalogOutput', data '{"presentValue":85}' from endpoint 1 with groupID 0
Info 2023-11-20 15:43:35MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c8141e0', payload '{"battery":null,"linkquality":148,"motor_state":null,"options":{"hand_open":true,"reverse_direction":false},"position":85,"power_outage_count":null,"running":null,"state":null,"update":{"installed_version":24,"latest_version":24,"state":"idle"},"update_available":null}'
```
Debug for fully working device 
```
Debug 2023-11-20 15:44:22Received MQTT message on 'zigbee2mqtt/0x54ef44100007151e/set' with data '{ "position": 93 }'
Debug 2023-11-20 15:44:22Publishing 'set' 'position' to '0x54ef44100007151e'
Debug 2023-11-20 15:44:23Received MQTT message on 'zigbee2mqtt/0x54ef44100007151e/set' with data '{ "position": 85 }'
Debug 2023-11-20 15:44:23Publishing 'set' 'position' to '0x54ef44100007151e'
Debug 2023-11-20 15:44:25Received Zigbee message from '0x54ef44100007151e', type 'attributeReport', cluster 'genMultistateOutput', data '{"61440":9818624,"presentValue":2}' from endpoint 1 with groupID 0
Info 2023-11-20 15:44:25MQTT publish: topic 'zigbee2mqtt/0x54ef44100007151e', payload '{"battery":89,"linkquality":208,"motor_state":"stop","options":{"hand_open":true,"reverse_direction":false},"position":96,"power_outage_count":19,"running":false,"state":"OFF","update":{"installed_version":24,"latest_version":24,"state":"idle"},"update_available":null,"voltage":3000}'
Debug 2023-11-20 15:44:25Received Zigbee message from '0x54ef44100007151e', type 'attributeReport', cluster 'genAnalogOutput', data '{"61440":131072,"presentValue":85}' from endpoint 1 with groupID 0
Info 2023-11-20 15:44:25MQTT publish: topic 'zigbee2mqtt/0x54ef44100007151e', payload '{"battery":89,"linkquality":208,"motor_state":"stop","options":{"hand_open":true,"reverse_direction":false},"position":85,"power_outage_count":19,"running":false,"state":"OFF","update":{"installed_version":24,"latest_version":24,"state":"idle"},"update_available":null,"voltage":3000}'
Debug 2023-11-20 15:44:25Received Zigbee message from '0x54ef44100007151e', type 'readResponse', cluster 'genAnalogOutput', data '{"presentValue":85}' from endpoint 1 with groupID 0
Info 2023-11-20 15:44:25MQTT publish: topic 'zigbee2mqtt/0x54ef44100007151e', payload '{"battery":89,"linkquality":208,"motor_state":"stop","options":{"hand_open":true,"reverse_direction":false},"position":85,"power_outage_count":19,"running":false,"state":"OFF","update":{"installed_version":24,"latest_version":24,"state":"idle"},"update_available":null,"voltage":3000}'
```

allistermaguire, 2023-11-20:
Logs from pairing:

```
Info 2023-11-20 16:43:11Device '0x04cf8cdf3c8141e0' joined
Info 2023-11-20 16:43:11MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x04cf8cdf3c8141e0","ieee_address":"0x04cf8cdf3c8141e0"},"type":"device_joined"}'
Info 2023-11-20 16:43:11Starting interview of '0x04cf8cdf3c8141e0'
Info 2023-11-20 16:43:11MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x04cf8cdf3c8141e0","ieee_address":"0x04cf8cdf3c8141e0","status":"started"},"type":"device_interview"}'
Info 2023-11-20 16:43:23Successfully interviewed '0x04cf8cdf3c8141e0', device has successfully been paired
Info 2023-11-20 16:43:23Device '0x04cf8cdf3c8141e0' is supported, identified as: Xiaomi Aqara B1 curtain motor (ZNCLDJ12LM)
Info 2023-11-20 16:43:23MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Aqara B1 curtain motor","exposes":[{"features":[{"access":7,"label":"State","name":"state","property":"state","type":"enum","values":["OPEN","CLOSE","STOP"]},{"access":7,"description":"Position of this cover","label":"Position","name":"position","property":"position","type":"numeric","unit":"%","value_max":100,"value_min":0}],"type":"cover"},{"access":1,"description":"Remaining battery in %, can take up to 24 hours before reported.","label":"Battery","name":"battery","property":"battery","type":"numeric","unit":"%","value_max":100,"value_min":0},{"access":1,"description":"Whether the motor is moving or not","label":"Running","name":"running","property":"running","type":"binary","value_off":false,"value_on":true},{"access":1,"description":"The current state of the motor.","label":"Motor state","name":"motor_state","property":"motor_state","type":"enum","values":["closing","opening","stop"]},{"access":1,"description":"Number of power outages (since last pairing)","label":"Power outage count","name":"power_outage_count","property":"power_outage_count","type":"numeric"},{"access":1,"description":"Link quality (signal strength)","label":"Linkquality","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"ZNCLDJ12LM","options":[{"access":2,"description":"Inverts the cover position, false: open=100,close=0, true: open=0,close=100 (default false).","label":"Invert cover","name":"invert_cover","property":"invert_cover","type":"binary","value_off":false,"value_on":true}],"supports_ota":true,"vendor":"Xiaomi"},"friendly_name":"0x04cf8cdf3c8141e0","ieee_address":"0x04cf8cdf3c8141e0","status":"successful","supported":true},"type":"device_interview"}'
Debug 2023-11-20 16:44:04Received Zigbee message from '0x04cf8cdf3c8141e0', type 'attributeReport', cluster 'genBasic', data '{"powerSource":3}' from endpoint 1 with groupID 0
```

and battery from the Aqara home app
[image]

maxs67, 2023-12-17:
Still no fix, have the same problem with an new installation.

allistermaguire, 2023-12-18:
@maxs67 Good to know I am not the only person with the issue. Are you using an EZSP Adapter? I am trying to narrow down if it is the Zigbee coordinator or something else.

maxs67, 2023-12-19:
@allistermaguire I'm using a zigstar adapter. I had to install domoticz compleet new, and on the new installation I don't get the  battery, motor_state, power_outage_count, running, state and voltage. In the old installation was all ok. I don't know where to look for a solution.

allistermaguire, 2023-12-19:
Thanks @maxs67, that is based on Texas Instruments chipset, mine is Silicon Labs which is still considered experimental as it doesn’t have backup/restore functionality with zigbee2mqtt.

My plan Z was to change my adapter, but now I know that won’t help.

I at least know which version it worked with and when it stopped, it isn’t a massive jump but still a reasonable number of PR’s.

maxs67, 2024-02-04:
@allistermaguire can I rolback to the working version? with version it was?

allistermaguire, 2024-02-07:
Hi @maxs67, the version I was running when I added my last ZNCLDJ12LM successfully was `1.28.4`.

I have since tested rolling back to that version, and my most recent ZNCLDJ12LM still doesn't have all the attributes.

zvezdochet2004, 2024-03-10:
Hello. I have the same problem.
Current version: 1.36.0-1

allistermaguire, 2024-03-11:
Hi @zvezdochet2004 can you provide some additional information on your setup? ie: do you have Aqara B1's that have all attributes and some with missing attributes? did they in the past? what sort of adapter are you using? etc

I am using EZSP, and the latest version has introduced Ember which is only supported with the latest adapter firmware, so I can not go back easily to a previous version.

allistermaguire, 2024-03-11:
Thanks @zvezdochet2004. The 6 month timeframe aligns with what I see, but I only have 1 of 6 B1's with missing attributes which was the last one added.

For yours to stop working; did they just stop? or did you have to repair them for some reason, and then once they were repaired they had missing attributes?

I have not wanted to try repairing a B1's that is currently reporting all attributes, in case it stops.

zvezdochet2004, 2024-03-11:
I don’t remember such details anymore.  I reconnected the devices and at some point noticed that the battery level attribute was missing

closed 2024-10-09 by github-actions[bot]