Koenkk/zigbee2mqtt #19787: AVATTO ME168 - bad temperature values when schedule is selected

zett93, 2023-11-20:
### What happened?

Hi!

I recently purchased several Avatto thermostatic valves. Everything works fine except for one stupid thing:
When I change from heating to automatic, the temperature at my valve always increases to 153.6 degrees.

Changing the hours or temperatures doesn't change anything. I use SONOFF Zigbee 3.0 USB Dongle Plus ZBDongle-P.

The problem was reproduced on my 7 ME168 valves. Always the same.

Below I have pasted an example schedule from my z2m configuration:

`00:00/16.0 07:00/18.0 15:00/19.0 18:00/18.0`

And some logs from my z2m (after setting a schedule)

Look at the last line.

Regards!

### What did you expect to happen?

I expected that my schedule will set proper temperature :) 

### How to reproduce it (minimal and precise)

Just set values on the valve (e.q. via "exposes" tab in z2m, or via MQTT `zigbee2mqtt/FRIENDLY_NAME/set`

### Zigbee2MQTT version

1.33.2 commit: 311ea07

### Adapter firmware version

20210708

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus ZBDongle-P

### Debug log

`debug 2023-11-20 14:55:37Publishing 'set' 'schedule_monday' to 'termostat-gabinet'
debug 2023-11-20 14:55:39Received MQTT message on 'zigbee2mqtt/termostat-gabinet/set' with data '{"system_mode":"auto"}'
debug 2023-11-20 14:55:39Publishing 'set' 'system_mode' to 'termostat-gabinet'
info 2023-11-20 14:55:40MQTT publish: topic 'zigbee2mqtt/termostat-gabinet', payload '{"battery_low":false,"child_lock":"UNLOCK","current_heating_setpoint":21,"error":null,"frost_protection":"OFF","linkquality":75,"local_temperature":21,"local_temperature_calibration":0,"running_state":"heat","scale_protection":"OFF","schedule_friday":"00:00/16.0 07:00/19.0 09:00/20.0 18:00/18.0","schedule_monday":"00:00/16.0 07:00/18.0 15:00/19.0 18:00/18.0","schedule_saturday":"00:00/16.0 08:00/18.0 10:00/20.0 18:00/18.0","schedule_sunday":"00:00/16.0 08:00/18.0 10:00/20.0 18:00/18.0","schedule_thursday":"00:00/16.0 07:00/18.0 15:00/19.0 18:00/18.0","schedule_tuesday":"00:00/16.0 07:00/18.0 15:00/19.0 18:00/18.0","schedule_wednesday":"00:00/16.0 07:00/19.0 09:00/20.0 18:00/18.0","system_mode":"heat"}'
debug 2023-11-20 14:55:40Received Zigbee message from 'termostat-gabinet', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[1,0,0,0,160,7,0,0,180,15,0,0,190,18,0,0,180],"type":"Buffer"},"datatype":0,"dp":28}],"seq":19712}' from endpoint 1 with groupID 0
info 2023-11-20 14:55:40MQTT publish: topic 'zigbee2mqtt/termostat-gabinet', payload '{"battery_low":false,"child_lock":"UNLOCK","current_heating_setpoint":21,"error":null,"frost_protection":"OFF","linkquality":69,"local_temperature":21,"local_temperature_calibration":0,"running_state":"heat","scale_protection":"OFF","schedule_friday":"00:00/16.0 07:00/19.0 09:00/20.0 18:00/18.0","schedule_monday":"00:00/16.0 07:00/18.0 15:00/19.0 18:00/18.0","schedule_saturday":"00:00/16.0 08:00/18.0 10:00/20.0 18:00/18.0","schedule_sunday":"00:00/16.0 08:00/18.0 10:00/20.0 18:00/18.0","schedule_thursday":"00:00/16.0 07:00/18.0 15:00/19.0 18:00/18.0","schedule_tuesday":"00:00/16.0 07:00/18.0 15:00/19.0 18:00/18.0","schedule_wednesday":"00:00/16.0 07:00/19.0 09:00/20.0 18:00/18.0","system_mode":"heat"}'
info 2023-11-20 14:55:40MQTT publish: topic 'zigbee2mqtt/termostat-gabinet', payload '{"battery_low":false,"child_lock":"UNLOCK","current_heating_setpoint":21,"error":null,"frost_protection":"OFF","linkquality":69,"local_temperature":21,"local_temperature_calibration":0,"running_state":"heat","scale_protection":"OFF","schedule_friday":"00:00/16.0 07:00/19.0 09:00/20.0 18:00/18.0","schedule_monday":"00:00/16.0 07:00/18.0 15:00/19.0 18:00/18.0","schedule_saturday":"00:00/16.0 08:00/18.0 10:00/20.0 18:00/18.0","schedule_sunday":"00:00/16.0 08:00/18.0 10:00/20.0 18:00/18.0","schedule_thursday":"00:00/16.0 07:00/18.0 15:00/19.0 18:00/18.0","schedule_tuesday":"00:00/16.0 07:00/18.0 15:00/19.0 18:00/18.0","schedule_wednesday":"00:00/16.0 07:00/19.0 09:00/20.0 18:00/18.0","system_mode":"auto"}'
debug 2023-11-20 14:55:40Received Zigbee message from 'termostat-gabinet', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":4,"dp":2}],"seq":19968}' from endpoint 1 with groupID 0
info 2023-11-20 14:55:40MQTT publish: topic 'zigbee2mqtt/termostat-gabinet', payload '{"battery_low":false,"child_lock":"UNLOCK","current_heating_setpoint":21,"error":null,"frost_protection":"OFF","linkquality":69,"local_temperature":21,"local_temperature_calibration":0,"running_state":"heat","scale_protection":"OFF","schedule_friday":"00:00/16.0 07:00/19.0 09:00/20.0 18:00/18.0","schedule_monday":"00:00/16.0 07:00/18.0 15:00/19.0 18:00/18.0","schedule_saturday":"00:00/16.0 08:00/18.0 10:00/20.0 18:00/18.0","schedule_sunday":"00:00/16.0 08:00/18.0 10:00/20.0 18:00/18.0","schedule_thursday":"00:00/16.0 07:00/18.0 15:00/19.0 18:00/18.0","schedule_tuesday":"00:00/16.0 07:00/18.0 15:00/19.0 18:00/18.0","schedule_wednesday":"00:00/16.0 07:00/19.0 09:00/20.0 18:00/18.0","system_mode":"auto"}'
debug 2023-11-20 14:55:59Received Zigbee message from 'termostat-gabinet', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,6,0],"type":"Buffer"},"datatype":2,"dp":4}],"seq":20224}' from endpoint 1 with groupID 0
info 2023-11-20 14:55:59MQTT publish: topic 'zigbee2mqtt/termostat-gabinet', payload '{"battery_low":false,"child_lock":"UNLOCK","current_heating_setpoint":153.6,"error":null,"frost_protection":"OFF","linkquality":66,"local_temperature":21,"local_temperature_calibration":0,"running_state":"heat","scale_protection":"OFF","schedule_friday":"00:00/16.0 07:00/19.0 09:00/20.0 18:00/18.0","schedule_monday":"00:00/16.0 07:00/18.0 15:00/19.0 18:00/18.0","schedule_saturday":"00:00/16.0 08:00/18.0 10:00/20.0 18:00/18.0","schedule_sunday":"00:00/16.0 08:00/18.0 10:00/20.0 18:00/18.0","schedule_thursday":"00:00/16.0 07:00/18.0 15:00/19.0 18:00/18.0","schedule_tuesday":"00:00/16.0 07:00/18.0 15:00/19.0 18:00/18.0","schedule_wednesday":"00:00/16.0 07:00/19.0 09:00/20.0 18:00/18.0","system_mode":"auto"}'`

drupry, 2023-11-21:
I have the same problem. It says there should be only 4 temperature transitions (no more and no less!) and one of my valve sets not to 153.6 but 0 when the schedule hits idk why.

robbi55, 2023-11-22:
Hello, I have the same issue. Do u find any solution?
Thanks

schablome, 2023-11-25:
Hello. Same issue here. 2 x Avatto ME168 with SONOFF Zigbee 3.0 USB Dongle Plus.
One ist setting 153.6 °C for target when switching to "auto", the other is setting 0°C for target.
Beside that both devices often do not accept set commands, whether from "exposes" in Zigbee2MQTT or from changing the entity.
LQI is around 50 for both devices.
I am also very interested in a solution. Thanks.

Micha-HGW, 2023-11-28:
Hello Koen,
I have the same problem with Avatto [ME168] and Sonoff Zigbee 3.0 USB Dongle.
The problem persists even though I brought the dongle close to the valves. The signal strength is now 96 to 160.
A solution would be great!
LG Micha

Skyr, 2023-11-29:
Same problem here: When the scheduler hits, I first see an mqtt message with a current_heating_setpoint with value "auto", and right after that a second message with a setpoint of 152.6 degrees.
I'm using the latest Docker image (package.json says 1.33.2), zigbee dongle: Slaesh's CC2652RB stick.

sl4m01, 2023-11-30:
Hello I also do have exactly the same issue with my new EM168 discovered device. Turning on "auto" mode results in target temperature 0.

zett93, 2023-12-01:
As a workaround i created flow in NodeRED, that changes temperature values in heating mode. I smart-scheduler and function nodes, if someone is interested in, please comment, i'll share my code

sychu, 2023-12-01:
I have exactly the same behawior as @zett93 described. I'm using SONOFF Zigbee 3.0 USB Dongle Plus-E.
What I also noticed that my device do not match picture from z2m devices page. So it is detected as [AVATTO  ME168](https://www.zigbee2mqtt.io/devices/ME168.html) but in reality  it look like on picture from [AVATTO ME167](https://www.zigbee2mqtt.io/devices/ME167.html).

Z2M About:
**Zigbee Model:** TS0601
**Zigbee Manufacturer:**_TZE200_p3dbf6qs
**Description:** Thermostatic radiator valve
**IEEE Address:** 0xa4c1381e73b5d3f6
**Network address:** 0x13F6
**Manufacturer:** [AVATTO](https://www.zigbee2mqtt.io/supported-devices/#v=AVATTO)
**Model:** [ME168](https://www.zigbee2mqtt.io/devices/ME168.html#avatto-me168)

ecotwist, 2023-12-01:
@sychu - same situation with models and pictures
I also noticed that the description of schedule says 6 values ('HH:MM/C HH:MM/C HH:MM/C HH:MM/C HH:MM/C HH:MM/C'), but the setting is only saved if I set only 4 values ('HH:MM/C HH:MM/C HH:MM/C HH:MM/C')

drupry, 2023-12-01:
I can confirm @sychu - it says ME167 in my manual given with the valves.
@zett93 the automation can be also very easily done with scheduler card but it would be nice to set it to the valve without HA interfeering

sychu, 2023-12-02:
> I have play around with this and it look like that main issue is that converter sends 4 values instead 6 (just like @ecotwist mentioned). I created custom converter which sends 6 values and it seem to work for me. I guessed some values for metadata days mappings. It work at least today (for Friday) but i have no idea what will happen tomorrows 😆. If anyone want to help with testing then I published fixed version here: https://github.com/sychu/avatto_me167_TZE200_p3dbf6qs/tree/main

Just updated code so it should work for Fridays and Saturdays also 😆. Few more days to check.

hacker, 2023-12-02:
> I have play around with this and it look like that main issue is that converter sends 4 values instead 6 (just like @ecotwist mentioned). I created custom converter which sends 6 values and it seem [...]

I can confirm that it does seem to work. The setpoint is way below 153ºC now :)

michaljuris, 2023-12-02:
> Just updated code so it should work for Fridays and Saturdays also 😆. Few more days to check.

I can also confirm that this works perfectly with my TZE200_p3dbf6qs/ - THX a lot for your fix @sychu 

Just one small note to others - when adding the external converter record - use full filename including extension (.js)
(README suggests it should be omitted )

michaljuris, 2023-12-02:
> avatto_me167

That's what was I mentioning in my previous post - when adding external convertor name, include extension (.js) - _avatto_me167**.js**_

That's why your z2m cannot find the file. README is wrong on this in 3rd step of the installation guide

mwxp06, 2023-12-02:
@SirUlbrich 
Be carefull about right format and content of file you downloaded from github.
And do not forget to extension.

For me this fix has solved problems! Thank you @sychu

SirUlbrich, 2023-12-02:
> > avatto_me167
> 
> That's what was I mentioning in my previous post - when adding external convertor name, include extension (.js) - _avatto_me167**.js**_
> 
> That's why your z2m cannot find the file. README is wrong on this in 3rd step of the installation guide

you posted after i did lol

SirUlbrich, 2023-12-02:
how did i change the external? GUI not open anymore. reinstall not working too =(

michaljuris, 2023-12-02:
> how did i change the external? GUI not open anymore. reinstall not working too =(

you need to open configuration.yaml and edit it manually there

[image]

SirUlbrich, 2023-12-02:
> > how did i change the external? GUI not open anymore. reinstall not working too =(
> 
> @SirUlbrich use File Editor. Open `/homeassistant/zigbee2mqtt/configuration.yaml` and look for ` external_converters:` ` - avatto-me168.js`

thx both of you. i found in configuration by samba

SirUlbrich, 2023-12-02:
@sychu did you delete battery low? in code i see:
```
        tuyaDatapoints: [
            [2, 'system_mode', tuya.valueConverterBasic.lookup({'auto': tuya.enum(0), 'heat': tuya.enum(1), 'off': tuya.enum(2)})],
            [3, 'running_state', tuya.valueConverterBasic.lookup({'heat': tuya.enum(0), 'idle': tuya.enum(1)})],
            [4, 'current_heating_setpoint', tuya.valueConverter.divideBy10],
            [5, 'local_temperature', tuya.valueConverter.divideBy10],
            [7, 'child_lock', tuya.valueConverter.lockUnlock],           
            [28, 'schedule_wednesday', fixedValueConverter.thermostatScheduleDayMultiDPWithDayNumber(1)],
            [29, 'schedule_thursday', fixedValueConverter.thermostatScheduleDayMultiDPWithDayNumber(2)],
            [30, 'schedule_friday', fixedValueConverter.thermostatScheduleDayMultiDPWithDayNumber(3)],
            [31, 'schedule_saturday', fixedValueConverter.thermostatScheduleDayMultiDPWithDayNumber(4)],
            [32, 'schedule_sunday', fixedValueConverter.thermostatScheduleDayMultiDPWithDayNumber(5)],
            [33, 'schedule_monday', fixedValueConverter.thermostatScheduleDayMultiDPWithDayNumber(6)],
            [34, 'schedule_tuesday', fixedValueConverter.thermostatScheduleDayMultiDPWithDayNumber(7)],  
            [35, null, tuya.valueConverter.errorOrBatteryLow],
            [36, 'frost_protection', tuya.valueConverter.onOff],
            [39, 'scale_protection', tuya.valueConverter.onOff],
            [47, 'local_temperature_calibration', tuya.valueConverter.localTempCalibration2],
```
row 35, "null"

sychu, 2023-12-02:
>r ow 35, "null"
@SirUlbrich no, but it's interesting. I will check history for this line in original repo.

TrolliRolli, 2023-12-21:
Dear all,
thx, the converter have fixed my issue too.
@sychu - In addition the AVATTO ME167, listed below, does really only accept 4 schedules a day but the z2m gui configuration example is showing 6 schedules (Schedule for monday, format: "HH:MM/C HH:MM/C HH:MM/C HH:MM/C HH:MM/C HH:MM/C")
"device": {
     "applicationVersion": 67,
     "dateCode": "",
     "friendlyName": "xxx",
     "hardwareVersion": 1,
     "ieeeAddr": "xxx",
     "manufacturerID": 4417,
     "manufacturerName": "_TZE200_6rdj8dzm",
     "model": "ME167",
     "networkAddress": xxx,
     "powerSource": "Battery",
     "stackVersion": 0,
     "type": "EndDevice",
     "zclVersion": 3
 }

Thanks to all for your Work! and have a good Christmas time

Koenkk, 2023-12-21:
@sychu could you create a PR to integrate this fix?

TrolliRolli, 2023-12-21:
sure

TrolliRolli, 2023-12-21:
> @sychu could you create a PR to integrate this fix?

done => [https://github.com/Koenkk/zigbee2mqtt/issues/20338](https://github.com/Koenkk/zigbee2mqtt/issues/20338)

stertinger, 2023-12-22:
Hi Guys, 

I did all steps like in instruction, on my devices list devices it changed from ME 168 to ME 167 and it's OK but can't do last step: Go to device and reconfigure 🔄️ - it doesn't work. 

Failed to configure (Read 0xa4c138322bc22c82/1 genBasic(["manufacturerName","zclVersion","appVersion","modelId","powerSource",65534], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 50867 - 1 - 67 - 0 - 1 after 10000ms))

and 

Failed to configure 'Termostat biuro', attempt 5 (Error: Read 0xa4c138322bc22c82/1 genBasic(["manufacturerName","zclVersion","appVersion","modelId","powerSource",65534], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 50867 - 1 - 67 - 0 - 1 after 10000ms) at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35) at listOnTimeout (node:internal/timers:559:17) at processTimers (node:internal/timers:502:7))

pgarnek, 2024-01-10:
does anyone have tested this fix with ME168 ?
can i give it a try ?

sychu, 2024-01-11:
@TrolliRolli Fixed converter sends 6 values instead 4 and it is matching only _TZE200_p3dbf6qs. Are You sure that You applied external converter for Yours_TZE200_6rdj8dzm?

sychu, 2024-01-11:
@pgarnek 
> does anyone have tested this fix with ME168 ? can i give it a try ?

Sure just tweak script to match Yours device, in script uncomment [line 69](https://github.com/sychu/avatto_me167_TZE200_p3dbf6qs/blob/8773ed9442614d02f4fc79be6112f86cc36e0993/avatto_me167.js#L69) and [line 79 ](https://github.com/sychu/avatto_me167_TZE200_p3dbf6qs/blob/8773ed9442614d02f4fc79be6112f86cc36e0993/avatto_me167.js#L79). Ensure that manufacturer match yours device `_TZE200_rxntag7i`.
Let us know if it worked :-)

sychu, 2024-01-11:
> @sychu could you create a PR to integrate this fix?

@Koenkk sure i can try, but i tested it only with TZE200_p3dbf6qs so it look like it should be separate definition for this device?

pgarnek, 2024-01-11:
i suspect, i should focus on the "_TZE200_p3dbf6qs", right ? :-)

sychu, 2024-01-11:
@pgarnek It is exactly fix for Yours device. Please read description https://github.com/sychu/avatto_me167_TZE200_p3dbf6qs?tab=readme-ov-file#avatto_me167_tze200_p3dbf6qs

>By default device _TZE200_p3dbf6qs is recognized by zigbee-herdsman-converters as [ME168](https://www.zigbee2mqtt.io/devices/ME168.html) but in fact it is [ME167](https://www.zigbee2mqtt.io/devices/ME167.html)

After applying this external converter device will be matched properly to ME167

pgarnek, 2024-01-11:
i've missed that, sorry.
Thanks, i'll try and let know if everything goes well

tdse13, 2024-01-28:
> As a workaround i created flow in NodeRED, that changes temperature values in heating mode. I smart-scheduler and function nodes, if someone is interested in, please comment, i'll share my code

Yes, I'm interested.

shirou93, 2024-02-08:
@sychu Can you prepare PR?

StefanLobbenmeier, 2024-04-18:
> I have play around with this and it look like that main issue is that converter sends 4 values instead 6 (just like @ecotwist mentioned). I created custom converter which sends 6 values and it seem [...]

The custom converter is no longer compatible with the latest zigbee2mqtt, so I will try to turn this into a PR now

StefanLobbenmeierObjego, 2024-04-18:
I created the PR now, but I need some input from the other people here - did anyone try this patch and had to to change the fingerprint? If so speak up now so we can change the PR to support the other fingerprints as well. I can only test my own one.

Vinceloud, 2024-06-15:
Hi, I have same issue, ME168 detected while I have ME167. I tried @sychu solution but submitting avatto-me168.js files I have an error : external converter must be array. Any idea please ? Thanks 🙏

MFlossmann, 2024-09-14:
@StefanLobbenmeier If you still need help (I'm guessing so, since the PR isn't merged yet - but it's also a few months old by now). However, I don't really know how to check out your PR on my setup and how to test it out. I use homeassistant with zigbee2mqtt.

richert-addisca, 2024-09-17:
Hello nice people, thank you so much for this script @sychu !

I was wondering if it is possible to disable scheduling altogether and only use automation to change the temperature. When disabling the schedule in MQTT Devices, it just hides them from the UI, but they get still executed.

StefanLobbenmeier, 2024-09-17:
Yeah I mostly need someone more knowledgeable to resolve the failing test there. But there is  another PR now so hopefully they will take care of it.

> disable scheduling altogether

scheduling is easy to disable, click the button to hide the „clock“ symbol in the device itself or in home assistant set the device from automatic to heating

richert-addisca, 2024-09-17:
@StefanLobbenmeier thanks for the advice, I can't seem to find the "clock" symbol, neither in Zigbee2MQTT -> Device nor in Settings -> Devices & Services -> MQTT Devices -> Device

I will try setting it to heating, I just expected the heater not to automatically shut off when the temperature is reached, if I manually set it to heating. But I will give it a try. Maybe the setting is just confusingly named.

ToMoHH, 2024-10-05:
Unfortunately, the solution with the converter no longer works with a current zigbee2mqtt version and the PR is open. How can I solve the problem at the moment? Unfortunately, I cannot use the thermostat like this.

StefanLobbenmeier, 2024-10-05:
@richert-addisca i meant on the physical device itself there is a clock led glowing when the schedule is activated

StefanLobbenmeier, 2024-10-05:
@ToMoHH For me the custom converter broke as well because of an import issue but I could resolve it a few months ago. Are you saying there is yet another issue / can you show the error message?

ToMoHH, 2024-10-06:
@StefanLobbenmeier 
I don't know why, but I reconnected the converter and the error message disappeared. I can operate the thermostat as intended. Sorry for the confusion.

akartla, 2024-11-05:
I there a solution for the auto mode problem on _TZE200_rxntag7i?
When i switch to auto mode the set point goes to 153°C.

SNX100, 2024-11-06:
Hello everyone. I'm currently facing the same issue with the ME167 thermostats. However, I'm not using HA but rather ioBroker. Unfortunately, I don't have the option to edit the configuration.yaml file, as it doesn't exist. In general, it seems to be set up a bit differently. I've copied the me167.js file to the directory /opt/iobroker/node_modules/zigbee-herdsman-converters/devices/ and changed the values in the object tree to

"type": "ME167",
"icon": "img/ME167.png"

and then restarted. However, this did not yield the desired result. On the contrary, after the restart, it reverted back to ME168 in the object tree. Could someone here possibly help me with this?

lexusburn, 2024-12-06:
Can anyone add the same path for _TZE200_9xfjixap please?

Schmallon, 2024-12-20:
I resolved this issue locally by applying the following patch, which leverages the ability of `thermostatScheduleDayMultiDPWithDayNumber` to specify the number of transitions, introduced in commit [9bf46b41ccc190860f5128f81a17264fac58d386](https://github.com/Koenkk/zigbee-herdsman-converters/commit/9bf46b41ccc190860f5128f81a17264fac58d386). Unfortunately, I don’t expect to have the time to create a PR for this in the near future, so please feel free to take this forward.

```
diff --git a/src/devices/tuya.ts b/src/devices/tuya.ts
index 69e991dd..faf5627d 100644
--- a/src/devices/tuya.ts
+++ b/src/devices/tuya.ts
@@ -4815,14 +4815,14 @@ const definitions: DefinitionWithExtend[] = [
                 [4, 'current_heating_setpoint', tuya.valueConverter.divideBy10],
                 [5, 'local_temperature', tuya.valueConverter.divideBy10],
                 [7, 'child_lock', tuya.valueConverter.lockUnlock],
-                [28, 'schedule_monday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(1)],
-                [29, 'schedule_tuesday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(2)],
-                [30, 'schedule_wednesday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(3)],
-                [31, 'schedule_thursday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(4)],
-                [32, 'schedule_friday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(5)],
-                [33, 'schedule_saturday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(6)],
-                [34, 'schedule_sunday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(7)],
-                [35, null, tuya.valueConverter.errorOrBatteryLow],
+                [28, 'schedule_monday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(1, 6)],
+                [29, 'schedule_tuesday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(2, 6)],
+                [30, 'schedule_wednesday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(3, 6)],
+                [31, 'schedule_thursday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(4, 6)],
+                [32, 'schedule_friday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(5, 6)],
+                [33, 'schedule_saturday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(6, 6)],
+                [34, 'schedule_sunday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(7, 6)],
+                [35, null, tuya.valueConverter.errorOrBatteryLow], //foo
                 [36, 'frost_protection', tuya.valueConverter.onOff],
                 [39, 'scale_protection', tuya.valueConverter.onOff],
                 [47, 'local_temperature_calibration', tuya.valueConverter.localTempCalibration2],
```

Schmallon, 2024-12-21:
Disregard my [earlier comment](https://github.com/Koenkk/zigbee2mqtt/issues/19787#issuecomment-2556808786). I just realized that [9bf46b41ccc190860f5128f81a17264fac58d386](https://github.com/Koenkk/zigbee-herdsman-converters/commit/9bf46b41ccc190860f5128f81a17264fac58d386) originates from @StefanLobbenmeier 's now closed [PR](https://github.com/Koenkk/zigbee-herdsman-converters/pull/7406) aimed at resolving this issue. So yes, Stefan's fix works for me as well and I'm using a _TZE200_9xfjixap.

tigattack, 2025-02-20:
Not stale.

tigattack, 2025-04-22:
Still an issue. Furthermore, the patches do not seem to work for my TRVs; _TZE200_p3dbf6qs (ME167).

StefanLobbenmeier, 2025-04-22:
@tigattack what do you mean the patches don’t work for you? If it doesn’t change anything maybe it’s not applied correctly, but if it breaks it for you you should raise more detail because the PR was based on the patch

closed 2025-06-29 by github-actions[bot]