Koenkk/zigbee2mqtt #19789: TO-Q-SY1-JZT is reported to support OTA but when I check for new firmware I got always an error

papperone, 2023-11-20:
### What happened?

I try to check if new firmware is available and only for these type of device I got below error:

info 2023-11-20 17:28:43MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/check', payload '{"data":{"id":"Piano Induzione"},"error":"Failed to check if update available for 'Piano Induzione' (No image available for imageType '5634')","status":"error","transaction":"dsqqv-13"}'
error 2023-11-20 17:28:43Failed to check if update available for 'Piano Induzione' (No image available for imageType '5634')

### What did you expect to happen?

no error and a confirmation if new firwmare is available or not

### How to reproduce it (minimal and precise)

pair one of these devices and try to check for OTA upgrades

### Zigbee2MQTT version

1.33.2

### Adapter firmware version

6.7.9.0 build 405

### Adapter

ZB-GW04 based on EFR32MG21 - EZSP v8

### Debug log

info 2023-11-20 17:28:43MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/check', payload '{"data":{"id":"Piano Induzione"},"error":"Failed to check if update available for 'Piano Induzione' (No image available for imageType '5634')","status":"error","transaction":"dsqqv-13"}'
error 2023-11-20 17:28:43Failed to check if update available for 'Piano Induzione' (No image available for imageType '5634')

Ra72xx, 2023-11-21:
Similar issue here: #19650

frayos, 2024-01-17:
The error I get is slightly different 
OTA: Images currently unavailable for device 'TS011F', hardwareVersion '1', manufacturerName _TZ3000_qeuvnohg, {"fieldControl":0,"manufacturerCode":4098,"imageType":5634,"fileVersion":65}'

papperone, 2024-07-18:
it seems I'm the only one with problem to make OTA works for this device?!??!?!

closed 2025-01-10 by github-actions[bot]

papperone, 2025-01-11:
ok thanks, as this is not planned I would suggest then to remove those device from the OTA page