Koenkk/zigbee2mqtt #19791: Bosch BTH-RM -> Setpoint is internally set by schedule

Boertie310, 2023-11-20:
### What happened?

It looks like there is a time schedule in the thermostat that is writing the setpoint.
This results in the behaviour that the setpoint is set randomly to e.g. 21°C.

The datapoint setpoint change source reports "schedule"

### What did you expect to happen?

I think there are missing some options to set the times and setpoints, and maybe a time synchronisation.
If there is no option to change it, I think there has to be some way to disable the internal schedule or set unocc and occ setpoint to the same setpoint or something like that.

This for example:
https://www.zigbee2mqtt.io/devices/TS0601_thermostat.html
{
    "workdays":[
        {"hour":6,"minute":0,"temperature":19},
        {"hour":8,"minute":0,"temperature":20},
        {"hour":18,"minute":0,"temperature":21},
        {"hour":20,"minute":30,"temperature":20},
        {"hour":22,"minute":0,"temperature":19},
        {"hour":23,"minute":30,"temperature":15}
    ]
}
{
    "holidays":[
        {"hour":6,"minute":0,"temperature":19},
        {"hour":8,"minute":0,"temperature":20},
        {"hour":18,"minute":0,"temperature":21},
        {"hour":20,"minute":30,"temperature":20},
        {"hour":22,"minute":0,"temperature":19},
        {"hour":23,"minute":30,"temperature":15}
    ]
}

### How to reproduce it (minimal and precise)

Change the mode from open window to automatic OR
Change the mode from off to automatic OR
Change the mode from heat to automatic OR
Change the setpoint and wait untill the moment the setpoint changes to 21°C by itself

### Zigbee2MQTT version

 1.33.2

### Adapter firmware version

20201026

### Adapter

zzh

burpy, 2023-11-20:
I have the same problem. It switches between 17 and 21 degree celcius if it is in the system mode "auto".

Boertie310, 2023-11-21:
> 

> I have the same problem. It switches between 17 and 21 degree celcius if it is in the system mode "auto".

I'm not going crazy then ;)

dierochade, 2023-12-16:
I do not think this is zigbee related, but part of the device firmware. Take a look at this discussion for reference:

https://community.home-assistant.io/t/bosch-radiator-thermostat-ii-does-anyone-use-these-without-hub-annoying-build-in-schedule-suggestions/622756/7

CoolID2, 2023-12-22:
I have the same problem. Can you fix it plse.

Boertie310, 2023-12-27:
> I do not think this is zigbee related, but part of the device firmware. Take a look at this discussion for reference:
> 
> https://community.home-assistant.io/t/bosch-radiator-thermostat-ii-does-anyone-use-these-without-hub-annoying-build-in-schedule-suggestions/622756/7

Well i think it is. There is a function (the schedule) in the thermostat that i think should be controlled from the bridge. That it is writing the setpoint is only good, and the use of the schedule inside of the thermostat would keep the screen off also. (Especially very useful at night for the WAF) 

The topic you described has the same problem, but does not excludes the option that it is zigbee related.

How would it work with a Bosch bridge then? At some moment the internal setpoints and times need to be written.

dierochade, 2024-01-02:
> Well i think it is. 

Yes, you are right from a technically point of view. It seems like a proprietary functionality without any standard though, I wonder if it can be done? Will probably need some/extensive reverse engineering...

I would love to see it implemented, thats for sure.

danielpetrovic, 2024-02-07:
I have implemented the heating state for BTH-RM so it can be used with the Heat mode now. HVAC_ACTION is properly propagated to Home Assistant now, so I have my Bosch Relay automated through HA that it turns on/off when the BTH-RM switches it's HVAC_ACTION to heating/idle.

I am leaving the thermostat on Heat all day and just change the set temperatures depending on time of day, outside temperature and if we are home or not.

The Auto setting is still available, but the schedule is hardcoded in the device. When you use the Bosch SHC II you can add extra schedules, but the default "optimized" cannot be removed from the settings.

Not sure if it is actually changing the schedule in the thermostat firmware or that the triggers are sent from the Bosch SHC II when you change the schedule from the default "optimized" .

Boertie310, 2024-02-16:
> I have implemented the heating state for BTH-RM so it can be used with the Heat mode now. HVAC_ACTION is properly propagated to Home Assistant now, so I have my Bosch Relay automated through HA that [...]

To bad it is hard coded, i'm on the point of just buying a Bosch bridge just to try it.

I'm using it roughly the same way right now, only without the heating state.
I'm using a PI controller on the temperature and setpoint that controls my eurotronic spirit valves on valve position.

closed 2024-09-14 by github-actions[bot]

wintersteiger, 2025-01-21:
Just FYI: the BTH-RM has a non-standard time cluster and the normal procedures for setting the time on the thermostat don't work. It will happily turn on the scheduler without knowing the time though. Perhaps, if we put the batteries in at exactly midnight...