Koenkk/zigbee2mqtt #19796: BTicino K4570CWI recognized as Legrand 064873 and impossible to OTA update

syntesys87, 2023-11-21:
### What happened?

Hello,

I have a couple of BTicino K4570CWI connected to my Z2M instance, but they are recognized as Legrand 064873 and it's impossible to start OTA update with the error _Update of 'Comando Terrazza' failed (Device didn't respond to OTA request)_.

Already tried to remove battery and reset without success.

### Zigbee2MQTT version

1.33.2-dev commit: 5889cdd

### Adapter firmware version

7.3.1.0 build 176

### Adapter

EZSP v12 SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2

### Debug log

Info 2023-11-21 08:30:55MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/update', payload '{"data":{"id":"Comando Terrazza"},"error":"Update of 'Comando Terrazza' failed (Device didn't respond to OTA request)","status":"error","transaction":"4d3o3-1"}'
Error 2023-11-21 08:30:55Update of 'Comando Terrazza' failed (Device didn't respond to OTA request)

syntesys87, 2023-12-21:
Same problem with 1.33.4.

ster1um, 2023-12-30:
confirm the problem, I change the battery and pair direct to coordinator, it's impossible to update.
`Update of 'Comando Entra E Esci' failed (Device didn't respond to OTA request)`

ster1um, 2024-01-11:
Update disappeared from z2m.

Bartok92, 2024-02-19:
ciao come sei riuscito a collegare l'entra e esci tramite zigbee ?

closed 2024-09-17 by github-actions[bot]

dinukro, 2025-02-14:
Hello, please let me know how I can associate K4570CWI to z2m.

Thank you

lucaam, 2026-01-09:
@syntesys87 any feedback on how you connected this device to z2m?