Koenkk/zigbee2mqtt #19797: IKEA TRADFRI lights turning on when sending `{"state": "OFF"}`

pbek, 2023-11-21:
### What happened?

Thanks a lot for this wonderful bridge!

I just found out that my IKEA TRADFRI lights are turning on when I send a `{"state": "OFF"}`. And they're still sending a state `OFF` when I do so.

I tried these two:

- https://www.zigbee2mqtt.io/devices/LED2201G8.html (at firmware 3.0.10)
- https://www.zigbee2mqtt.io/devices/LED2005R5.html (at firmware 3.0.10)

### What did you expect to happen?

I expect that the lights stay off when I send `{"state": "OFF"}`.

### How to reproduce it (minimal and precise)

- Send `{"state": "OFF"}` via MQTT (I used Node-RED) to a IKEA TRADFRI light while it is off.
- See that it turns on and not showing via state that it is on
- Send `{"state": "OFF"}` again
- See that it turns off
- Send `{"state": "OFF"}` again
- See that it turns on again

### Zigbee2MQTT version

1.33.2 commit: 9996c93 (docker)

### Adapter firmware version

20221226

### Adapter

SONOFF Zigbee 3.0 USB Dongle

### Debug log

```
info 2023-11-21 08:49:13MQTT publish: topic 'zigbee2mqtt/ikea-tradfri-light/05/dressing-room', payload '{"brightness":55,"color_mode":"color_temp","color_options":{"execute_if_off":false},"color_temp":403,"level_config":{"on_level":"previous"},"linkquality":72,"power_on_behavior":"off","state":"OFF","update":{"installed_version":50331664,"latest_version":50331664,"state":"idle"}}'
info 2023-11-21 08:49:14MQTT publish: topic 'zigbee2mqtt/ikea-tradfri-light/05/dressing-room', payload '{"brightness":55,"color_mode":"color_temp","color_options":{"execute_if_off":false},"color_temp":403,"level_config":{"on_level":"previous"},"linkquality":72,"power_on_behavior":"off","state":"OFF","update":{"installed_version":50331664,"latest_version":50331664,"state":"idle"}}'
info 2023-11-21 08:49:16MQTT publish: topic 'zigbee2mqtt/ikea-tradfri-light/05/dressing-room', payload '{"brightness":55,"color_mode":"color_temp","color_options":{"execute_if_off":false},"color_temp":403,"level_config":{"on_level":"previous"},"linkquality":72,"power_on_behavior":"off","state":"OFF","update":{"installed_version":50331664,"latest_version":50331664,"state":"idle"}}'
```

pbek, 2023-11-21:
After finding a bulb where I couldn't reproduce this issue, I removed the two devices where I had the issue from zigbee2mqtt and joined them again. Afterward, I didn't have the issue anymore!

Should I close this issue, or is this something that still needs to be addressed?

happosade, 2024-02-16:
I'm having the same issue. I have `LED2201G8` with firmware 3.0.10

I could try to get more logs if that's helpful, but if not, I'll try to remove and add bulbs again to see if that fixes the issue. I don't remember having the issue with previous firmware and I did upgrade the firmware with z2m, if that helps at all to figure out what might cause this bug.

whcrg, 2024-04-14:
I seem to have same problem with IKEA LED2201G8.
Not turning fully off, transition involved.
Everything updated to newest stable (home assistant, z2m) and bulbs reset and rejoined etc.

Need to try what happens without transition.

pbek, 2024-04-14:
Maybe just remove the device and reconnect it, like I did.

whcrg, 2024-04-14:
As clearly stated above, reset and rejoin not helping.

When transition is 2 seconds or less the bulbs seem to go off normally, with 3 seconds there is random behavior between going off or remaining on, with 4 seconds or more bulbs consistently remain on low brightness. 

Exactly same behavior for both of this model bulbs I have. Firmware 3.0.10 and did same on previous out-of-the-box, whatever number it was.

Somewhere is a bug (firmware or z2m? Workaround possible if in firmware?)

pbek, 2024-04-14:
> As clearly stated above, reset and rejoin not helping.

Ah, sorry. I wasn't sure what you meant by "reset", because I only have a "reconfigure" button, and that didn't help. I removed the device completely and started the pairing process again via turning on/off the bulb 5 times. Furthermore, I then had to set the old name manually again too...

djurny, 2024-10-12:
Hi there,
I had something similar with IKEA [LED1924G9](https://www.zigbee2mqtt.io/devices/LED1924G9.html#ikea-led1924g9) bulbs. Turning the bulb off twice - sending an "OFF" to a bulb that's already off, would turn the bulb back ON.
Have a look at these issues that show similar effects:
- https://github.com/Koenkk/zigbee2mqtt/issues/23406
- https://github.com/Koenkk/zigbee2mqtt/issues/22030

To work around this, I check if the bulbs are OFF before turning the off in HASS. (I use scripts for all light controls.)

What also caused some issues in my setup, is mixing HASS `light` groups with Zigbee groups, in particular addressing lights individually in `light` groups while they were still part of a Zigbee group. Reason for ending up in that situation is that I did not properly dissolve the Zigbee groups before adding the individual bulbs to a `light` group.
End result there was lights unexpectedly turning on/off due to `light` group behaviour turning entire group on if one light is ON. (Little more to it than that, but that's more or less the gist of it.)

Groetjes,

closed 2025-01-10 by github-actions[bot]