Koenkk/zigbee2mqtt #19799: Vimar 14592.0 not report status to Home Assistant

ontirius, 2023-11-21:
### What happened?

Hi, I have 5 vimar 14592.0 integrated in HA with conbee II and zigbee2mqtt, all working perfectly until I upgraded zigbee2mqtt to the latest release 1.33. 2, now on the interface of zigbee2mqtt, the 5 vimar switches work, I see their status and can change it without any problem, only on HA I don't see their status and I can't even change it (I attach two screenshot, one for zigbee2mqtt interface in which you can see the correct status, and the presence of button switch to change the status, and one for the home assisntance intervace, in which you are able to see that is showed as "disconnect" an no actions available to change the status), they appear as devices, if I change the name on zigbee2mqtt the name also updates on HA, but there is no way to change\see the status.
I have already tried removing the MQTT integration on HA and reinstalling it, unfortunately without success.
Can you please verify this problem?
Thank you very much!
Note: all other zigbee devices (occupancy, temp, luminosity sensors) work without problems, I see them correctly on HA
[image]
[image]

### Zigbee2MQTT version

1.33.2

### Adapter firmware version

0x26580700

### Adapter

ConBee2/RaspBee2

ontirius, 2023-11-28:
Hi, any news about that?

ontirius, 2023-12-01:
Hi all, are there some news?

Koenkk, 2023-12-02:
Could you check if the issue is fixed with the following external converter: https://gist.github.com/Koenkk/49608776b57009e68d2e350b21f40c45

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

ontirius, 2023-12-02:
Hi Koenkk, many thanks for your support.
I did it, create a new file ext_converter.js inside folder of zigbee2mqtt, in that folder also exists the configuration.yaml, and I added to configuration.yaml the two rows:

external_converters:
   - ext_converter.js

Note: I added this two rows below section experimental: {} and before the section frontend:

I restarted zigbee2mqtt with command "docker-compose restart zigbee2mqtt"

I waited a few minutes... But the situation is still the same. On zigbee2mqtt interface all zigbee devices are see properly, and I can change their status, also vimar devices, but in HA I am able to see only this:

[image]

Just to translate from Italy: "has become unknown" means "has become unknown", and as you can see no actions are available.

Maybe I wrong to configure the configuration.yaml? I need to put the two rows in a proper position in configuration.yaml?
Many thanks for your patience!
Best regards
Giuseppe

Koenkk, 2023-12-07:
Can you try with z2m 1.34.0 and the latest HA?

ontirius, 2023-12-07:
Hi Koenkk,
many thanks for your support! Fixed with upgrade to HA 2023.9.2!
Best regards
Giuseppe

closed 2023-12-07 by ontirius