Koenkk/zigbee2mqtt #19803: [New device support]: for presence detector TS0225 / manufacter _TZE200_2aaelwxk

patrick-foubert-56770, 2023-11-21:
### Link

https://fr.aliexpress.com/item/1005005780858603.html?spm=a2g0o.order_list.order_list_main.28.631e5e5bE0beJX&gatewayAdapt=glo2fra

### Database entry

{"id":77,"type":"Router","ieeeAddr":"0xa4c13860917e1353","nwkAddr":31107,"manufId":4417,"manufName":"_TZE200_2aaelwxk","powerSource":"DC Source","modelId":"TS0225","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,3,1280,57346,61184,60928,57344,1024],"outClusterList":[3,57346,61184,60928,57344],"clusters":{"genBasic":{"attributes":{}},"ssIasZone":{"attributes":{"iasCieAddr":"0x00124b00257ccbe9","zoneState":0}},"msIlluminanceMeasurement":{"attributes":{"measuredValue":33729}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":130,"stackVersion":2,"hwVersion":1,"swBuildId":"0122052017","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1700571523713,"defaultSendRequestWhen":"immediate"}

### Comments

 need custom script i can't update z2m version (issue with switch)

5.8G Zigbee Presence small version (not square=

pejotigrek, 2023-12-30:
+1 for this.
I've got the same sensor (_TZE200_2aaelwxk) and I think it's wrongly recognized as [ZG-205Z/A](https://www.zigbee2mqtt.io/devices/ZG-205Z_A.html#tuya-zg-205z%252Fa) - it's possible that those sensors are very similar, but mine seems like is going to sleep mode VERY OFTEN and VERY QUICK - the connection drops which renders it almost unusable, as it stops detecting people nearby until I press the button on sensor, or change some settings to "refresh" the connection. I'm thinking if that wrong recognition is causing this issue as other devices in the network work fine. I've got quite a few routers nearby so I doubt it's a link quality problem or something network-based.

kkondyl, 2024-01-26:
+1
I have the same sensor and it is being recognized as ZG-205Z/A.
It is working somehow but it looks to be too sensitive (I lowered the sensitivity to 0).
Main issue is that the reported Illuminance is 100x higher.
As a workaround I have configured the setting "Illuminance calibration"=-99, which I understand that subtracts 99% of original value.
Moreover, for some reason the illuminance was disabled in mqtt. I needed to enable it manually...

pejotigrek, 2024-07-25:
bad bot

Koenkk, 2024-07-26:
Could you try to, stop z2m, in the `data/database.db` change `_TZE200_2aaelwxk` to `_TZE200_hl0ss9oa`, start z2m and see if it works? If it doesn't, you can also try `_TZE200_2aaelwxk`

pejotigrek, 2024-07-27:
> @pejotigrek I'm not sure if I can fix yours from the z2m side

no problem, I lowered the priority of that sensor in my automations, wasn't too reliable unfortunately ;(
who knows, maybe some update will fix it ;)

kkondyl, 2024-07-28:
> @kkondyl I've fixed the illuminance issue, @pejotigrek I'm not sure if I can fix yours from the z2m side, @kkondyl for the rest everything works?
> 
> Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

I am using three sensors for 6 months. Apart from illuminance, no other issue. Thanks for the fix!

maiersoft, 2024-08-02:
My [Tuya ZG-205Z/A] now reports 20lx where it used to report 2000, which 2000 seems correct.

poostrom, 2024-08-02:
Exact the same here with Tuya ZG-205Z/A, when putting a bright light on the sensor, it goes to 32 lux, where it should be 100 times higher.

kkondyl, 2024-08-02:
> I am using three sensors for 6 months. Apart from illuminance, no other issue. Thanks for the fix!

After upgrade to 1.39.1, all my _TZE200_2aaelwxk sensors report correct illuminance, confirmed with external lux meter! Fix works fine!

Koenkk, 2024-08-03:
> After upgrade to 1.39.1, all my _TZE200_2aaelwxk sensors report correct illuminance, confirmed with external lux meter! Fix works fine!

Amazing, then I assume this can be closed now.

closed 2024-08-03 by Koenkk

Koenkk, 2024-08-12:
Unfortunately had to revert this, see https://github.com/Koenkk/zigbee2mqtt/issues/23496#issuecomment-2283803423