Koenkk/zigbee2mqtt #19804: [New device support]: OWON THS317-ET, model: TS0201, manufacturer: _TZE200_01fvxamo

soukal24, 2023-11-21:
### Link

https://www.pamitech.cz/kabelovy-zigbee-sensor-teploty-tuya/

### Database entry

{"id":2,"type":"EndDevice","ieeeAddr":"0xb43a31fffe66d186","nwkAddr":34055,"manufId":4098,"manufName":"_TZE200_01fvxamo","powerSource":"Battery","modelId":"TS0201","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,1,61184,1026,1029,65382],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65534":0,"modelId":"TS0201","manufacturerName":"_TZE200_01fvxamo","powerSource":3,"zclVersion":3,"appVersion":66,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":66,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-1440156621},"lastSeen":1700575181645,"defaultSendRequestWhen":"immediate"}

### Comments

It is a new device similar to THS317-ET but it is based on ZS3L to be supported by Tuya Zigbee gateway. Pairing was successful in Z2M, it looks like TS0201 but does not display any values. It works properly with Tuya Zigbee gateway.
[image]
[image]
[image]

saba01, 2024-01-09:
I have the same problem,
tested with last dev release, 1.35.1-dev commit: ce7d022

soukal24, 2024-03-01:
Here is log from Z2M afr´ter pair device:
`2024-03-01 12:47:49Device '0x3c6a2cfffed3e89a' joined
Info 2024-03-01 12:47:49MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x3c6a2cfffed3e89a","ieee_address":"0x3c6a2cfffed3e89a"},"type":"device_joined"}'
Info 2024-03-01 12:47:49Starting interview of '0x3c6a2cfffed3e89a'
Info 2024-03-01 12:47:49MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x3c6a2cfffed3e89a","ieee_address":"0x3c6a2cfffed3e89a","status":"started"},"type":"device_interview"}'
Debug 2024-03-01 12:47:49Device '0x3c6a2cfffed3e89a' announced itself
Info 2024-03-01 12:47:49MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x3c6a2cfffed3e89a","ieee_address":"0x3c6a2cfffed3e89a"},"type":"device_announce"}'
Debug 2024-03-01 12:47:49Received Zigbee message from '0x3c6a2cfffed3e89a', type 'readResponse', cluster 'genBasic', data '{"manufacturerName":"_TZE200_01fvxamo","modelId":"TS0201"}' from endpoint 1 with groupID 0
Debug 2024-03-01 12:47:49Skipping message, still interviewing
Debug 2024-03-01 12:47:50Received Zigbee message from '0x3c6a2cfffed3e89a', type 'readResponse', cluster 'genBasic', data '{"powerSource":3}' from endpoint 1 with groupID 0
Debug 2024-03-01 12:47:50Skipping message, still interviewing
Debug 2024-03-01 12:47:51Received Zigbee message from '0x3c6a2cfffed3e89a', type 'readResponse', cluster 'genBasic', data '{"zclVersion":3}' from endpoint 1 with groupID 0
Debug 2024-03-01 12:47:51Skipping message, still interviewing
Debug 2024-03-01 12:47:51Received Zigbee message from '0x3c6a2cfffed3e89a', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,209],"type":"Buffer"},"datatype":2,"dp":1}],"seq":22784}' from endpoint 1 with groupID 0
Debug 2024-03-01 12:47:51Skipping message, still interviewing
Debug 2024-03-01 12:47:51Received Zigbee message from '0x3c6a2cfffed3e89a', type 'readResponse', cluster 'genBasic', data '{"appVersion":65}' from endpoint 1 with groupID 0
Debug 2024-03-01 12:47:51Skipping message, still interviewing
Debug 2024-03-01 12:47:51Received Zigbee message from '0x3c6a2cfffed3e89a', type 'readResponse', cluster 'genBasic', data '{"stackVersion":0}' from endpoint 1 with groupID 0
Debug 2024-03-01 12:47:51Skipping message, still interviewing
Debug 2024-03-01 12:47:51Received Zigbee message from '0x3c6a2cfffed3e89a', type 'readResponse', cluster 'genBasic', data '{"hwVersion":1}' from endpoint 1 with groupID 0
Debug 2024-03-01 12:47:51Skipping message, still interviewing
Debug 2024-03-01 12:47:52Received Zigbee message from '0x3c6a2cfffed3e89a', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,70],"type":"Buffer"},"datatype":2,"dp":4}],"seq":23040}' from endpoint 1 with groupID 0
Debug 2024-03-01 12:47:52Skipping message, still interviewing
Debug 2024-03-01 12:47:52Received Zigbee message from '0x3c6a2cfffed3e89a', type 'readResponse', cluster 'genBasic', data '{"dateCode":""}' from endpoint 1 with groupID 0
Debug 2024-03-01 12:47:52Skipping message, still interviewing
Debug 2024-03-01 12:47:52Received Zigbee message from '0x3c6a2cfffed3e89a', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 1 with groupID 0
Debug 2024-03-01 12:47:52Skipping message, still interviewing
Info 2024-03-01 12:47:52Successfully interviewed '0x3c6a2cfffed3e89a', device has successfully been paired
Info 2024-03-01 12:47:52Device '0x3c6a2cfffed3e89a' is supported, identified as: TuYa Temperature & humidity sensor with display (TS0201)
Info 2024-03-01 12:47:52MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Temperature & humidity sensor with display","exposes":[{"access":1,"category":"diagnostic","description":"Remaining battery in %, can take up to 24 hours before reported","label":"Battery","name":"battery","property":"battery","type":"numeric","unit":"%","value_max":100,"value_min":0},{"access":1,"description":"Measured temperature value","label":"Temperature","name":"temperature","property":"temperature","type":"numeric","unit":"°C"},{"access":1,"description":"Measured relative humidity","label":"Humidity","name":"humidity","property":"humidity","type":"numeric","unit":"%"},{"access":1,"category":"diagnostic","description":"Voltage of the battery in millivolts","label":"Voltage","name":"voltage","property":"voltage","type":"numeric","unit":"mV"},{"access":1,"category":"diagnostic","description":"Link quality (signal strength)","label":"Linkquality","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"TS0201","options":[{"access":2,"description":"Calibrates the temperature value (absolute offset), takes into effect on next report of device.","label":"Temperature calibration","name":"temperature_calibration","property":"temperature_calibration","type":"numeric"},{"access":2,"description":"Number of digits after decimal point for temperature, takes into effect on next report of device. This option can only decrease the precision, not increase it.","label":"Temperature precision","name":"temperature_precision","property":"temperature_precision","type":"numeric","value_max":3,"value_min":0},{"access":2,"description":"Calibrates the humidity value (absolute offset), takes into effect on next report of device.","label":"Humidity calibration","name":"humidity_calibration","prope [... cut ...] ,"seq":25856}'
Debug 2024-03-01 12:54:55Received Zigbee message from '0x3c6a2cfffed3e89a', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,70],"type":"Buffer"},"datatype":2,"dp":4}],"seq":26112}' from endpoint 1 with groupID 0
Debug 2024-03-01 12:54:55No converter available for 'TS0201' with cluster 'manuSpecificTuya' and type 'commandDataReport' and data '{"dpValues":[{"data":{"data":[0,0,0,70],"type":"Buffer"},"datatype":2,"dp":4}],"seq":26112}'`

awhitwam, 2024-05-01:
Same issue here.  Any luck with a solution ?

awhitwam, 2024-05-02:
ok, i got it working without entirely knowing what i was doing;

Here's the code;

```javascript

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
	fingerprint: tuya.fingerprint('TS0201', ['_TZE200_01fvxamo']),
    model: 'THS317-ET-EY',
    vendor: 'Tuya',
    description: 'Temperature Sensor with Probe THS 317-ET',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.fz.datapoints],
    configure: tuya.configureMagicPacket,
    exposes: [e.temperature(), e.battery()],
    meta: {
        tuyaDatapoints: [
            [1, 'temperature', tuya.valueConverter.divideBy10],
            [4, 'battery', tuya.valueConverter.raw],

        ],
    },

};

module.exports = definition;

```

Koenkk, 2024-05-02:
@awhitwam do you see logging for any other DPs, like for humidity?

awhitwam, 2024-05-02:
> @awhitwam do you see logging for any other DPs, like for humidity?

No, nothing at all in the logs.  I'm pretty sure it only supports temperature and battery.

see [https://zigbee.blakadder.com/OWON_THS317-ET.html](url)

closed 2024-05-02 by Koenkk

Hoween, 2024-08-20:
Hi @soukal24. Looking that we have same issue with same sensor from same e-shop :) Please, can you confirm that THS 317-ET-YT is fully supported in Z2M? According to this closed issue it should be - I have latest stable Z2M, but this sensors is still identified as [TS0201](https://www.zigbee2mqtt.io/devices/TS0201.html#tuya-ts0201). Or do you have any clue how to fix this behaviour?

soukal24, 2024-08-20:
Hi, here is the newly tested and working converter. Is it possible to add it to the next version, please? @Koenkk

```javascript
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
	fingerprint: tuya.fingerprint('TS0201', ['_TZE200_iq4ygaai']),
    model: 'THS317-ET-TY',
    vendor: 'Tuya',
    description: 'Temperature Sensor with Probe THS 317-ET-TY',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.fz.datapoints],
    configure: tuya.configureMagicPacket,
    exposes: [e.temperature(), e.battery()],
    meta: {
        tuyaDatapoints: [
            [1, 'temperature', tuya.valueConverter.divideBy10],
            [4, 'battery', tuya.valueConverter.raw],

        ],
    },

};

module.exports = definition;
```

Koenkk, 2024-08-21:
Done!