Koenkk/zigbee2mqtt #19813: Schneider Electric WV704R0A0902 - Running_state is not working

juliennnnn, 2023-11-22:
### What happened?

Hello
I bought 3 Schneider Electric WV704R0A0902 valves (https://www.zigbee2mqtt.io/devices/WV704R0A0902.html) but there is no valid value for "running_state". Current value is "no value". 

### What did you expect to happen?

As the documentation shows, I excepted to have the values "idle" when not running, and "heat" when running.

### Zigbee2MQTT version

1.33.2

### Adapter firmware version

0x26580700

### Adapter

ConBee2

closed 2024-06-20 by github-actions[bot]

juliennnnn, 2024-11-16:
Hello, any update on this please? Still the same issue, even with the last updates. Thanks a lot :)

pbranly, 2025-02-18:
Hi I would also deeply need that feature for controlling my central heating system
Please push it
Phil