Koenkk/zigbee2mqtt #19816: [New device support]: TS0601 Boiler Switch with timers 15,30,60,90 min

moshiko2312, 2023-11-22:
### Link

https://www.smarthome4u.co.il/products.html

### Database entry

{"id":2,"type":"Router","ieeeAddr":"0xa4c138192510945e","nwkAddr":60927,"manufId":4417,"manufName":"_TZE204_vqaajp1a","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"stackVersion":0,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1700660489583,"defaultSendRequestWhen":"immediate"}

### Comments

Circuit breaker for electric boiler with timer 15,30,60,90 minutes
[image]
[image]

pantherale0, 2023-11-22:
Not able to test as its not a device I have, but see if this converter works, I found the datapoints over on the smartthings forum.

https://gist.github.com/pantherale0/cca06b9669c07e68addd22d74f0e555d

Sources:
https://docs.google.com/spreadsheets/d/12AsWMyQfc8MoNTGed6p7BPIvi3H2pkar/edit#gid=725399417
https://community.smartthings.com/t/st-edge-personal-tuya-devices-generic-ef00-device/255270/330?page=17

moshiko2312, 2023-11-23:
Hi thanks 
Its not working good on off not working , what I need to do?
￼

closed 2023-11-25 by moshiko2312