Koenkk/zigbee2mqtt #19817: Zemismart ZM25RX-08/30 (TS0601/_TZE200_bv1jcqqu): Battery drain/Log messages flood

donbobka, 2023-11-22:
### What happened?

From here https://github.com/Koenkk/zigbee2mqtt/issues/18685#issuecomment-1820156784
> I too have these roller shades and I'm experiencing pretty fast battery drain on them once paired to Z2M. Looking in my logs, it seems the shades are frequently messaging and (maybe?) not receiving a response due to the missing converter. I wonder if the shades are expecting a response. Roughly every minute, the lights on the roller shade light up without any commands issued.

```
Received Zigbee
message from 'Bedroom Shades', type
'commandMcuVersionResponse', cluster
'manuSpecificTuya', data
'{"seq" :256, "version":65}' from endpoint 1 with groupID 0
Debug
2023-11-20 22:05:26 No converter available
for 'ZM25RX-08/30' with cluster
'manuSpecificTuya' and type
'commandMcuVersionResponse' and data
'{"seq" :256, "version": 65}'

```

https://github.com/Koenkk/zigbee2mqtt/assets/589057/829bb475-6206-4a9d-a71f-7990708ad0eb

### What did you expect to happen?

mcuVersionResponse sent only once

### How to reproduce it (minimal and precise)

- send message `mcuVersionRequest`

cluster: `0xEF00`
command: `0x10`
payload: `{"seq":2}`

### Zigbee2MQTT version

1.33.2

### Adapter firmware version

0x26720700

### Adapter

ConBee2

### Debug log

```
Received Zigbee message from 'office-blinds', type 'commandMcuVersionResponse', cluster 'manuSpecificTuya', data '{"seq":256,"version":65}' from endpoint 1 with groupID null
debug 2023-11-23 00:38:10No converter available for 'ZM25RX-08/30' with cluster 'manuSpecificTuya' and type 'commandMcuVersionResponse' and data '{"seq":256,"version":65}'
info 2023-11-23 00:38:10MQTT publish: topic 'zigbee2mqtt/office-blinds', payload '{"battery":null,"click_control":"lower","color_mode":"xy","last_seen":"2023-11-22T13:38:11.622Z","linkquality":255,"motor_direction":"normal","position":0,"program":null,"reverse_direction":"false","state":"STOP","work_state":"closing"}'
info 2023-11-23 00:38:11MQTT publish: topic 'zigbee2mqtt/office-blinds', payload '{"battery":null,"click_control":"lower","color_mode":"xy","last_seen":"2023-11-22T13:38:11.849Z","linkquality":255,"motor_direction":"normal","position":0,"program":null,"reverse_direction":"false","state":"STOP","work_state":"closing"}'
debug 2023-11-23 00:38:11Received Zigbee message from 'office-blinds', type 'commandMcuVersionResponse', cluster 'manuSpecificTuya', data '{"seq":256,"version":65}' from endpoint 1 with groupID null
debug 2023-11-23 00:38:11No converter available for 'ZM25RX-08/30' with cluster 'manuSpecificTuya' and type 'commandMcuVersionResponse' and data '{"seq":256,"version":65}'
info 2023-11-23 00:38:11MQTT publish: topic 'zigbee2mqtt/office-blinds', payload '{"battery":null,"click_control":"lower","color_mode":"xy","last_seen":"2023-11-22T13:38:11.956Z","linkquality":255,"motor_direction":"normal","position":0,"program":null,"reverse_direction":"false","state":"STOP","work_state":"closing"}'
info 2023-11-23 00:38:11MQTT publish: topic 'zigbee2mqtt/office-blinds', payload '{"battery":null,"click_control":"lower","color_mode":"xy","last_seen":"2023-11-22T13:38:12.174Z","linkquality":255,"motor_direction":"normal","position":0,"program":null,"reverse_direction":"false","state":"STOP","work_state":"closing"}'
debug 2023-11-23 00:38:11Received Zigbee message from 'office-blinds', type 'commandMcuVersionResponse', cluster 'manuSpecificTuya', data '{"seq":256,"version":65}' from endpoint 1 with groupID null
debug 2023-11-23 00:38:11No converter available for 'ZM25RX-08/30' with cluster 'manuSpecificTuya' and type 'commandMcuVersionResponse' and data '{"seq":256,"version":65}'
```

Koenkk, 2023-11-22:
Can you provide the herdsman debug log of this? I don't know why the device keeps sending this command, I would expect that z2m sends back the `commandMcuVersionRequest` which should make the device stop requesting it.

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

donbobka, 2023-11-23:
Log attached [herdsman.log](https://github.com/Koenkk/zigbee2mqtt/files/13445882/herdsman.log).

My understanding of current code/problem is that [mcuVersionResponse from device](https://github.com/Koenkk/zigbee-herdsman-converters/blob/24baf6800c109ad891899e580659d800ab4918ad/src/lib/tuya.ts#L45) triggers [mcuVersionRequest](https://github.com/Koenkk/zigbee-herdsman-converters/blob/24baf6800c109ad891899e580659d800ab4918ad/src/lib/tuya.ts#L46) and then it's in loop:

0. Something (I'm not entirely sure what triggers it initially, but i assume it's z2m; i simulate it manually with z2m dev console) triggers initial mcuVersionRequest to device
1. Device receives mcuVersionRequest and respond with mcuVersionResponse back
2. Z2M receives mcuVersionResponse, calls onEvent that again triggers mcuVersionRequest
3. Go to 1

Koenkk, 2023-11-23:
Maybe it's because we are replying with the wrong sequence number, can you try changing:

```js
await endpoint.command('manuSpecificTuya', 'mcuVersionRequest', {'seq': 0x0002});
```

to 

```js
await endpoint.command('manuSpecificTuya', 'mcuVersionRequest', {'seq': data.data.seq});
```

donbobka, 2023-11-23:
Same thing (also tested `{'seq': data.data.seq + 1}`).

> I don't know why the device keeps sending this command, I would expect that z2m sends back the commandMcuVersionRequest which should make the device stop requesting it.

I think it's an opposite: z2m sending mcuVersionRequest triggers device to send response back

from [here](https://github.com/Koenkk/zigbee2mqtt/issues/18685#issuecomment-1821564210)
> @donbobka the naming of the commands is wrong I think, but the temperature sensor that was added here required this (and I think it's according to the TuYa protocol).

Is there documentation/explanation of requirement for temperature sensor? Is it just needed to send `mcuVersionRequest` on `deviceAnnounce`?

Koenkk, 2023-11-24:
> Is it just needed to send mcuVersionRequest on deviceAnnounce?

It doesn't send a mcuVersionRequest on deviceAnnounce (it's sends a `dataQuery`). For the temperature sensor I'm sure the `mcuVersionRequest`/`mcuVersionResponse` is correct. The sensor sends the `mcuVersionResponse`, z2m replies with the `mcuVersionRequest` after which the device doesn't do the `mcuVersionResponse` anymore. I've confirmed this behaviour myself through sniffing.

I think `onEvent` should be extended with a parameter to allow disabling responding to `mcuVersionResponse` for this device, could you make a PR for that?

SolidElectronics, 2023-12-21:
@RikJ01 Can you try reverting zigbee2mqtt to 1.33.1 and see if that stops the message flood?
I'm seeing a similar behaviour on anything past that, and I suspect the cause might be related to this update: https://github.com/Koenkk/zigbee-herdsman-converters/pull/6341
```
Info 2023-12-20 22:14:08MQTT publish: topic 'zigbee2mqtt/Mollys room blind', payload '{"battery":null,"last_seen":"2023-12-20T22:14:08-05:00","linkquality":40,"motor_direction":"normal","position":100,"running":false,"state":"OPEN","work_state":"opening"}'
Info 2023-12-20 22:14:11MQTT publish: topic 'zigbee2mqtt/Mollys room blind', payload '{"battery":null,"last_seen":"2023-12-20T22:14:11-05:00","linkquality":40,"motor_direction":"normal","position":100,"running":false,"state":"OPEN","work_state":"opening"}'
Info 2023-12-20 22:14:13MQTT publish: topic 'zigbee2mqtt/Mollys room blind', payload '{"battery":null,"last_seen":"2023-12-20T22:14:13-05:00","linkquality":40,"motor_direction":"normal","position":100,"running":false,"state":"OPEN","work_state":"opening"}'
Info 2023-12-20 22:14:13MQTT publish: topic 'zigbee2mqtt/Mollys room blind', payload '{"battery":null,"last_seen":"2023-12-20T22:14:13-05:00","linkquality":32,"motor_direction":"normal","position":100,"running":false,"state":"OPEN","work_state":"opening"}'
Info 2023-12-20 22:14:14MQTT publish: topic 'zigbee2mqtt/Mollys room blind', payload '{"battery":null,"last_seen":"2023-12-20T22:14:14-05:00","linkquality":32,"motor_direction":"normal","position":100,"running":false,"state":"OPEN","work_state":"opening"}'

```

donbobka, 2023-12-21:
@Koenkk I think for that particular problem we can fix it easier: just delete onEvent callback (https://github.com/Koenkk/zigbee-herdsman-converters/pull/6748). I copied it from another blinds configuration and blinds works perfectly without that.

Koenkk, 2023-12-21:
Thanks @donbobka, merged https://github.com/Koenkk/zigbee-herdsman-converters/pull/6748

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-07-19 by github-actions[bot]