Koenkk/zigbee2mqtt #19820: [Wrong device]: EARU VOLTAGE METER  WRONG RECOGNIZED

basalisco, 2023-11-22:
### Link

https://ae01.alicdn.com/kf/S1ec89b600c7045fb953929e00a0fef99K/1P-2P-TUYA-APP-Zigbee-WIFI-consumo-energetico-intelligente-kWh-Meter-interruttore-del-circuito-interruttore-del.jpg_.webp

### Model

EAYCBM

### Description

voltage meter

### Vendor

EARU

### Picture (link)

https://ae01.alicdn.com/kf/S1ec89b600c7045fb953929e00a0fef99K/1P-2P-TUYA-APP-Zigbee-WIFI-consumo-energetico-intelligente-kWh-Meter-interruttore-del-circuito-interruttore-del.jpg

### Database entry

{"id":2,"type":"EndDevice","ieeeAddr":"0xa4c138703eab6435","nwkAddr":38062,"manufId":4417,"manufName":"_TZ3000_xr3htd96","powerSource":"Battery","modelId":"TS0201","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":770,"inClusterList":[1,3,1026,1029,0],"outClusterList":[3,25,10],"clusters":{"genBasic":{"attributes":{"65503":"#\u0017\u000f,\u0011@M\u0016,\u0011Y� ,\u0012�� ,\u0011J� ,\u0012","65506":51,"65508":0,"65534":0,"modelId":"TS0201","manufacturerName":"_TZ3000_xr3htd96","powerSource":3,"zclVersion":3,"appVersion":70,"stackVersion":0,"hwVersion":1,"dateCode":""}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":24,"batteryVoltage":25}},"msTemperatureMeasurement":{"attributes":{"measuredValue":2342}},"msRelativeHumidity":{"attributes":{"measuredValue":5373}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":70,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-708457359},"lastSeen":1687031923076,"defaultSendRequestWhen":"immediate"}

### Notes

this device is recognized as [TS011F_plug_1](https://www.zigbee2mqtt.io/devices/TS011F_plug_1.html#tuya-ts011f_plug_1)

basalisco, 2023-12-01:
> ### Link
> https://it.aliexpress.com/item/1005005999108883.html?spm=a2g0o.order_list.order_list_main.20.6cde3696FUAHuy&gatewayAdapt=glo2ita
> 
> ### Model
> _TZ3000_zrm3oxsh
> 
> ### Description
> voltage meter
> 
> ### Vendor
> EARU
> 
> ### Picture (link)
> https://ae01.alicdn.com/kf/S1ec89b600c7045fb953929e00a0fef99K/1P-2P-TUYA-APP-Zigbee-WIFI-consumo-energetico-intelligente-kWh-Meter-interruttore-del-circuito-interruttore-del.jpg
> 
> ### Database entry
> {"id":12,"type":"Router","ieeeAddr":"0xe0798dfffedde7cd","nwkAddr":7423,"manufId":4098,"manufName":"_TZ3000_zrm3oxsh","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":266,"inClusterList":[0,3,4,5,6,1794,2820,1026,57344,57345],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"�\u001c�,\u0012","65506":31,"65508":0,"65534":0,"modelId":"TS011F","manufacturerName":"_TZ3000_zrm3oxsh","stackVersion":0,"dateCode":"","zclVersion":3,"appVersion":67,"powerSource":1}},"haElectricalMeasurement":{"attributes":{"rmsVoltage":233,"rmsCurrent":2876,"activePower":563,"acCurrentDivisor":1000,"acCurrentMultiplier":1}},"seMetering":{"attributes":{"currentSummDelivered":[0,8315],"divisor":100,"multiplier":1}},"manuSpecificBosch":{"attributes":{"53251":"AAAA"}},"msTemperatureMeasurement":{"attributes":{"measuredValue":2702}},"genOnOff":{"attributes":{"32768":0,"onOff":1,"onTime":0,"offWaitTime":0,"tuyaBacklightMode":0,"moesStartUpOnOff":2,"tuyaBacklightSwitch":1}},"manuSpecificTuya_3":{"attributes":{"53248":0,"53249":0,"53250":0,"53251":0,"53252":0,"53253":0,"powerOnBehavior":2,"switchType":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b002a2ed6ec","endpointID":1},{"cluster":2820,"type":"endpoint","deviceIeeeAddress":"0x00124b002a2ed6ec","endpointID":1},{"cluster":1794,"type":"endpoint","deviceIeeeAddress":"0x00124b002a2ed6ec","endpointID":1},{"cluster":1026,"type":"endpoint","deviceIeeeAddress":"0x00124b002a2ed6ec","endpointID":1}],"configuredReportings":[{"cluster":1026,"attrId":0,"minRepIntval":60,"maxRepIntval":3600,"repChange":0,"manufacturerCode":null},{"cluster":2820,"attrId":1285,"minRepIntval":5,"maxRepIntval":3600,"repChange":5,"manufacturerCode":null},{"cluster":2820,"attrId":1288,"minRepIntval":5,"maxRepIntval":3600,"repChange":50,"manufacturerCode":null},{"cluster":2820,"attrId":1291,"minRepIntval":5,"maxRepIntval":3600,"repChange":10,"manufacturerCode":null},{"cluster":1794,"attrId":0,"minRepIntval":5,"maxRepIntval":3600,"repChange":[1,1],"manufacturerCode":null}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":67,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-1941818122},"lastSeen":1701468303243,"defaultSendRequestWhen":"immediate"}
> 
> ### Notes
> this device is recognized as [TS011F_plug_1](https://www.zigbee2mqtt.io/devices/TS011F_plug_1.html#tuya-ts011f_plug_1) but I guess it should be recognized as [this](https://www.zigbee2mqtt.io/devices/EAKCB-T-M-Z.html) (2P with protection 30ma)
> 
> These are the available sizes and options ![Cattura](https://private-user-images.githubusercontent.com/16223678/285006850-e0ab9e42-5496-4380-96f0-5ddc4209a296.PNG?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDE0Njk0NjcsIm5iZiI6MTcwMTQ2OTE2NywicGF0aCI6Ii8xNjIyMzY3OC8yODUwMDY4NTAtZTBhYjllNDItNTQ5Ni00MzgwLTk2ZjAtNWRkYzQyMDlhMjk2LlBORz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFJV05KWUFYNENTVkVINTNBJTJGMjAyMzEyMDElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjMxMjAxVDIyMTkyN1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWVlN2EzMjQ5ZjNmNTM1NTU2N2QyZDczMDk2M2YzZDVkZTQyZTQ4YmYyM2I1Yzc0NDdhOTRkYTc1MGUyODkzYzAmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.P-EfbQF0p3XrGAWL4IfoJsMctqc2Ffz7k-3Q-bqMnao)

Nothing?

closed 2023-12-01 by basalisco

git-std, 2024-02-14:
Hi @basalisco,

I received mine circuit breaker (EAYCBM-Z-2P) today and having the same issues it seems.
(Zigbee Mode TS011F_plug_3 (OTA screen)   , Zigbee Manufacturer  _TZ3000_zrm3oxsh)

no adjustable power breaking settings as in EAKCB-Z-M (i have two different versions of this model, 25amp and 1-63amp model, both expost as _TZ3000_lepzuhto), these have both adjustable power breaking settings, but exact same limits even if the models have 25 or 63 maximum rathings, probably due to same Manufactor setting, don't know what happens if i set my 25amp version to, say 35amp) .
wondering if these different versions have the same physical HW inside, probably not..

can anyone help us.
when pictures or other info is required, do let me know. 
Thanks

Perfect-Web, 2024-05-28:
@basalisco why was this issue closed ? the problem still persists ...