Koenkk/zigbee2mqtt #19824: Aqara WS-USC04 not reporting energy or power

ic-dev21, 2023-11-23:
### What happened?

Energy and power are both said to be exposed by the doc but neither populates. Both are null. 

See here:
https://www.zigbee2mqtt.io/devices/WS-USC04.html

### What did you expect to happen?

I’d like to have those values populated. 

### How to reproduce it (minimal and precise)

Add to network, see that no values are being popuplated for those two fields.

### Zigbee2MQTT version

1.33.2

### Adapter firmware version

 20221226

### Adapter

Zdongle-P

### Debug log

N/A

[image]

ic-dev21, 2023-12-05:
I can add logs tonight if need be.

ic-dev21, 2023-12-18:
```
2023-12-17 22:08:22Request 'zigbee2mqtt/bridge/request/device/configure_reporting' failed with error: 'ConfigureReporting 0x0022a300002531de/1 seMetering([{"attribute":"currentSummReceived","minimumReportInterval":5,"maximumReportInterval":3600,"reportableChange":[1,1]}], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'
2023-12-17 22:08:55Request 'zigbee2mqtt/bridge/request/device/configure_reporting' failed with error: 'ConfigureReporting 0x0022a300002531de/1 seMetering([{"attribute":"currentSummReceived","minimumReportInterval":5,"maximumReportInterval":65535,"reportableChange":[1,1]}], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'
2023-12-17 22:09:15Request 'zigbee2mqtt/bridge/request/device/configure_reporting' failed with error: 'ConfigureReporting 0x0022a300002531de/1 genOta([{"attribute":"currentFileVersion","minimumReportInterval":5,"maximumReportInterval":3600,"reportableChange":[1,1]}], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'
2023-12-17 22:10:36Request 'zigbee2mqtt/bridge/request/device/configure_reporting' failed with error: 'ConfigureReporting 0x54ef4410004b7e0a/242 genPowerCfg([{"attribute":"mainsVoltage","minimumReportInterval":60,"maximumReportInterval":3600,"reportableChange":0}], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 55282 - 242 - 28 - 1 - 7 after 10000ms)'
```

[image]

realbuz, 2024-03-07:
same issue as this closed (stale) ticket. 

#18380

Simon-CR, 2024-09-15:
still an issue

Simon-CR, 2025-01-02:
Anti stale comment

closed 2025-03-12 by github-actions[bot]