Koenkk/zigbee2mqtt #19827: Neo NAS-AB06B2 doesn't report battery percentage

zoll38, 2023-11-23:
### What happened?

Neo NAS-AB06B2 is supported since v1.33.0. However, battery percentage, battery charging and battery low are always reported as Null or NA. It was working fine before v1.33.0 with external converter. I also tried external converter with v1.33.2, but same result. other folks are reporting same issue in this discussion : https://github.com/Koenkk/zigbee2mqtt/discussions/17325

### What did you expect to happen?

battery percentage, battery charging and battery low reporting

### How to reproduce it (minimal and precise)

Install 1.33.2, register neo-AB06B2 device and look at the Exposes tab after 24 hours : 
[image]

### Zigbee2MQTT version

1.33.2

### Adapter firmware version

20221226

### Adapter

Sonoff 3.0

kwetnico, 2023-12-17:
No solution to indicate the battery under z2m?

kwetnico, 2024-01-01:
With the last version, always no battery for me 
[image]

emilianogetino, 2024-01-12:
I have the same problem, any solution? thank you.

Sabjul33, 2024-01-16:
I have the same problem, battery unknown (N/A)

constapel, 2024-02-11:
Please look into the battery reporting. Security Sirene  is important! for implemtations

Koenkk, 2024-02-12:
@zoll38 what external converter were you using?

zoll38, 2024-02-13:
I am using zigbee-herdsman-converters 18.9.0 for now.

percentage (Issue #19827)

 

JYXZAVCNFSM6AAAAAA7XTCX52VHI2DSMVQWIX3LMV43OSLTON2WKQ3PNVWWK3TUHMYTSMZZGQ3TI
NJSGU> .
<https://github.com/notifications/beacon/BEBSKKYH2Y6TI4EVJZUIRWLYTJYXZA5CNFS
M6AAAAAA7XTCX52WGG33NNVSW45C7OR4XAZNMJFZXG5LFINXW23LFNZ2KUY3PNVWWK3TUL5UWJTT

kwetnico, 2024-02-22:
> @zoll38 what external converter were you using?

Hi did you find a solution to see the battery information? Thanks

zoll38, 2024-02-22:
Hi,

After installing latest versions including converter, it seems that I can get the battery percentage now. 

[image]

kwetnico, 2024-02-22:
> Hi,
>
> After installing latest versions including converter, it seems that I can get the battery percentage now. [...]

So we have to manually install this converter? It doesn't work for me just with the latest Z2M version (1.35.3), could you tell me the procedure to install the converter and or download it? THANKS

Sabjul33, 2024-02-22:
> Hi,
>
> After installing latest versions including converter, it seems that I can get the battery percentage now. [...]

I'm interested too, i have the 1.35.3 version and no battery perrcentage.

zoll38, 2024-02-23:
I can't tell you how I made it working... I installed latest versions and after few weeks, battery status suddenly appeared. I didn't do anything else. But it took much more that 24 hours contrary to what is written in the web interface

Sabjul33, 2024-03-02:
I install the last firmware 1.36.0 same thing no battery percentage..

kwetnico, 2024-03-05:
Same for me, I have the impression that no solution or research has been provided.. 👎

Sabjul33, 2024-03-05:
I also flashed my sonoff dongle with a most recent firmware (7.3.1) same thing no percentage.
It will be great if a solution can be found.
Thanks

Koenkk, 2024-03-05:
@zoll38 

>  It was working fine before v1.33.0 with external converter. 

Do you have the external converter?

Sabjul33, 2024-03-05:
on other thread, a comment about an external convert .. May be the same ???

https://github.com/Koenkk/zigbee2mqtt/discussions/17325#discussioncomment-7592589

zoll38, 2024-03-06:
> @zoll38
> 
> > It was working fine before v1.33.0 with external converter.
> 
> Do you have the external converter?

Not sure to understand your question. Yes I have external converter. Here are the various versions I am using : 

Coordinator type
zStack3x0
Coordinator revision
20221226
Coordinator IEEE Address
0x00124b002a2ec38f
Frontend version
0.6.156
zigbee_herdsman_converters_version
18.28.3
zigbee_herdsman_version
0.33.8

Koenkk, 2024-03-06:
can you post the external converter where the battery % works here?

emilianogetino, 2024-03-06:
I am also interested in the converter, thank you.

zoll38, 2024-03-06:
Converter that I use can be found here : https://github.com/Koenkk/zigbee-herdsman-converters/tree/v18.28.3
@Koenkk  this is on your github page, so not sure why you ask me to post them here as I didn't update anything from this branch

Koenkk, 2024-03-06:
@zoll38 that's not a link to a converter, just a zhc version

Sabjul33, 2024-03-06:
Sorry but i'm a little confused sor this device we must install the converter or the zigbee2mqtt main installation is sufficient ?
For me i never install the converter, i only have zigbee2mqtt 1.36.0 installation.
Thanks for your answer.

zoll38, 2024-03-07:
Hi, 
After checking again, I am using the attached external converter on top of the standard zhc ones.
[siren.zip](https://github.com/Koenkk/zigbee2mqtt/files/14519124/siren.zip)

zoll38, 2024-03-08:
Of course it doesn't :  

Without the external converter : 

[image]

with the external converter : 

[image]

kwetnico, 2024-03-08:
I installed the converter with version 1.36 of z2m and for me it doesn't change anything, still no increase in the battery..

zoll38, 2024-03-08:
You need to wait few days,. it took a while to me. not sure why.

Sabjul33, 2024-03-08:
> I installed the converter with version 1.36 of z2m and for me it doesn't change anything, still no increase in the battery..

How do you install it ?

@Koenkk  i need to install the converter or i wait a modification in the main zigbee2mqtt install ?

Koenkk, 2024-03-08:
@zoll38 I think you also need to wait a few days when not using the external converter. If that doesn't help, can you try to:

- stop z2m
- clear `battpercentage` from the `data/state.json` file
- start z2m, see if `battpercentage` comes back

zoll38, 2024-03-09:
I am away from my house now for a week so cannot do too much tests.
When not using external converter, the battery percentage never showed up.

kwetnico, 2024-03-09:
> @zoll38 I think you also need to wait a few days when not using the external converter. If that doesn't help, can you try to:
> 
> * stop z2m
> * clear `battpercentage` from the `data/state.json` file
> * start z2m, see if `battpercentage` comes back

Hello i don't have battery in the data/state.json for this device :
"0xa4c138d338cd55db": {
        "alarm_switch": "OFF",
        "alarm_state": "normal",
        "tamper_alarm_switch": "OFF",
        "alarm_melody": "melody_2",
        "alarm_mode": "alarm_sound_light",
        "last_seen": "2024-03-09T08:21:12+01:00",
        "linkquality": 196,
        "alarm_time": 5,
        "charging": false
    },

kwetnico, 2024-03-09:
> > @zoll38 I think you also need to wait a few days when not using the external converter. If that doesn't help, can you try to:
> > 
> > * stop z2m
> > * clear `battpercentage` from the `data/state.json` file
> > * start z2m, see if `battpercentage` comes back
> 
> Hello i don't have battpercentage in the data/state.json for this device : "0xa4c138d338cd55db": { "alarm_switch": "OFF", "alarm_state": "normal", "tamper_alarm_switch": "OFF", "alarm_melody": "melody_2", "alarm_mode": "alarm_sound_light", "last_seen": "2024-03-09T08:21:12+01:00", "linkquality": 196, "alarm_time": 5, "charging": false },

[image]
[image]

emilianogetino, 2024-03-09:
I have also installed the converter and waited more than 24 hours and the battery percentage does not appear.

Sabjul33, 2024-03-09:
I've installed the external converter done by @zoll38 yesterday.
Not working and not battery information in state.json.
Now i view your post @kwetnico and in external converter menu the file not appear. I add the filename and restart zigbee2mqtt.
Oh my god it's work !!!!!

[image]

What happen !!! 🥳🥳🥳😜😜

kwetnico, 2024-03-10:
When adding a converter, do you have to delete and pair the siren?  Thank you carie still don't have any feedback for the battery

Sabjul33, 2024-03-20:
Hello, No update of this issue ??
Always the same for me battery percentage never appear..
Thanks

kwetnico, 2024-04-01:
Hi
Can you tell me if the new version 1.36.1 finally corrects the problem of increasing the battery percentage on the siren? 
 Thanks

kwetnico, 2024-04-02:
Hi @Koenkk 

Can you repair please ?
Thanks

zoll38, 2024-04-02:
Hello,

On my side, after upgrading to 1.36.1, the battery percentage is back to NA :-( 
I will wait few hours and if it doesn't evolve, I will roll back to 1.36.0.

zoll38, 2024-04-03:
After 24 hours, still no progress so I rolled back to 1.36.0

Sabjul33, 2024-04-03:
After 48h with the 1.36.1 version nothing change always N/A and ?.

gvillo, 2024-04-25:
> After 24 hours, still no progress so I rolled back to 1.36.0

It's not related to 1.36.0, the issue with the external_converters (siren.zip), it's the deprecated extend import

`const extend = require('zigbee-herdsman-converters/lib/extend');`

It should be replaced with:

`const {} = require('zigbee-herdsman-converters/lib/modernExtend');`

And it works ok!

Sabjul33, 2024-04-25:
I have the same model _TZE200_nlrfgpny .
If i understand i open my siren.js file and i change the line 
const extend = require('zigbee-herdsman-converters/lib/extend');
by 
const { } = require( 'ziqbee-herdsman-converters/lib/modernExtend' ) ;

i reboot zigbee2mqtt

and that's all . I wait if the value appear 

Thanks for your answer

gvillo, 2024-04-25:
I tried using that external converter, but some features (like tamper) wasn't working for my `_TZE204_nlrfgpny`, so for now I'll stick to default setup (which loads default zhc). No battery info btw, but at least everything else works fine.

emilianogetino, 2024-04-30:
I'm in the same situation, for my model: _TZE204_nlrfgpny
The battery indication does not work in any way, neither updating z2mqtt to the latest version, nor siren.js, nor modified siren.js.
I have modified siren.js, line 59 where it says:
manufacturerName: '_TZE200_nlrfgpny',
putting up:
manufacturerName: '_TZE204_nlrfgpny',
The result is always the same, even waiting 24 hours it never indicates battery percentage.
Does anyone know how to request that this model be included in z2mqtt or has the knowledge to modify the siren.js for this model ?

Koenkk, 2024-05-01:
Please check wether battery reporting works with the original gateway.

gvillo, 2024-05-02:
I can confirm it's not working correctly with `_TZE204_nlrfgpny` model. It's running for days (stable and lot of routers in mesh network) and nothing is being reported.

guix77, 2024-05-17:
@kwetnico I think he meant trying with the official Tuya gateway & app tied to this product.

I've been reading issues including the one in which the original converter has been developed. It looks like battery never worked except when forcing a battery charging event and for very low battery percentage.

So it's possible the device firmware does not work as we want for battery. This can only be confirmed using official Tuya gateway & app.

rafpigna, 2024-05-24:
Hello, I have the same issue.
I looked in all the previous posts, but I cant understand if there is a way to have the battery reporting working or not. 
Please someone that has this working now, could you make a simple and clear step-by-step on what to do to have it working? 
Because I was lost between updates, different converters, changes in .js files and cant understand what to do.

This is really important security information for an external siren/alarm that basically is mounted in a non easily reachable place, so if the battey is not charging or faulty, it's important to see the information to make the needed fixes or change the battery with a new one! Having the battey not reported is an high security risk

guix77, 2024-05-24:
For the 3rd time: until someone shows that battery is working consistently, for instance with the Tuya gateway, we can only suspect battery is not working at all, independently of z2m, zha or any other gateway and that it's because of the firmware.

xekil, 2024-06-13:
Hi did you find a solution to see the battery information? Thanks

Sabjul33, 2024-08-02:
Hi, everyone 

No news about this issue ?
Best regards,

kwetnico, 2024-08-02:
> Hi, everyone
> 
> No news about this issue ? Best regards,

Hello, no the battery level indication still does not work.

averter, 2024-10-12:
Hi all, Same thing for manufacturer `_TZE284_nlrfgpny`

averter, 2024-11-05:
> Hello, I have the same issue. I looked in all the previous posts, but I cant understand if there is a way to have the battery reporting working or not. Please someone that has this working now, [...]

It seems to be difficult to create such a report, since the battery starts to work "out of the blue" or after some undefined period of time (24h or even days) and there is a lot of trial and error involved. 

Similarly to others I also don't have any `battpercentage` field in the `data/state.json` file. However, interestingly within the file the siren is showing a `"charging": true,` field while being charged via the USB-C cable. So it "knows" that the battery is being charged, it's just not displaying the SoC for some reason. I'll keep investigating and report back if I find anything. 
`

davidclaverie, 2024-12-09:
hi all
im new here
i have a NAS-AB06B2 Siren TS0601
and i dont have battery status??
a solution pelase?
a read a lot of thing but anybody FIND the correct solution?
Thanks a lot (im new in this "world" domitik...and im french language people...)

averter, 2024-12-09:
As ridiculous as it sounds I'm about to buy [one tuya gateway](https://www.aliexpress.com/item/1005006863886308.html?algo_exp_id=7c86dab8-bf20-4099-9490-f2d270633eb0-0&pdp_npi=4%40dis%21GBP%2120.09%215.40%21%21%21180.98%2148.65%21%402103894417337630280341841e48e3%2112000038553154879%21sea%21UK%210%21ABX&curPageLogUid=FqeBxlHVeRyM&utparam-url=scene%3Asearch%7Cquery_from%3A) for this one testing purpose. Before doing so, if anyone has a tuya gateway can you please step forward and test if the battery works or not? Thanks.

gregoriusus, 2024-12-19:
I tested with tuya gateway. 

I disconnected it from HA and connected to Tuya. It shows battery 100%, although I do some testing with sound and lights and it passed now more than one month. So, I think that siren only sends battery value when level changes. That's why after few months, magically, battery status shows up in HA. As siren is not battery hungry device and also on daily light, it recharge itself.
[image]

kwetnico, 2024-12-20:
> I tested with tuya gateway.
>
> I disconnected it from HA and connected to Tuya. It shows battery 100%, although I do some testing with sound and lights and it passed now more than one month. So, I [...]

Hello, under z2m nothing is displayed for the battery level. We should have at least 100%.

kwetnico, 2024-12-20:
A person with a Tuya gateway with the siren would this not help to troubleshoot and finally display the battery information? @Koenkk

[46 more replies]

closed 2025-12-07 by github-actions[bot]