Koenkk/zigbee2mqtt #19832: [New device support]:  Tuya Smart Home Multiple Zigbee Smart Home Control Panel

MiAutomations, 2023-11-23:
### Link

https://www.aliexpress.com/item/1005005857660356.html

### Database entry

{"id":64,"type":"Router","ieeeAddr":"0xa4c1389e2caf2f02","nwkAddr":39228,"manufId":4417,"manufName":"_TZE200_mua6ucdj","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"��u,\u0017\b\u0000\u0000\u0000i\b\u0000\u0000\u0000i\b\u0000\u0000\u0000i\t\u0000\u0000\u0000i\t\u0000\u0000\u0000i\n\u0000\u0000\u0000i\n\u0000\u0000\u0000i\n\u0000\u0000\u0000i \u0000\u0000\u0000f!\u0000\u0000\u0000\u0012","65506":53,"65508":0,"modelId":"TS0601","manufacturerName":"_TZE200_mua6ucdj","powerSource":1,"zclVersion":3,"appVersion":65,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":65,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1700688783184,"defaultSendRequestWhen":"immediate"}

### Comments

Hello 

I try to add this device without success for now the device appear as not Supported 

Can you please add this device to the list of supported devices? 

Thank you

closed 2024-06-21 by github-actions[bot]

MiAutomations, 2024-12-24:
Any news about this request? 

Thank you

sleepypig2003, 2025-05-06:
This is a duplicate of https://github.com/Koenkk/zigbee2mqtt/issues/18982

The device is the Golden Security ESP32-S3 480x480 panel
Chip is ESP32-S3 (QFN56) (revision v0.2)
Features: WiFi, BLE, Embedded PSRAM 8MB (AP_3v3)
Tuya ZT3L for Zigbee

To integrate to Homeassistant or to use MQTT it can be reflashed to OpenHASP (which means you can customize any button/user interface/images/etc to whatever you want but using OpenHASP firmware will disable ZigBee since Openhasp doesn't support Tuya's ZT3L zigbee chip).

See here: 
https://github.com/HASwitchPlate/openHASP/discussions/458
https://github.com/HASwitchPlate/openHASP/discussions/578
https://www.openhasp.com/0.7.0/hardware/golden-security/gs-t3e/