Koenkk/zigbee2mqtt #19833: [New device support]: Aqara T1 light strip controller ZNDDQDQ12LM

MiAutomations, 2023-11-23:
### Link

https://www.aqara.cn/light-strip-controller_overview

### Database entry

{"id":80,"type":"Router","ieeeAddr":"0x54ef4410008de36c","nwkAddr":15179,"manufId":4447,"manufName":"Aqara","powerSource":"DC Source","modelId":"lumi.dimmer.acn004","epList":[1,21],"endpoints":{"1":{"profId":260,"epId":1,"devId":258,"inClusterList":[5,4,3,0,768,8,6,2820,1794,64704],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"lumi.dimmer.acn004","manufacturerName":"Aqara","powerSource":4,"zclVersion":3,"appVersion":23,"stackVersion":27,"hwVersion":1,"dateCode":"20230605"}}},"binds":[],"configuredReportings":[],"meta":{}},"21":{"profId":260,"epId":21,"devId":0,"inClusterList":[12],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":23,"stackVersion":27,"hwVersion":1,"dateCode":"20230605","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1700755744718,"defaultSendRequestWhen":"immediate"}

### Comments

Hello Team 

Can you please try to add this new device from the Aqara? 

Thank you

MiAutomations, 2024-02-09:
Hello Any news regarding this device ? 

Thank you

Koenkk, 2024-02-10:
Does it work out of the box with 1.35.3?

MiAutomations, 2024-02-16:
> Does it work out of the box with 1.35.3?

No, Still facing the issue as unsupported device even with the last update on Z2M and the device FW 

[image]

Koenkk, 2024-02-18:
Is nothing exposes when clicking on the device -> expose page? 

Could you try with the [generated external converter](https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html#_2-creating-the-external-definition)?

MiAutomations, 2024-02-25:
> Is nothing exposes when clicking on the device -> expose page?
> 
> Could you try with the [generated external converter](https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html#_2-creating-the-external-definition)?

Hello 

In the exposes tab nothing appears as you van check in this image:
[image]

As you suggested I try also generated external converter: 

[image]

But from my understanding the device still without any functions available ....

Koenkk, 2024-02-25:
What color modes does it support?

MiAutomations, 2024-02-25:
> What color modes does it support?

Only this ones 

[image]

Koenkk, 2024-02-26:
Added!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-02-26 by Koenkk