Koenkk/zigbee2mqtt #19835: [New device support]:  Aqara Switch Z1 model ZNQBKG41LM

MiAutomations, 2023-11-23:
### Link

https://www.aqara.cn/Canon_Smart_Wall_Switch_Z1_overview

### Database entry

{"id":82,"type":"EndDevice","ieeeAddr":"0x54ef44100096dd21","nwkAddr":20903,"manufId":4447,"manufName":"Aqara","powerSource":"DC Source","modelId":"lumi.switch.acn055","epList":[1,2,3,4,21],"endpoints":{"1":{"profId":260,"epId":1,"devId":0,"inClusterList":[2820,1794,18,5,4,6,3,0,64704],"outClusterList":[10,25],"clusters":{"genBasic":{"attributes":{"modelId":"lumi.switch.acn055","manufacturerName":"Aqara","appVersion":21,"powerSource":4,"zclVersion":3,"stackVersion":27,"hwVersion":1,"dateCode":"20230526"}}},"binds":[],"configuredReportings":[],"meta":{}},"2":{"profId":260,"epId":2,"devId":0,"inClusterList":[18,5,4,6,64704],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"3":{"profId":260,"epId":3,"devId":0,"inClusterList":[18,5,4,6,64704],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"4":{"profId":260,"epId":4,"devId":0,"inClusterList":[18,64704],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"21":{"profId":260,"epId":21,"devId":0,"inClusterList":[12],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":21,"stackVersion":27,"hwVersion":1,"dateCode":"20230526","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1700757823520,"defaultSendRequestWhen":"immediate"}

### Comments

Hello Team 

Can you please include this device in the list of devices supported ? 

Thank you

MiAutomations, 2023-11-26:
> See if it helps? Seems [nickgoh] did something for that. [#18135 (comment)](https://github.com/Koenkk/zigbee2mqtt/issues/18135#issuecomment-1778487502)

Great Thank you I will try

leeaash, 2023-12-07:
> See if it helps? Seems [nickgoh] did something for that. [#18135 (comment)](https://github.com/Koenkk/zigbee2mqtt/issues/18135#issuecomment-1778487502)

I have tried that, but i can’t only get 3 button show up, no button 4 shows up on my end. 
[image]

jomud9, 2024-01-03:
@leeaash It should be normal as 4 gang switch is 3 relay + 1 virtual, and the pic you shown is the "physical relay".

I did not have that switch but you could check the action output and see if it shown the button4 is functioning

leeaash, 2024-01-03:
> @leeaash It should be normal as 4 gang switch is 3 relay + 1 virtual, and the pic you shown is the "physical relay".
> 
> I did not have that switch but you could check the action output and see if it shown the button4 is functioning

You're right, the gan 4 action shows as action output such as 'single_button_4', 'double_button_4'

closed 2024-08-01 by github-actions[bot]