Koenkk/zigbee2mqtt #19837: [New device support]: Namron 2 channel switch 4512768

Aeliassen, 2023-11-23:
### Link

https://www.elektroimportoren.no/namron-zigbee-2-kanal-16a-bryter-hvit/4512768/Product.html

### Database entry

not sure how to find it

### Comments

It's a 2 rocker switch. I tried to follow the guide and look at other Namron devices. The code below lets me control one of the rocker, it also shows me voltage for the switch, watt usage and accumulated usage. 

I can't figure out how to get the other rocker to work. 

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['4512768'],
    model: '4512768',
    vendor: 'Namron',
    description: 'Zigbee 2 channel switch',
    fromZigbee: [fz.metering, fz.electrical_measurement, fz.on_off],
    toZigbee: [tz.on_off],
    exposes: [e.power(), e.current(), e.voltage(), e.energy(), e.switch()],
    extend: extend.switch(),
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1) || device.getEndpoint(3);
        const binds = ['seMetering', 'haElectricalMeasurement', 'genOnOff',];
        await reporting.bind(endpoint, coordinatorEndpoint, binds);
        await reporting.onOff(endpoint);

        // Metering
        await reporting.readEletricalMeasurementMultiplierDivisors(endpoint);
        await reporting.readMeteringMultiplierDivisor(endpoint);
        await reporting.rmsVoltage(endpoint, {min: 10, change: 20}); // Voltage - Min change of 2v
        await reporting.rmsCurrent(endpoint, {min: 10, change: 10}); // A - z2m displays only the first decimals, so change of 10 (0,01)
        await reporting.activePower(endpoint, {min: 10, change: 15}); // W - Min change of 1,5W
        await reporting.currentSummDelivered(endpoint, {min: 300}); // Report KWH every 5min
    },
};

module.exports = definition;
```

Aeliassen, 2023-11-24:
I figured it out :)

closed 2023-11-24 by Aeliassen

Aeliassen, 2023-12-04:
Support for this switch is added in the newest release of zigbee2mqtt.

AndKe, 2024-09-22:
@Aeliassen Hi, I am using Z2M ver 1.39.1  , which surely has this device, but my issue it that it won't pair (I see no attempt to pair.)
Did you had any issues making it pair? - maybe upgraded firmware or something?
My coordinator revision is 
20230507