Koenkk/zigbee2mqtt #19839: TS0202 Presence sensor _TZ3000_6ygjfyll -> Keep time and Sensitivity

Evililim, 2023-11-23:
### What happened?

Hello,

I have add 10 of these  TS0202 presence sensor _TZ3000_6ygjfyll in Z2M. They work very well, but i found 2 issues.

First, the expose "sensitivity parameter" (low/medium/hight) :

- no Sensitivity parameter in TUYA app
- not in tuya device debogging
- When i try to change the parameter in Z2M in HA i have this error message :

> Publish 'set' 'sensitivity' to '*************' failed: 'Error: Write 0xa4c1386ab50a6a5a/1 ssIasZone({"currentZoneSensitivityLevel":0}, {"sendWhen":"active","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received (197))'
- This expose can be delete for this device

 Second, the expose "Keep time" (30/60/120) :
- This parameter is in TUYA app
- In the IOT TUYA platform / device debogging I found the Data Point ID is 10.
- When i try to change the parameter in Z2M in HA i have this error message :
> Publish 'set' 'keep_time' to '************' failed: 'Error: Write 0xa4c1386ab50a6a5a/1 ssIasZone({"61441":{"value":0,"type":32}}, {"sendWhen":"active","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received (54))'
- What information can I provide to help solve this problem?

### What did you expect to happen?

- no "sensitivity" parameter in expose
- be able to change the "keep time" parameter in expose

### How to reproduce it (minimal and precise)

Try to change the parameters "Keep time" or "Sensivity" in the Exposes for TS0202  _TZ3000_6ygjfyll devices.

### Zigbee2MQTT version

1.33.2

### Adapter firmware version

0x26780700

### Adapter

ConBee2/RaspBee2

udayms, 2023-11-28:
I am also facing this problem.

Tonny56, 2024-02-01:
I have this problem too.

G-glop, 2024-02-02:
I've managed to successfully change the parameters by activating the sensor (it makes a red blink) right after clicking the button in Z2M. It's a bit of a pain in the ass with longer keep times and multiple sensors

closed 2024-09-01 by github-actions[bot]