Koenkk/zigbee2mqtt #19842: [New device support]: ZNQBKG38LM Xiaomi Aqara Z1 1 gang smart wall switch

jomud9, 2023-11-24:
### Link

https://www.aqara.com/cn/Canon_Smart_Wall_Switch_Z1_overview

### Database entry

This is a wrong one. Will replace once got one {"id":3,"type":"EndDevice","ieeeAddr":"0x54ef4410008ea781","nwkAddr":47837,"manufId":4447,"manufName":"Aqara","powerSource":"DC Source","modelId":"lumi.switch.acn049","epList":[1,2,3,4,21],"endpoints":{"1":{"profId":260,"epId":1,"devId":0,"inClusterList":[2820,1794,18,5,4,6,3,0,64704],"outClusterList":[10,25],"clusters":{"genBasic":{"attributes":{"modelId":"lumi.switch.acn049","manufacturerName":"Aqara","powerSource":4,"zclVersion":3,"appVersion":21,"stackVersion":27,"hwVersion":1,"dateCode":"20230526"}},"aqaraOpple":{"attributes":{"4":4,"247":{"type":"Buffer","data":[3,40,38,5,33,11,0,9,33,0,0,10,33,132,210,12,32,10,13,35,21,0,0,0,17,35,1,0,0,0,100,16,1,101,16,1,149,57,172,71,225,60,150,57,51,171,14,69,152,57,82,184,53,66,151,57,205,108,180,67,154,32,0]},"512":1,"617":1,"645":0,"646":1,"648":0,"1303":1}},"genOnOff":{"attributes":{"245":62577920,"onOff":1}}},"binds":[],"configuredReportings":[],"meta":{}},"2":{"profId":260,"epId":2,"devId":0,"inClusterList":[18,5,4,6,64704],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"245":117440538,"onOff":1}},"aqaraOpple":{"attributes":{"512":0,"617":1,"645":0}},"genMultistateInput":{"attributes":{"presentValue":1}}},"binds":[],"configuredReportings":[],"meta":{}},"3":{"profId":260,"epId":3,"devId":0,"inClusterList":[64704],"outClusterList":[],"clusters":{"aqaraOpple":{"attributes":{"512":1,"617":1,"645":0}}},"binds":[],"configuredReportings":[],"meta":{}},"4":{"profId":260,"epId":4,"devId":0,"inClusterList":[64704],"outClusterList":[],"clusters":{"aqaraOpple":{"attributes":{"617":0}}},"binds":[],"configuredReportings":[],"meta":{}},"21":{"profId":260,"epId":21,"devId":0,"inClusterList":[12],"outClusterList":[],"clusters":{"genAnalogInput":{"attributes":{"presentValue":45.91999816894531}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":21,"stackVersion":27,"hwVersion":1,"dateCode":"20230526","zclVersion":3,"interviewCompleted":true,"meta":{"configured":2114939527},"lastSeen":1687378993322,"defaultSendRequestWhen":"immediate"}

### Comments

requested from various z1 1 gang owners, including @maodun96 who dropped me a message and nice comment from @nickgoh. I tried to draft one for z1 1 gang.

As I did not own one, please comment 

### External converter

```shell
//. Changlog 20231124 - first ver for ZNQBKG38LM
//. Continue work from nickgoh - https://github.com/Koenkk/zigbee2mqtt/issues/18135#issuecomment-1778487502
//. 
//. 1. still thinking this version of firmware only support
//.     - action: single (no double, triple, hold)
//.     - no multiple button pressed together (no both, all)
//. 2. as button arranged in "top to buttom" instead of "standard" left to right, I renamed as button_1 instead of left
//. 3. Need help on function to make these working
//.   - flip_indicator_light, power_outage_memory and click mode
//.   - Seems address got changed but I have no idea

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');

//. PLEASE READ: These are present from device/xiaomi.ts, only needed for external converter
const herdsman = require('zigbee-herdsman');
const e = exposes.presets;
const ea = exposes.access;

const manufacturerOptions = {
    xiaomi: {manufacturerCode: herdsman.Zcl.ManufacturerCode.LUMI_UNITED_TECH, disableDefaultResponse: true},
};

const { hasAlreadyProcessedMessage } = require('zigbee-herdsman-converters/lib/utils');

const preventReset = async (type, data, device) => { 
    if (
        // options.allow_reset ||
        type !== 'message' ||
        data.type !== 'attributeReport' ||
        data.cluster !== 'genBasic' ||
        !data.data[0xfff0] ||
        // eg: [0xaa, 0x10, 0x05, 0x41, 0x87, 0x01, 0x01, 0x10, 0x00]
        !data.data[0xFFF0].slice(0, 5).equals(Buffer.from([0xaa, 0x10, 0x05, 0x41, 0x87]))
    ) {
        return;
    }
    const options = {manufacturerCode: 0x115f};
    const payload = {[0xfff0]: {
        value: [0xaa, 0x10, 0x05, 0x41, 0x47, 0x01, 0x01, 0x10, 0x01],
        type: 0x41,
    }};
    await device.getEndpoint(1).write('genBasic', payload, options);
}; // PLEASE READ: THE END OF THE BLOCK THAT CAN BE DELELED after released

//. PLEASE READ: These should be update to FromZigbee.js 
const fzLocal = {
    xiaomi_multistate_action: {
        cluster: 'genMultistateInput',
[... 148 lines cut ...]
    exposes: [
        e.power(), e.voltage(), e.device_temperature(),
        e.switch().withEndpoint('button_1'), 
        exposes.enum('power_outage_memory', ea.ALL, ['on', 'electric_appliances_on', 'electric_appliances_off', 'Inverted'])
                .withDescription('Power Outage Memory'),
        exposes.enum('operation_mode', ea.ALL, ['control_relay', 'decoupled'])
                .withDescription('Decoupled mode for button 1 button')
                .withEndpoint('button_1'),
                
        exposes.enum('lock_relay', ea.ALL, ['On', 'Off'])
                .withDescription('Lock Relay mode for button 1 button')
                .withEndpoint('button_1'),
                
        exposes.enum('click_mode', ea.ALL, ['fast', 'multi'])
    ],
    onEvent: preventReset,
    configure: async (device, coordinatorEndpoint, logger) => {
        await device.getEndpoint(1).write('aqaraOpple', {'mode': 1}, {manufacturerCode: 0x115f, disableResponse: true});
//       await device.getEndpoint(1).write('aqaraOpple', {0x0125: {value: 0x02, type: 0x20}}, {manufacturerCode: 0x115f});
    },

};

module.exports = definition;
```

jomud9, 2023-11-24:
Dear @maodun96 , Would you provide me the Database entry after your put these code into your device? 

If the code works I will submit changes to  source.

maodun96, 2023-11-24:
> Dear @maodun96 , Would you provide me the Database entry after your put these code into your device?
> 
> If the code works I will submit changes to source.

I would be happy to serve you. This is my Database entry. My Aqara Z1 1 gang has been successfully driven.

Database entry:
{"id":2,"type":"EndDevice","ieeeAddr":"0x54ef44100094d5ea","nwkAddr":36341,"manufId":4447,"manufName":"Aqara","powerSource":"DC Source","modelId":"lumi.switch.acn048","epList":[1,2,3,4,21],"endpoints":{"1":{"profId":260,"epId":1,"devId":0,"inClusterList":[2820,1794,18,5,4,6,3,0,64704],"outClusterList":[10,25],"clusters":{"genBasic":{"attributes":{"modelId":"lumi.switch.acn048","manufacturerName":"Aqara","powerSource":4,"zclVersion":3,"appVersion":31,"stackVersion":27,"hwVersion":1,"dateCode":"20231102"}},"aqaraOpple":{"attributes":{"4":4,"223":{"type":"Buffer","data":[3,35,92,0,0,0,4,33,5,0,5,33,0,0,41,35,82,162,50,0,42,35,71,179,0,8,40,35,64,0,0,0,43,33,0,0,44,41,160,255]},"247":{"type":"Buffer","data":[3,40,27,5,33,131,0,9,33,0,0,10,33,0,0,12,32,10,13,35,31,0,0,0,17,35,1,0,0,0,100,16,0,149,57,0,0,0,0,150,57,102,102,27,69,152,57,0,0,0,0,151,57,0,0,0,0,154,32,0]},"512":1,"617":1,"645":0,"646":1,"648":0,"651":1,"1303":1}},"genOnOff":{"attributes":{"245":42857728,"onOff":0}}},"binds":[],"configuredReportings":[],"meta":{}},"2":{"profId":260,"epId":2,"devId":0,"inClusterList":[64704],"outClusterList":[],"clusters":{"aqaraOpple":{"attributes":{"512":1,"617":1,"645":0}}},"binds":[],"configuredReportings":[],"meta":{}},"3":{"profId":260,"epId":3,"devId":0,"inClusterList":[64704],"outClusterList":[],"clusters":{"aqaraOpple":{"attributes":{"512":1,"617":1,"645":0}}},"binds":[],"configuredReportings":[],"meta":{}},"4":{"profId":260,"epId":4,"devId":0,"inClusterList":[64704],"outClusterList":[],"clusters":{"aqaraOpple":{"attributes":{"617":0}}},"binds":[],"configuredReportings":[],"meta":{}},"21":{"profId":260,"epId":21,"devId":0,"inClusterList":[12],"outClusterList":[],"clusters":{"genAnalogInput":{"attributes":{"presentValue":0}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":31,"stackVersion":27,"hwVersion":1,"dateCode":"20231102","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1854471209},"lastSeen":1700850656590,"defaultSendRequestWhen":"immediate"}

jomud9, 2023-11-27:
@maodun96 那太好了. 如果周末有時間我也試試 2 /3 gang 改動一下, 沒大問題就 submit 有關的 check, 那就不再需要用 converter 了. :-)

closed 2024-03-14 by jomud9