Koenkk/zigbee2mqtt #19847: [New device support]: Tuya ZigBee Mini Dimmable Switch FS-05R (TS0052 / _TZ3000_mgusv51k)

talkout, 2023-11-24:
### Link

https://www.aliexpress.com/item/1005005824579031.html

### Database entry

{"id":11,"type":"Router","ieeeAddr":"0xa4c1388aa502fbec","nwkAddr":56498,"manufId":4417,"manufName":"_TZ3000_mgusv51k","powerSource":"Mains (single phase)","modelId":"TS0052","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[3,4,5,6,57344,57345,8,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"^��,\u0012ч�,\u0012@��,\u0012У�,\u0012\u0005\u0000\u0000\u0000\u0012\u0005\u0000\u0000\u0000\u0012\u0005\u0000\u0000\u0000\u0012�\u000b\u0000\u0000\u0012","65506":54,"65508":0,"65534":0,"modelId":"TS0052","manufacturerName":"_TZ3000_mgusv51k","powerSource":1,"zclVersion":3,"appVersion":68,"stackVersion":0,"hwVersion":1,"dateCode":""}},"manuSpecificBosch":{"attributes":{"53251":"AAAA"}},"genOnOff":{"attributes":{"onOff":1,"onTime":0,"offWaitTime":0,"moesStartUpOnOff":2}},"manuSpecificTuya_3":{"attributes":{"switchType":2}},"genLevelCtrl":{"attributes":{"currentLevel":2,"minLevel":2,"maxLevel":254,"onOffTransitionTime":0}}},"binds":[],"configuredReportings":[],"meta":{"onLevelSupported":false}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":68,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-1440156621},"lastSeen":1700818080387,"defaultSendRequestWhen":"immediate"}

### Comments

By default this pairs as "TS0052", but the functionalities and controls are not working correctly. 
1. the on/'off toggle doesn't work
2. when toggled light dims down to 2, and doesn't goes back
3. only way to make on work is to slide brightness back. 

I added an external converter and fixed this for myself and it work.

Not sure if this needs to be a new device or something that already exists, but pairs incorrectly.

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require("zigbee-herdsman-converters/lib/tuya");
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');

const {
    precisionRound, mapNumberRange, isLegacyEnabled, toLocalISOString, numberWithinRange, hasAlreadyProcessedMessage,
    calibrateAndPrecisionRoundOptions, addActionGroup, postfixWithEndpointName, getKey,
    batteryVoltageToPercentage, getMetaValue,
} = require('zigbee-herdsman-converters/lib/utils');

const fzLocal = {
    brightness: {
        cluster: 'genLevelCtrl',
        type: ['attributeReport', 'readResponse'],
        convert: (model, msg, publish, options, meta) => {
            if (msg.data.hasOwnProperty('61440')) {
                const property = postfixWithEndpointName('brightness', msg, model);
                return {[property]: msg.data['61440']};
            }
        },
    },
}

const definition = {
    // Since a lot of Tuya devices use the same modelID, but use different data points
    // it's usually necessary to provide a fingerprint instead of a zigbeeModel
    fingerprint: [
        {
            // The model ID from: Device with modelID 'TS0601' is not supported
            // You may need to add \u0000 at the end of the name in some cases
            modelID: 'TS0052',
            // The manufacturer name from: Device with modelID 'TS0601' is not supported.
            manufacturerName: '_TZ3000_mgusv51k'
        },
    ],
    model: 'FS-05R', // Vendor model number, look on the device for a model number
    vendor: 'TuYa', // Vendor of the device (only used for documentation and startup logging)
    description: 'Mini Dimmable Switch', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    fromZigbee: [fz.on_off, fzLocal.brightness], // We will add this later
    toZigbee: [tz.on_off, tz.light_onoff_brightness], // Should be empty, unless device can be controlled (e.g. lights, switches).
    exposes: [e.linkquality(),e.light_brightness()], // Defines what this device exposes, used for e.g. Home Assistant discovery and in the frontend
};

module.exports = definition;
```

boojew, 2024-01-14:
Amazing! Thanks for this! 

I had to change the manufacturer to _TZ3000_zjtxnoft, but otherwise works perfectly for on/off. Setting the transition time doesnt work, but definitely progress :)

Koenkk, 2024-01-15:
Could you check if the issue is fixed with the following [external converter](https://gist.github.com/Koenkk/7b18f5be3ea07bc2cc94fe08c56c09e4): 

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

KipK, 2024-01-26:
This fix some problems, but min/max brightness settings are missing here

RemCom, 2024-01-28:
@Koenkk I found out today that I had the same 2% issue when turning it off. Tested the external converter and solved the issue for me

Koenkk, 2024-01-28:
Integrated the fix, thanks @KipK 

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-01-28 by Koenkk

upandabout, 2024-02-27:
I have the FS-05R dimmer and it's not working 100% in the current version of zigbee2mqtt. I have several problems.
1. I can not in the frontend set minimum dimmer level when in the device Exposes tab.
2. Switching the Do not disturb on makes the switch uncontrollable for a while, perhaps needing a power cycle to get back to normal.
3. The transition setting under Settings (Specific) is making the light impossible to turn off, when set. It will only change the dimmer to minimum Dim level. (I had to set the minimum dim level using the dev console)
4. I think I should be able to change the light type between LED, Incandescent and Halogen light type, but not sure if it's applicable to this unit, but as my LED bulb is blitzing when dimmed from time to time, I think this might be the issue. The reason I think this might be supported is this post:/comment: https://github.com/Koenkk/zigbee2mqtt/issues/12793#issuecomment-1157933067
However I still don't have a Tuya Gateway Hub to test out what exactly the Tuya app would let me do... 

Below is a pic of my Exposes page. 
[image]

upandabout, 2024-02-27:
> @Koenkk I found out today that I had the same 2% issue when turning it off. Tested the external converter and solved the issue for me

Did you enable a transition time in the Settings (specific) tab under the device? If yes, try to delete the value. In my case it enabled the light to switch off.

strati64, 2024-06-07:
I have the same problem with this hardware in home assistant Zigbee2mqtt integration (but on/off work !) and it is recognized as a Tuya TS0052 and not FS-05R

Zigbee2MQTT  Version     [1.38.0] 
Zigbee-herdsman-converters  Version     19.47.1
Zigbee-herdsman Version     0.49.2

upandabout, 2024-06-07:
> Quote reply

What exactly is your problem, many different problems has been on the table for this product.

vega6763, 2024-06-29:
I have the same problem, when configured by default the ON/OFF button does not turn to OFF but stays at 2%.
If i change the transition to '0', the ON/OFF button is OK, the dimming either, but the bulb still slightly ON.

I can see in mqtt explorer, for the first case that a set = brightness : "2" is send a couple seconds after the setstate : OFF causing the command to not shutoff but the bulb is OFF contrary the second case.

I attempted to modify the toZigbee.js but it does not save modification after HA reboot.
[image]

vega6763, 2024-07-01:
I have two FS-05R, the first works flawlessy, the second one has the issues listed in my previous post.

This is an extraction from zigbee2mqtt/database.db, id18 is the working one, id23 has with issue. Something may lead to manu specificbosch missing in the second.

1. {"id":18,"type":"Router","ieeeAddr":"0xa4c1387fa20791d2","nwkAddr":60915,"manufId":4417,"manufName":"_TZ3000_mgusv51k","powerSource":"Mains (single phase)","modelId":"TS0052","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[3,4,5,6,57344,57345,8,0],"outClusterList":[25,10],"clusters":{"57344":{"attributes":{"53251":"AAAA"}},"genBasic":{"attributes":{"modelId":"TS0052","manufacturerName":"_TZ3000_mgusv51k","powerSource":1,"zclVersion":3,"appVersion":68,"stackVersion":0,"hwVersion":1,"dateCode":""}},"genOnOff":{"attributes":{"onOff":1,"onTime":0,"offWaitTime":0,"moesStartUpOnOff":2}},"genLevelCtrl":{"attributes":{"currentLevel":2,"minLevel":2,"maxLevel":254,"onOffTransitionTime":50}},"manuSpecificBosch":{"attributes":{"53251":"AAAA"}},"manuSpecificTuya_3":{"attributes":{"switchType":0,"powerOnBehavior":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b002c3ad33b","endpointID":1},{"cluster":8,"type":"endpoint","deviceIeeeAddress":"0x00124b002c3ad33b","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":65000,"repChange":1,"manufacturerCode":null},{"cluster":8,"attrId":0,"minRepIntval":10,"maxRepIntval":65000,"repChange":1,"manufacturerCode":null}],"meta":{"onLevelSupported":false}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":68,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":332242049},"lastSeen":1719820177262}
2. {"id":23,"type":"Router","ieeeAddr":"0xa4c13890532c1180","nwkAddr":20797,"manufId":4417,"manufName":"_TZ3000_mgusv51k","powerSource":"Mains (single phase)","modelId":"TS0052","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[3,4,5,6,57344,57345,8,0],"outClusterList":[25,10],"clusters":{"57344":{"attributes":{"53251":"AAAA"}},"genBasic":{"attributes":{"modelId":"TS0052","manufacturerName":"_TZ3000_mgusv51k","powerSource":1,"zclVersion":3,"appVersion":68,"stackVersion":0,"hwVersion":1,"dateCode":""}},"genOnOff":{"attributes":{"onOff":0,"onTime":0,"offWaitTime":0,"moesStartUpOnOff":2}},"genLevelCtrl":{"attributes":{"currentLevel":149,"minLevel":2,"maxLevel":254,"onOffTransitionTime":50}},"manuSpecificTuya_3":{"attributes":{"switchType":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b002c3ad33b","endpointID":1},{"cluster":8,"type":"endpoint","deviceIeeeAddress":"0x00124b002c3ad33b","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":65000,"repChange":1,"manufacturerCode":null},{"cluster":8,"attrId":0,"minRepIntval":10,"maxRepIntval":65000,"repChange":1,"manufacturerCode":null}],"meta":{"onLevelSupported":false}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":68,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":332242049},"lastSeen":1719819899724}

FedericoKlappenbach, 2025-01-27:
Any development on this? Which converter works the best?

FedericoKlappenbach, 2025-01-27:
> I have the same problem, when configured by default the ON/OFF button does not turn to OFF but stays at 2%. If i change the transition to '0', the ON/OFF button is OK, the dimming either, but the [...]

I have the same problem with latest Z2M 2.0.0

vega6763, 2025-01-27:
In settings (specific) leave transition blank (not 0 but blank). Restart Z2M and test again

[image]