Koenkk/zigbee2mqtt #19849: TS0004 - _TZ3000_hdlpifbk - cannot control individual switches

wizu, 2023-11-24:
### What happened?

When I turn on one of the switches, the other three turns off immediately.
I'm almost sure, that it wasn't working like that a few years ago when I had it installed.

### What did you expect to happen?

I would expect, that I would be able to turn on two or all of them at same time

### How to reproduce it (minimal and precise)

Turn on first switch, then turn on second. First one if turned off automatically.

### Zigbee2MQTT version

 1.33.2

### Adapter firmware version

20230507

### Adapter

zStack3x0

micahed, 2023-11-25:
One of the three modes for the device is an interlock mode , it may have been accidentally selected using the 'mode switch' on the front of the device.

wizu, 2023-11-26:
🤦‍♂️ you are right. Thanks

closed 2023-11-26 by wizu