Koenkk/zigbee2mqtt #19853: [EZSP] ZGP / greenPower messages are not relayed by HUE routers

rsegers, 2023-11-24:
### What happened?

Zigbee Green Power (ZGP) messages are only broadcasted by the coordinator, so they only work in close proximity to it.

### What did you expect to happen?

I have several HUE routers in range of the ZGP switches (EnOcean PTM 215Z).
I expect those device to also pick up and relay ZGP messages.

### How to reproduce it (minimal and precise)

1.) Create a Zigbee network with z2m and a conbee II coordinator on channel 11
2.) Configure several ZGP switches for channel 11
3.) Add HUE routers to the network
4.) Position the ZGP switches far from the coordinator but in close range to HUE routers

### Zigbee2MQTT version

1.33.2-dev commit: 746066c

### Adapter firmware version

6.10.3.0 build 297

### Adapter

EZSP Sonoff Dongle Plus-E EFR32MG21 

### Debug log

None

rsegers, 2023-11-24:
Cloud this solve the issue? https://github.com/Koenkk/zigbee-herdsman/compare/master...rsegers:zigbee-herdsman:master

posixx, 2023-11-30:
Well this shoud be solved before the release of tomorrow as we use this a lot, and with the current release there are no problems!

rsegers, 2023-12-07:
> > Cloud this solve the issue? [Koenkk/zigbee-herdsman@master...rsegers:zigbee-herdsman:master](https://github.com/Koenkk/zigbee-herdsman/compare/master...rsegers:zigbee-herdsman:master)
> 
> Does it fix the issue for you?

Nope, it doesn't. But after adding the change, pairing a PTM 215Z through a Hue router now shows up in the Herdsman logging. Pairing does not occur, because Zigbee2MQTT thinks it is the coordinator?

```
2023-12-06T21:07:35.648Z zigbee-herdsman:adapter:ezsp:uart <-- [0132b1a90d2a92abdaf0b12eabdd8549f0f925b0e4320e7ff17e]
2023-12-06T21:07:35.648Z zigbee-herdsman:adapter:ezsp:uart <-- DATA (0,1,0): 0132b1a90d2a92abdaf0b12eabdd8549f0f925b0e4320e7ff17e
2023-12-06T21:07:35.649Z zigbee-herdsman:adapter:ezsp:uart --> ACK  (1)
2023-12-06T21:07:35.649Z zigbee-herdsman:adapter:ezsp:uart --> [8160597e]
2023-12-06T21:07:35.649Z zigbee-herdsman:adapter:ezsp:ezsp <== Frame: 709001590087198364fb0b018817006cb7021b09fc69
2023-12-06T21:07:35.650Z zigbee-herdsman:adapter:ezsp:ezsp <== 0x59: {"cls":"incomingRouteRecordHandler","id":89,"isRequest":false,"source":6535,"longId":{"type":"Buffer","data":[0,23,136,1,11,251,100,131]},"lastHopLqi":108,"lastHopRssi":-73,"relay":{"type":"Buffer","data":[27,9]}}
2023-12-06T21:07:35.650Z zigbee-herdsman:adapter:ezsp:driv handleRouteRecord: nwk=6535, ieee=
2023-12-06T21:07:35.705Z zigbee-herdsman:adapter:ezsp:uart <-- [7d3132b1a97d312a1552f8b54ad7585593499ccc431e6ad79874ebd7778ddc75b2429dcc596f8fffd2db52cbbd6750fb73ea81407e]
2023-12-06T21:07:35.705Z zigbee-herdsman:adapter:ezsp:uart <-- DATA (1,1,0): 1132b1a9112a1552f8b54ad7585593499ccc431e6ad79874ebd7778ddc75b2429dcc596f8fffd2db52cbbd6750fb73ea81407e
2023-12-06T21:07:35.705Z zigbee-herdsman:adapter:ezsp:uart --> ACK  (2)
2023-12-06T21:07:35.706Z zigbee-herdsman:adapter:ezsp:uart --> [82503a7e]
2023-12-06T21:07:35.706Z zigbee-herdsman:adapter:ezsp:ezsp <== Frame: 709001450000e0a12100f2f2000100008264b58719ffff16111404200b8de576018700000015008719d4eb16d8da06
2023-12-06T21:07:35.707Z zigbee-herdsman:adapter:ezsp:ezsp <== 0x45: {"cls":"incomingMessageHandler","id":69,"isRequest":false,"type":0,"apsFrame":{"profileId":41440,"clusterId":33,"sourceEndpoint":242,"destinationEndpoint":242,"options":256,"groupId":0,"sequence":130},"lastHopLqi":100,"lastHopRssi":-75,"sender":6535,"bindingIndex":255,"addressIndex":255,"message":{"type":"Buffer","data":[17,20,4,32,11,141,229,118,1,135,0,0,0,21,0,135,25,212,235,22,216,218]}}
2023-12-06T21:07:35.707Z zigbee-herdsman:adapter:ezsp:debg processMessage: {"messageType":0,"apsFrame":{"profileId":41440,"clusterId":33,"sourceEndpoint":242,"destinationEndpoint":242,"options":256,"groupId":0,"sequence":130},"lqi":100,"rssi":-75,"sender":6535,"bindingIndex":255,"addressIndex":255,"message":{"type":"Buffer","data":[17,20,4,32,11,141,229,118,1,135,0,0,0,21,0,135,25,212,235,22,216,218]}}
2023-12-06T21:07:35.709Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"reservedBits":0,"frameType":1,"direction":0,"disableDefaultResponse":true,"manufacturerSpecific":false},"transactionSequenceNumber":130,"manufacturerCode":null,"commandIdentifier":0},"Payload":{"type":"Buffer","data":[17,20,4,32,11,141,229,118,1,135,0,0,0,21,0,135,25,212,235,22,216,218]},"Command":{"ID":0,"parameters":[{"name":"options","type":33},{"name":"srcID","type":35,"conditions":[{"type":"bitFieldEnum","param":"options","offset":0,"size":3,"value":0}]},{"name":"gpdIEEEAddr","type":240,"conditions":[{"type":"bitFieldEnum","param":"options","offset":0,"size":3,"value":2}]},{"name":"gpdEndpoint","type":32,"conditions":[{"type":"bitFieldEnum","param":"options","offset":0,"size":3,"value":2}]},{"name":"frameCounter","type":35},{"name":"commandID","type":32},{"name":"payloadSize","type":32},{"name":"commandFrame","type":1009},{"name":"gppNwkAddr","type":33,"conditions":[{"type":"bitMaskSet","param":"options","mask":16384}]},{"name":"gppGddLink","type":32,"conditions":[{"type":"bitMaskSet","param":"options","mask":16384}]}],"name":"notification"}},"address":6535,"endpoint":242,"linkquality":100,"groupID":null,"wasBroadcast":true,"destinationEndpoint":242}'
2023-12-06T21:07:35.709Z zigbee-herdsman:controller:greenpower GP Undefined Command
Zigbee2MQTT:debug 2023-12-06 22:07:35: Received Zigbee message from 'Coordinator', type 'commandNotification', cluster 'greenPower', data '{"data":[17,20,4,32,11,141,229,118,1,135,0,0,0,21,0,135,25,212,235,22,216,218],"type":"Buffer"}' from endpoint 242 with groupID null, ignoring since it is from coordinator
```

posixx, 2023-12-07:
I can confirm it still works like expected after upgrading to 1.34.0, using my tubeszb-cc2652-poe-2023.

rsegers, 2023-12-07:
> I can confirm it still works like expected after upgrading to 1.34.0, using my tubeszb-cc2652-poe-2023.
This issue is related to an EZSP adapter and not a z-stack adapter

closed 2023-12-07 by rsegers