Koenkk/zigbee2mqtt #19856: Device disappeared from Z2M

Nezz, 2023-11-24:
### What happened?

One of my Ubisys S2 switches disappeared from Z2M two days ago. The device is still on the network because it can be controlled via groups and it is directly bound to other devices and other devices are bound to it. However, home assistant can no longer see it's state and control it.

The device had disappeared from the coordinator backup when the issue happened. However, it's still there under Frontend > Settings > Raw.

Is there a way to bring this device back? Physical access to the device is a problem because it's in the wall.

### Zigbee2MQTT version

1.33.2

### Adapter firmware version

20221226

### Adapter

Sonoff ZBdongle-P

Ra72xx, 2023-11-25:
I have the same issue repeatedly with an Skyconnect dongle. Devices of all kinds are simply no longer present in the network. In most cases this can be solved by allowing devices to join in Z2M and initiating the respective pairing procedure of the device - then all settings etc. are restored, too.
However, this and other problems with the network stability currently make me consider switching to ZHA, unfortunately.

Nezz, 2023-12-07:
I had to re-pair the device