Koenkk/zigbee2mqtt #19857: [New device support]: A9MEM1522

HSPilot2008, 2023-11-25:
### Link

https://www.se.com/za/en/product/A9MEM1522/energy-sensor-powertag-monoconnect-63a-1p+n-bottom-position/

### Database entry

{"id":2,"type":"GreenPower","ieeeAddr":"0x00000000e20702f6","nwkAddr":758,"manufId":null,"modelId":"GreenPower_165","epList":[242],"endpoints":{"242":{"epId":242,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"interviewCompleted":true,"meta":{},"lastSeen":1700899010552,"defaultSendRequestWhen":"immediate"}

### Comments

The A9MEM1522 is part of the Schneider PowerTag product family. Currently the A9MEM1570 from the same product family is supported. I tried renaming the modelID to GreenPower_165 since I am not using the A9MEM1570 but it still does not work.

HSPilot2008, 2023-12-27:
Any suggestions?

pissten, 2024-04-20:
Any updates on this?

HSPilot2008, 2024-10-18:
Any feedback if this can be added?

pissten, 2024-10-18:
Very sad that it seems to be a very difficult task to add support for these devices 🥺

closed 2025-01-10 by github-actions[bot]