Koenkk/zigbee2mqtt #19864: Wrong converter / duplicate fingerprint for different hardware

Stratagm, 2023-11-25:
### Link

https://www.aliexpress.com/item/1005005802197374.html

### Database entry

{"id":30,"type":"Router","ieeeAddr":"0xa4c13851c72d1f91","nwkAddr":50847,"manufId":4417,"manufName":"_TZE204_vmcgja59","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"P��,iP��,i","65506":56,"65508":0,"65534":0,"modelId":"TS0601","manufacturerName":"_TZE204_vmcgja59","powerSource":1,"zclVersion":3,"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-1440156621},"lastSeen":1700941570698,"defaultSendRequestWhen":"immediate"}

### Comments

Hi

I recently purchased this device, knowing there was no converter available for it in Z2M. To my surprise it showed up as supported, and reporting 24 switches. Variants of this board with anything from 4 to 24 relays are available, but in this case mine has 16. 

The board identifies itself as "_TZE204_vmcgja59", which references this converter for the 24 relay variant
https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/devices/tuya.ts#L6330

I then decided to have a crack at making my first ever converter, to match my board, which was a success I suppose.
In the process I noticed a debug log showing up for datapoints 182-189. I have no clue what to do with this as I don't seem to be missing any apparent features, and it shows up with both the existing, and my converter.
Example: "Datapoint 188 not defined for '_TZE204_vmcgja59' with value 150" 

If any further information is required, I'll be happy to provide it.

- Thanks

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    // Since a lot of TuYa devices use the same modelID, but use different datapoints
    // it's necessary to provide a fingerprint instead of a zigbeeModel
    fingerprint: [
        {
            // The model ID from: Device with modelID 'TS0601' is not supported
            // You may need to add \u0000 at the end of the name in some cases
            modelID: 'TS0601',
            // The manufacturer name from: Device with modelID 'TS0601' is not supported.
            manufacturerName: '_TZE204_vmcgja59',
        },
    ],
    model: 'TS0601_switch_16',
    vendor: 'TuYa',
    description: 'ZYXH 16 gang switch',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    //  onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
    configure: tuya.configureMagicPacket,
    exposes: [
        // Here you should put all functionality that your device exposes
        ...Array.from({length: 16}, (_, i) => tuya.exposes.switch().withEndpoint(`l${i + 1}`))
    ],
    endpoint: (device) => {
        return {'l1': 1, 'l2': 1, 'l3': 1, 'l4': 1, 'l5': 1, 'l6': 1, 'l7': 1, 'l8': 1, 'l9': 1, 'l10': 1, 'l11': 1, 'l12': 1, 'l13': 1, 'l14': 1, 'l15': 1, 'l16': 1};
    },
    meta: {
        // All datapoints go in here
        multiEndpoint: true,
        tuyaDatapoints: [
            [1, 'state_l1', tuya.valueConverter.onOff],
            [2, 'state_l2', tuya.valueConverter.onOff],
            [3, 'state_l3', tuya.valueConverter.onOff],
            [4, 'state_l4', tuya.valueConverter.onOff],
            [5, 'state_l5', tuya.valueConverter.onOff],
            [6, 'state_l6', tuya.valueConverter.onOff],
            [0x65, 'state_l7', tuya.valueConverter.onOff],
            [0x66, 'state_l8', tuya.valueConverter.onOff],
            [0x67, 'state_l9', tuya.valueConverter.onOff],
            [0x68, 'state_l10', tuya.valueConverter.onOff],
            [0x69, 'state_l11', tuya.valueConverter.onOff],
            [0x6A, 'state_l12', tuya.valueConverter.onOff],
            [0x6B, 'state_l13', tuya.valueConverter.onOff],
            [0x6C, 'state_l14', tuya.valueConverter.onOff],
            [0x6D, 'state_l15', tuya.valueConverter.onOff],
            [0x6E, 'state_l16', tuya.valueConverter.onOff],
//            [0xB6],           //Datapoint 182?
//            [0xB7],           //Datapoint 183?
//            [0xB8],           //Datapoint 184?
//            [0xB9],           //Datapoint 185?
//            [0xBA],           //Datapoint 186?
//            [0xBB],           //Datapoint 187?
//            [0xBC],           //Datapoint 188?
//            [0xBD],           //Datapoint 189?
        ],
    },
};

module.exports = definition;
```

Koenkk, 2023-11-26:
I guess the vendor users the same firmware for both the 16 and 24 relais version, from the z2m perspective it is not possible to distinguish since the fingerprint is the same.

Stratagm, 2023-11-26:
> I guess the vendor users the same firmware for both the 16 and 24 relais version, from the z2m perspective it is not possible to distinguish since the fingerprint is the same.  

Bugger, that was my guess too. 

Do you have any comments regarding the undefined datapoints?

Koenkk, 2023-11-26:
You can ignore it by adding `[188, null, null],`

Stratagm, 2023-11-26:
> You can ignore it by adding `[188, null, null],` or figure out what it is through https://www.zigbee2mqtt.io/advanced/support-new-devices/03_find_tuya_data_points.html

Thanks, I'll have to ignore it as I don't have a tuya gateway.

Stratagm, 2023-12-01:
Oh, wild guess. I assume the data points relate to the 433MHz radio on it

closed 2024-06-29 by github-actions[bot]