Koenkk/zigbee2mqtt #19865: [Wrong device]: SA-003-Zigbee

lrljoe, 2023-11-25:
### Link

https://www.zigbee2mqtt.io/devices/SA-003-Zigbee.html

### Model

SA-003-Zigbee

### Description

Smart Socket

### Vendor

eWeLink 

### Picture (link)

https://www.zigbee2mqtt.io/images/devices/SA-003-Zigbee.jpg

### Database entry

Unknown

### Notes

Believe this may relate to #19653  where the opposite was happening for someone else (their ZB-R01 was being detected as a SA-003-Zigbee)

lrljoe, 2023-11-25:
[image]

Koenkk, 2023-11-26:
Fixed in https://github.com/Koenkk/zigbee2mqtt/issues/19653

closed 2023-11-26 by Koenkk

lrljoe, 2023-11-26:
@Koenkk not fixed!  The plug is being detected as a repeater rather than a switch.

Koenkk, 2023-11-26:
I've reverted the commit.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

mrmaximas, 2023-12-12:
@Koenkk i have another eWeLink device in another housing with the another clusters and same name - SA-003-Zigbee
https://api.sprut.ai/static/media/cache/00/00/70/81/1685/55777/600x600x_image.jpeg
Please fix it is it possible.
My DB recorrd is:
{"id":55,"type":"Router","ieeeAddr":"0x00124b001d43736c","nwkAddr":49894,"manufId":0,"manufName":"eWeLink","powerSource":"Mains (single phase)","modelId":"SA-003-Zigbee","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":9,"inClusterList":[0,3,4,5,6],"outClusterList":[0],"clusters":{"genOnOff":{"attributes":{"onOff":1}},"genBasic":{"attributes":{"modelId":"SA-003-Zigbee","manufacturerName":"eWeLink","powerSource":1,"zclVersion":1,"hwVersion":1,"dateCode":"20190604-V102"}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b002a1aa3c4","endpointID":1}],"configuredReportings":[],"meta":{}}},"hwVersion":1,"dateCode":"20190604-V102","zclVersion":1,"interviewCompleted":true,"meta":{"configured":1},"lastSeen":1702378390106,"defaultSendRequestWhen":"immediate"}

Koenkk, 2023-12-12:
@mrmaximas these devices all use the same `modelId`, therefore we cannot distinguish them from each other.

mrmaximas, 2023-12-12:
> @mrmaximas these devices all use the same `modelId`, therefore we cannot distinguish them from each other.

right now found issue with this my device after restart z2m 1.34.0-dev commit: [bcf50bd0](https://github.com/Koenkk/zigbee2mqtt/commit/bcf50bd0)
Failed to configure 'Kitchen Hood', attempt 1 (Error: ConfigureReporting 0x00124b001d43736c/1 genOnOff([{"minimumReportInterval":0,"maximumReportInterval":65000,"reportableChange":1,"attribute":"onOff"}], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 49894 - 1 - 13 - 6 - 7 after 10000ms) at Timeout._onTimeout (/Users/max/zigbee2mqtt/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35) at listOnTimeout (node:internal/timers:573:17) at processTimers (node:internal/timers:514:7))

Error 2023-12-12 22:22:36 Failed to configure 'Kitchen Hood', attempt 2 (Error: ConfigureReporting 0x00124b001d43736c/1 genOnOff([{"minimumReportInterval":0,"maximumReportInterval":65000,"reportableChange":1,"attribute":"onOff"}], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 49894 - 1 - 14 - 6 - 7 after 10000ms) at Timeout._onTimeout (/Users/max/zigbee2mqtt/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35) at listOnTimeout (node:internal/timers:573:17) at processTimers (node:internal/timers:514:7))

Koenkk, 2023-12-13:
Could you check if the issue is fixed with the following [external converter](https://gist.github.com/Koenkk/a554ed0a7a7e28ed83e9e894b506f0d4): 

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

mrmaximas, 2023-12-13:
> Could you check if the issue is fixed with the following [external converter](https://gist.github.com/Koenkk/a554ed0a7a7e28ed83e9e894b506f0d4):
>
> * save this as file next to `configuration.yaml` as [...]

Error 2023-12-13 22:31:40Failed to configure 'Kitchen Hood', attempt 1 (Error: ConfigureReporting 0x00124b001d43736c/1 genOnOff([{"minimumReportInterval":0,"maximumReportInterval":65000,"reportableChange":1,"attribute":"onOff"}], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 49894 - 1 - 48 - 6 - 7 after 10000ms) at Timeout._onTimeout (/Users/max/zigbee2mqtt/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35) at listOnTimeout (node:internal/timers:573:17) at processTimers (node:internal/timers:514:7))
Info 2023-12-13 22:32:12Configuring 'Kitchen Hood'
Info 2023-12-13 22:32:12MQTT publish: topic 'zigbee2mqtt/Kitchen Hood', payload '{"action":"on","linkquality":112,"state":"ON"}'
Error 2023-12-13 22:32:35Failed to configure 'Kitchen Hood', attempt 2 (Error: ConfigureReporting 0x00124b001d43736c/1 genOnOff([{"minimumReportInterval":0,"maximumReportInterval":65000,"reportableChange":1,"attribute":"onOff"}], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 49894 - 1 - 68 - 6 - 7 after 10000ms) at Timeout._onTimeout (/Users/max/zigbee2mqtt/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35) at listOnTimeout (node:internal/timers:573:17) at processTimers (node:internal/timers:514:7))

mrmaximas, 2023-12-13:
W/o ext converter 
Error 2023-12-13 22:37:05Failed to configure 'Kitchen Hood', attempt 1 (Error: Bind 0x00124b001d43736c/1 genOnOff from '0x00124b002a1aa3c4/1' failed (AREQ - ZDO - bindRsp after 10000ms) at Timeout._onTimeout (/Users/max/zigbee2mqtt/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35) at listOnTimeout (node:internal/timers:573:17) at processTimers (node:internal/timers:514:7))

Koenkk, 2023-12-13:
fixed!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.