Koenkk/zigbee2mqtt #19868: [New device support]: add feature to Bosch BSD-2 smoke alarm detector to use the inbuilt alarm siren for other purposes

Mikkooooo, 2023-11-26:
### Link

https://www.zigbee2mqtt.io/devices/BSD-2.html

### Database entry

{"id":9,"type":"EndDevice","ieeeAddr":"0x9035eafffe123b2f","nwkAddr":10333,"manufId":4617,"manufName":"Bosch","powerSource":"Battery","modelId":"RBSH-SD-ZB-EU","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,1,3,32,1280,2821],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"RBSH-SD-ZB-EU","manufacturerName":"Bosch","powerSource":3,"zclVersion":8,"appVersion":1,"stackVersion":6,"hwVersion":1,"dateCode":"20220222","swBuildId":"1.10"}},"ssIasZone":{"attributes":{"iasCieAddr":"0xe0798dfffed1cd50","zoneState":1}},"genPollCtrl":{"attributes":{"checkinInterval":6480}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":200,"batteryAlarmState":0}}},"binds":[{"cluster":32,"type":"endpoint","deviceIeeeAddress":"0xe0798dfffed1cd50","endpointID":1},{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0xe0798dfffed1cd50","endpointID":1},{"cluster":64684,"type":"endpoint","deviceIeeeAddress":"0xe0798dfffed1cd50","endpointID":1}],"configuredReportings":[{"cluster":1,"attrId":33,"minRepIntval":3600,"maxRepIntval":62000,"repChange":0,"manufacturerCode":null}],"meta":{}}},"appVersion":1,"stackVersion":6,"hwVersion":1,"dateCode":"20220222","swBuildId":"1.10","zclVersion":8,"interviewCompleted":true,"meta":{"configured":1394888596},"lastSeen":1700983962337,"defaultSendRequestWhen":"active","checkinInterval":1620} ﻿

### Comments

This smoke detector already works fine with Z2M 1.30.3 but does not expose the alarm as a switch.
Via the Bosch gateway it is possible to e.g. trigger the smoke alarm when a door is opened or similar triggers are fired.

Can this attribute be made available as a switch also in Z2M?

This has already been mentioned in #17229   and  #18509   .

vjonas, 2024-01-15:
Merged in master. You can pull the the branch or wait for the docker release.
Can be closed.

Mikkooooo, 2024-01-15:
Thanks a lot vjonas. Do I only have to replace the content of the src/devices/bosch.ts    ?

Regards, Mikkooooo

vjonas, 2024-01-16:
> Thanks a lot vjonas. Do I only have to replace the content of the src/devices/bosch.ts ?
> 
> Regards, Mikkooooo

I'm sorry, you should use the dev branch to test it out. Master gets released a bit later.

vjonas, 2024-02-05:
> How much DB does the device create when triggerd for the alarm?

Smoke or intruder siren?

thecobra666, 2024-02-05:
> > How much DB does the device create when triggerd for the alarm?
> 
> Smoke or intruder siren?

Intruder :)

vjonas, 2024-02-06:
> > > How much DB does the device create when triggerd for the alarm?
> > 
> > 
> > Smoke or intruder siren?
> 
> Intruder :)

A quick local measurement ranged between 100-110 dB for the intruder alarm.

thecobra666, 2024-02-07:
> > > > How much DB does the device create when triggerd for the alarm?
> > > 
> > > 
> > > Smoke or intruder siren?
> > 
> > Intruder :)
> 
> A quick local measurement ranged between 100-110 dB for the intruder alarm.

How about false positives on the fire alarm? Are they stable or very sensitive ?

vjonas, 2024-02-07:
> > > > > How much DB does the device create when triggerd for the alarm?
> > > > 
> > > > 
> > > > Smoke or intruder siren?
> > > 
> > > 
> > > Intruder :)
> > 
> > 
> > A quick local measurement ranged between 100-110 dB for the intruder alarm.
> 
> How about false positives on the fire alarm? Are they stable or very sensitive ?

One was chilling in the garage and later removed due to too much false positives. 
1 in a tall room very high on the ceiling and it send out 2x false positives throughout 1 year.

Not sure but it might be dust sensitive? The other 4 detectors in the house have not triggered falsely. There is even 1 in the kitchen that has been quite up until now.

Pferdebockwurst, 2024-06-15:
I think we can close this too as fixed.

closed 2025-01-10 by github-actions[bot]