Koenkk/zigbee2mqtt #19869: Startup crash on Alpine (APK)

SoftMadHatter, 2023-11-26:
### What happened?

After upgrading to latest (1.33.2-r0) apk on alpine (3.19) zigbee2mqtt is not starting anymore.

(Updated with basic apk commands) 

Startup crached with error :

Error: Error relocating /usr/lib/zigbee2mqtt/node_modules/unix-dgram/build/Release/unix_dgram.node: _ZN2v88internal20CanHaveInternalFieldEi: symbol not found
    at Object.Module._extensions..node (node:internal/modules/cjs/loader:1327:18)
    at Module.load (node:internal/modules/cjs/loader:1091:32)
    at Function.Module._load (node:internal/modules/cjs/loader:938:12)
    at Module.require (node:internal/modules/cjs/loader:1115:19)
    at require (node:internal/modules/helpers:130:18)
    at bindings (/usr/lib/zigbee2mqtt/node_modules/bindings/bindings.js:112:48)
    at Object.<anonymous> (/usr/lib/zigbee2mqtt/node_modules/unix-dgram/lib/unix_dgram.js:4:34)
    at Module._compile (node:internal/modules/cjs/loader:1241:14)
    at Object.Module._extensions..js (node:internal/modules/cjs/loader:1295:10)
    at Module.load (node:internal/modules/cjs/loader:1091:32)

Any Idea ?

### Zigbee2MQTT version

1.33.2

### Adapter firmware version

/

### Adapter

/

jckriplr, 2023-11-27:
Actually is a very huge problem, since this update all Zigbee devices are unavailable.
I didn't find a way to restore previous version from apk package manager.. Someone knows a way to do this or a workaround for the problem?

LesterNJ, 2023-11-27:
I just experienced the same issue. I don't have a workaround for this.

Koenkk, 2023-11-28:
Could you contact the maintainer of this package? This is a third party zigbee2mqtt distribution.

tweisberger, 2023-11-28:
@jckriplr I believe that you are the maintainer of this package on Alpine. Are you having similar results? 

@SoftMadHatter I am also having this issue with 1.33.2-r0. I was able to recover a backup from before the update, and can confirm that z2m 1.33.1-r0 works as expected.

tweisberger, 2023-11-28:
@jirutka I believe that you are the maintainer of this package on Alpine. Are you having similar results?

@SoftMadHatter I am also having this issue with 1.33.2-r0. I was able to recover a backup from before the update, and can confirm that z2m 1.33.1-r0 works as expected.

SoftMadHatter, 2023-11-28:
Same for me reverted to an older backup in 1.33.1-r0 and all is ok.

@jirutka yes all up to date and no mix stable-edge

jckriplr, 2023-11-28:
> Yes, I am. zigbee2mqtt 1.33.2-r0 is only in Alpine Edge (unstable), v3.19 hasn’t been released yet. Have you installed all packages from the Edge repositories (and `apk upgrade -sal` returns no packages for upgrade), or are you mixing stable and edge? If the latter, then the problem is on your side – you should never mix stable and edge.

Personally, I use the tteck's ( https://github.com/tteck/Proxmox) script for creating lxc on proxmox, so I don 't know what is the environment used (mixed or not).. In the case I'm in the mixed state, are a way to resolve this issue moving on the edge repos?

jirutka, 2023-11-28:
It seems that the package is really broken in Edge, I’ll fix it soon.

tweisberger, 2023-11-28:
@jirutka, now that I have you here, I found that z2m on Alpine point to a configuration in `~/.z2m/`
```
alpine:~# zigbee2mqtt 
Error: ENOENT: no such file or directory, open '/root/.z2m/configuration.yaml'
```
but the default configuration files are placed in `/etc/zigbee2mqtt/`. This seems to be handled in Debian, but in Alpine. It requires the user (or an install script like tteck's) to create a link to `/etc/zigbee2mqtt` (or move the config files to) `~/.z2m`. It seems like you are trying to handle this by making simlinks in `/usr/lib/zigbee2mqtt/`, but it doesn't seem like the trick.

jckriplr, 2023-11-28:
> It seems that the package is really broken in Edge, I’ll fix it soon.

Thank you very much! I really apreciate your work!

mikaelsundin, 2023-12-01:
I also spotted the problem yesterday.  

I run zigbee2mqtt in alpine LXC container and ussually i do a backup before upgrading, so in this instance it was easy to do a server rollback.

jirutka, 2023-12-02:
> After upgrading to latest (1.33.2-r0) apk on alpine (3.19) zigbee2mqtt is not starting anymore.

I’ve upgraded zigbee2mqtt aport to 1.34.0. This problem should be fixed now.

> The default configuration files are placed in /etc/zigbee2mqtt/.
> It requires the user (or an install script like tteck's) to create a link to /etc/zigbee2mqtt (or move the config files to) ~/.z2m.

It’s supposed to be started by an OpenRC service which sets the environment variable `ZIGBEE2MQTT_DATA=/var/lib/zigbee2mqtt`. If you don’t use OpenRC, it’s up to you to set this variable. I don’t patch the default paths in zigbee2mqtt. If I did, other users might complain that it behaves differently than upstream.

jckriplr, 2023-12-02:
> > After upgrading to latest (1.33.2-r0) apk on alpine (3.19) zigbee2mqtt is not starting anymore.
> 
> I’ve upgraded zigbee2mqtt aport to 1.34.0. This problem should be fixed now.

I can confirm that now the issue is solved! Thank you very much!

partikus, 2023-12-03:
JFYI, before fixes from @jirutka, the way of switching back to `v3.18` was

```patch
--- /etc/apk/repositories
+++ /etc/apk/repositories
@@ -1,3 +1,2 @@
-https://dl-cdn.alpinelinux.org/alpine/edge/main
-https://dl-cdn.alpinelinux.org/alpine/edge/community
-https://dl-cdn.alpinelinux.org/alpine/edge/testing
+https://dl-cdn.alpinelinux.org/alpine/v3.18/main
+https://dl-cdn.alpinelinux.org/alpine/v3.18/community
```

```bash
apk upgrade -sal
apk del nodejs zigbee2mqtt
apk add nodejs zigbee2mqtt
```