Koenkk/zigbee2mqtt #19870: Moes Carbon Monoxide Alarm - extension?

mtausk, 2023-11-26:
### Link

https://www.zigbee2mqtt.io/devices/ZC-HM.html

### Database entry

{"id":31,"type":"EndDevice","ieeeAddr":"0xa4c1387b5a634c6e","nwkAddr":28394,"manufId":4417,"epList":[1],"endpoints":{"1":{"epId":1,"inClusterList":[],"outClusterList":[],"clusters":{"genBasic":{"attributes":{"65506":56,"65508":1,"appVersion":72}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":72,"interviewCompleted":false,"meta":{},"lastSeen":1700990045511,"defaultSendRequestWhen":"immediate"}

### Comments

Hi,
I've just bought Moes Carbon Monoxide Alarm and it does not pair (I have updated zigbee2mqtt prior pairing to current version today).
I suspect the configuration needs to be extended to accept this version/modification?

Is that possible to do easily without further investigation?

info  2023-11-26 10:13:34: Device '0xa4c1387b5a634c6e' joined
info  2023-11-26 10:13:34: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c1387b5a634c6e","ieee_address":"0xa4c13
87b5a634c6e"},"type":"device_joined"}'
info  2023-11-26 10:13:34: Starting interview of '0xa4c1387b5a634c6e'
info  2023-11-26 10:13:35: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c1387b5a634c6e","ieee_address":"0xa4c13
87b5a634c6e","status":"started"},"type":"device_interview"}'
info  2023-11-26 10:13:35: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c1387b5a634c6e","ieee_address":"0xa4c13
87b5a634c6e"},"type":"device_announce"}'
error 2023-11-26 10:13:45: Failed to interview '0xa4c1387b5a634c6e', device has not successfully been paired
info  2023-11-26 10:13:46: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c1387b5a634c6e","ieee_address":"0xa4c13
87b5a634c6e","status":"failed"},"type":"device_interview"}'
warn  2023-11-26 10:13:46: Received message from unsupported device with Zigbee model 'undefined' and manufacturer name 'undefined'
warn  2023-11-26 10:13:46: Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html

Pls. add information to the doc. that to enter pairing mode the only button on the top needs to be pushed for at least 5 secs.

Thank you.

danieledwardgeorgehitchcock, 2023-11-26:
It looks like the device is failing the initial interview rather than being unsupported.. Try moving the Moes device closer to your co-ordinator during the interview stage...

mtausk, 2023-12-01:
You were correct. After few more retries, it finally got paired. Thanks.

closed 2023-12-02 by Koenkk