Koenkk/zigbee2mqtt #19871: [Feature request]: Split data on there own topics

Pimmetje, 2023-11-26:
### Is your feature request related to a problem? Please describe

I like to use a external tool to record some parts of my MQTT traffic using a path like some/topic/+/temperature. But currently z2mqtt is only giving out json. It would be nice if it also has the option to split it over topics

### Describe the solution you'd like

publish every value on it's own topic /temperature /link /battery for every device when the value changes.

### Describe alternatives you've considered

Publish all values every time a value comes in

### Additional context

This makes it easier for devices to capture specific values without having to parse the json. And only get values when the data there interested in change.

Koenkk, 2023-11-26:
See the [`advanced.output`](https://www.zigbee2mqtt.io/guide/configuration/mqtt.html#mqtt-behaviour) option.

closed 2023-11-26 by Koenkk

Pimmetje, 2023-11-26:
@Koenkk Thanks sorry i have overlooked that :)