Koenkk/zigbee2mqtt #19876: [Wrong device]: TS0003 _TZ3000_ouwfc1qj

jocamane, 2023-11-26:
### Link

https://pt.aliexpress.com/item/1005002950510187.html?spm=a2g0o.order_list.order_list_main.15.6c1e5e5bpEPgqO&gatewayAdapt=glo2bra

### Model

TS0003 _TZ3000_ouwfc1qj

### Description

After update 1.33.2 z2m the picture of device change to TUYA logo

### Vendor

tuya

### Picture (link)

https://ae01.alicdn.com/kf/H8f3d6abafe4f4289947132d745903b26Z.jpg

### Database entry

{"id":24,"type":"Router","ieeeAddr":"0xa4c1389e45517183","nwkAddr":41698,"manufId":4417,"manufName":"_TZ3000_ouwfc1qj","powerSource":"Mains (single phase)","modelId":"TS0003","epList":[1,2,3,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[3,4,5,6,57344,57345,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"���,\u0012","65506":54,"65508":0,"65534":0,"modelId":"TS0003","manufacturerName":"_TZ3000_ouwfc1qj","powerSource":1,"zclVersion":3,"appVersion":82,"stackVersion":0,"hwVersion":1,"dateCode":""}},"manuSpecificBosch":{"attributes":{"53251":"AAAAAgAABAAA"}},"genOnOff":{"attributes":{"32768":0,"onOff":1,"onTime":0,"offWaitTime":0,"tuyaBacklightMode":1,"moesStartUpOnOff":2,"tuyaBacklightSwitch":1}},"manuSpecificTuya_3":{"attributes":{"powerOnBehavior":2,"switchType":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00212effff07afaa","endpointID":1}],"configuredReportings":[],"meta":{}},"2":{"profId":260,"epId":2,"devId":256,"inClusterList":[4,5,6,57345],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"onOff":0,"onTime":0,"offWaitTime":0}},"manuSpecificTuya_3":{"attributes":{"powerOnBehavior":2}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00212effff07afaa","endpointID":1}],"configuredReportings":[],"meta":{}},"3":{"profId":260,"epId":3,"devId":256,"inClusterList":[4,5,6,57345],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"onOff":0,"onTime":0,"offWaitTime":0}},"manuSpecificTuya_3":{"attributes":{"powerOnBehavior":2}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00212effff07afaa","endpointID":1}],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":82,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-150731130},"lastSeen":1701022523163,"defaultSendRequestWhen":"immediate"}

jocamane, 2024-01-20:
hi, it is possible change this image ?

[image]

jocamane, 2024-02-11:
someone here ?

jocamane, 2024-03-30:
After update 1.36.0-1 the picture of device mantain TUYA logo. please correct

Koenkk, 2024-03-31:
Will be fixed in the 1 April release.

closed 2024-03-31 by Koenkk

jocamane, 2024-03-31:
thanks @Koenkk