Koenkk/zigbee2mqtt #19884: [New device support]: Aqara WS-USC01(1gang no neutral, single rocker) request supports Power outage memory. Please see the details below for my test results

a29559089, 2023-11-27:
### Link

https://www.zigbee2mqtt.io/devices/WS-USC01.html

### Database entry

{"id":14,"type":"EndDevice","ieeeAddr":"0x54ef4410006193a4","nwkAddr":15534,"manufId":4447,"manufName":"LUMI","powerSource":"Mains (single phase)","modelId":"lumi.switch.b1laus01","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[0,2,3,4,5,6,9],"outClusterList":[10,25],"clusters":{"genBasic":{"attributes":{"dateCode":"06-09-2020","appVersion":32}},"aqaraOpple":{"attributes":{"4":1,"240":1,"247":{"type":"Buffer","data":[100,16,0,3,40,24,5,33,5,0,154,32,0,8,33,32,1,10,33,173,132,11,32,0]},"512":1,"513":0}},"genOnOff":{"attributes":{"245":37531136,"onOff":0}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":32,"stackVersion":2,"hwVersion":1,"dateCode":"06-09-2020","zclVersion":3,"interviewCompleted":true,"meta":{"configured":503270415},"lastSeen":1701080297161,"defaultSendRequestWhen":"immediate"}

### Comments

Because WS-USC01 is not supported for Power outage memory in z2m, but WS-USC02 does support Power outage memory. Therefore, I created an External converter by modifying the model of WS-USC01 to WS-USC02 , tozigbee adding tz.xiaomi_switch_power_outage_memory , and including e.power_outage_memory() under exposes. Currently, testing indicates that Power outage memory is functioning correctly. If possible, I would appreciate it if you could follow the External converter config below to integrate the Power outage memory feature for WS-USC01. Thank you.
[image]
[image]
[image]
[image]

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
 
const definition = {
        zigbeeModel: ['lumi.switch.b1laus01'],
        model: 'WS-USC02',
        vendor: 'Xiaomi',
        description: 'Aqara smart wall switch (no neutral, single rocker)',
        fromZigbee: [fz.on_off, fz.xiaomi_multistate_action, fz.aqara_opple],
        toZigbee: [tz.on_off, tz.xiaomi_switch_operation_mode_opple,
            tz.xiaomi_flip_indicator_light, tz.aqara_switch_mode_switch, tz.xiaomi_switch_power_outage_memory],
        exposes: [e.switch(), e.action(['single', 'double']),
		e.flip_indicator_light(),e.power_outage_memory(),
		e.enum('operation_mode', ea.ALL, ['control_relay', 'decoupled']).withDescription('Decoupled mode'),
        e.enum('mode_switch', ea.ALL, ['anti_flicker_mode', 'quick_mode'])
                .withDescription('Anti flicker mode can be used to solve blinking issues of some lights.' +
                    'Quick mode makes the device respond faster.'),
            e.power_outage_count(), 
            e.device_temperature().withAccess(ea.STATE)],
       // onEvent: preventReset,
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint1 = device.getEndpoint(1);
            // set "event" mode
            await endpoint1.write('aqaraOpple', {'mode': 1}, {manufacturerCode: 0x115f, disableResponse: true});
        },       
};
module.exports = definition;
```

Koenkk, 2023-12-05:
@caryyu0306 please create a pull request to the zigbee-herdsman-converters repo

Koenkk, 2023-12-06:
Integrated the changes!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2023-12-06 by Koenkk

caryyu0306, 2023-12-07:
> Integrated the changes!
> 
> 
> 
> Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

 Thx love you