Koenkk/zigbee2mqtt #19885: Aqara Xiaomi Switches not reporting temp, power, energy, power outage count or on/off state

RN-Say, 2023-11-27:
### What happened?

Most Xiaomi WS-EUKO3 and WS-EUK04 switches do not report states to home assistant. Problem not solved by removing/re-pairing.

There are several other bug reports that may be related: 17477, 12365, 17306, 16910, 19824, 19478

### What did you expect to happen?

After pairing, I expect the switches to report temperature, power, energy, power outage count and on/off states to home assistant and Z2M

### How to reproduce it (minimal and precise)

Pair device with Zigbee2mqtt
Open newly paired device in Z2M>exposes
See Null, Null, Null, Null, Null
Open device page in HA for confirmation
See Unknown, Unknown, Unknown, Unknown, Unknown
Push physical button on device
See nothing happen to the controls in home assistant or Z2M
Open HA Dev Tools>Events and listen for state changes
Push physical button in device
See nothing

### Zigbee2MQTT version

1.33.2

### Adapter firmware version

7.3.1.0 build 176 (Same behavior with 6.10.3)

### Adapter

Sonoff Zigbee Dongle E 3.0 v2

### Debug log

Nothing happens in Z2M debug log when the physical button is pushed.

The following error does occur if power is queried from the UI:

[Debug 2023-11-27 045339Received MQT.txt](https://github.com/Koenkk/zigbee2mqtt/files/13472870/Debug.2023-11-27.045339Received.MQT.txt)

rjaros87, 2024-03-03:
I've got the same problem. Adapter Skyconnect EZSP v12 version 7.3.2.0 build 212; Z2M: 1.35.3

mojiro, 2024-03-04:
Similar issue here.. but with two different incidents

Back in November 2023, I was having troubles pairing a WS-EUK04.

**Setup of November 2023**
- Device: Aqara WS-EUK04 (double rocker with neutral)
- Device Firmware: 11-10-2021
- Adapter: Sonoff Zigbee Dongle E 3.0 v2
- Adapter Firmware: 6.10.3
- Adapter Driver: ezsp
- Zigbee2MQTT Version: I suppose it was, v1.33.2-1

My problem on that day, was solved by upgrading the Adapter's Firmware to 7.3.1.0 build 176.

Today, I installed another Aqara WS-EUK04, with same Firmware (11-10-2021) and once it gets paired, I have the same exact troubles.

**Setup of March 2024**
- Device: Aqara WS-EUK04 (double rocker with neutral)
- Device Firmware: 11-10-2021
- Adapter: Sonoff Zigbee Dongle E 3.0 v2
- Adapter Firmware: 7.3.1.0 build 176 & 7.4.1.0 build 0 (tried with both)
- Adapter Driver: ezsp & ember (tried with both)
- Zigbee2MQTT Version: v1.36.0-1

As you see, I have tried several combinations, apart of downgrading Zigbee2MQTT using an old home-assistant backup.

mojiro, 2024-03-05:
## Similar issues with decoupled mode (this issue seems to be fixed)
#12365 - Xiaomi Aqara WS-EUK03 not reporting Action in Decoupled Mode
#17305 - Aqara WS-EUK04 not emitting actions in decoupled mode
#17477 - Xiaomi/Aqara QBKG26LM doesn't report action state in operation_mode: decoupled
#19478 - Aqara WS-EUK04 no actions in decoupled mode

## Similar issues regardless mode
#11507 - Xiaomi (aquara) WS-EUK03 not updating state in z2m
#17306 - Pressing physical button on Xiaomi WS-USC04 does not update state
#19824 - Aqara WS-USC04 not reporting energy or power
#20575 - Conbee3 and Aqara WS-EUK03/WS-EUK04 - missing attributes
#21679 - Aqara wall switch H1 (WS-EUK04) - not syncing state, not showing power measurement anymore

[Aqara Wall Switch NOT working in Decoupled Mode](https://community.home-assistant.io/t/aqara-wall-switch-not-working-in-decoupled-mode/537582)

## Similar issues on other devices
#16910 - Aqara roller shade companion E1 (Xiaomi ZNJLBL01LM) state reporting

nepofigist, 2024-03-22:
Looks like I've found a solution. Try this option →

[image]

RN-Say, 2024-03-22:
> Looks like I've found a solution. Try this option →
> 
> <img alt="Снимок экрана 2024-03-22 в 10 16 26" width="1626" src="https://private-user-images.githubusercontent.com/83584820/315937424-7fa524eb-18c5-40f7-a55b-c36ef535ab60.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTExMDQ0MDQsIm5iZiI6MTcxMTEwNDEwNCwicGF0aCI6Ii84MzU4NDgyMC8zMTU5Mzc0MjQtN2ZhNTI0ZWItMThjNS00MGY3LWE1NWItYzM2ZWY1MzVhYjYwLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAzMjIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMzIyVDEwNDE0NFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTkzZDRhYjZlMjhhY2VlMDg2ZjEzZjE5MzI3MzliNmYxNTM5M2JmZjk2ODE2NWJjNWExOGZkYmU0MTkxYWRhM2QmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.QL--KcYc3hqyOdW9hRlqNM0E6oFgIiKx2_5j_QsmyBs">

This is a workaround that fixes only one state of the many states that aren't being reported for one switch(OnOff). This is not a solution to the broader issue of some (but not all) Aqara devices not properly reporting their states.  For the ones that somehow work, this workaround isn't necessary. Unfortunately for me, my first Aqara switch worked flawlessly and I believed they would be a good solution for my needs. Only when I ordered and  installed the rest of the switches in my house did I learn that the majority of them wouldn't configure correctly. I am using the above workaround to get minimal functionality but I don't fully understand the impact of this on my overall Zigbee network function. My concern (and time investment) multiplies if I am to implement something similar for every state of my 20+ switches.

closed 2024-10-20 by github-actions[bot]

Simon-CR, 2024-10-20:
This is still an issue.