Koenkk/zigbee2mqtt #19898: [New device support]:  SIXWGH Zigbee AU Smart Plug with Power Monitoring

hndrk, 2023-11-28:
### Link

https://www.aliexpress.com/item/1005002605377621.html

### Database entry

{"id":5,"type":"Router","ieeeAddr":"0xa4c13828e299aeba","nwkAddr":16915,"manufId":4107,"manufName":"_TZ3000_0zfrhq4i","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,3,4,5,6,10,57344,1794,2820,4096,6280],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"hwVersion":0,"manufacturerName":"_TZ3000_0zfrhq4i","zclVersion":3,"appVersion":112,"modelId":"TS011F","powerSource":1}},"genOnOff":{"attributes":{"32768":0,"onOff":1,"tuyaBacklightMode":1,"moesStartUpOnOff":0,"onTime":0,"offWaitTime":0}},"haElectricalMeasurement":{"attributes":{"rmsVoltage":236,"rmsCurrent":110,"activePower":18}},"seMetering":{"attributes":{"currentSummDelivered":[0,30]}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0029e7307d","endpointID":1},{"cluster":2820,"type":"endpoint","deviceIeeeAddress":"0x00124b0029e7307d","endpointID":1},{"cluster":1794,"type":"endpoint","deviceIeeeAddress":"0x00124b0029e7307d","endpointID":1}],"configuredReportings":[{"cluster":2820,"attrId":1285,"minRepIntval":5,"maxRepIntval":3600,"repChange":5,"manufacturerCode":null},{"cluster":2820,"attrId":1288,"minRepIntval":5,"maxRepIntval":3600,"repChange":50,"manufacturerCode":null}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":112,"stackVersion":2,"hwVersion":0,"swBuildId":"0122052017","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1701135818630,"defaultSendRequestWhen":"immediate"}

### Comments

I note that last year that this or a similar device from the manufacturer _TZ3000_0zfrhq4i was added to the dev branch, however I'm experiencing a few issues with these plugs:

- The current value is reported as Amps whereas the value given is in milliamps (off by a factor of 1000). 
- The energy consumption is given as null.
- Cannot check for OTA updates `"Failed to check if update available for '0xa4c13828e299aeba' (No image available for imageType '524')"`
- Always encounter configuration error on restart `Failed to configure '0xa4c13828e299aeba', attempt 4 (Error: ConfigureReporting 0xa4c13828e299aeba/1 haElectricalMeasurement([{"attribute":"activePower","minimumReportInterval":5,"maximumReportInterval":3600,"reportableChange":10}], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'INVALID_DATA_TYPE') at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:421:28) at Endpoint.configureReporting (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:762:22) at Object.activePower (/app/node_modules/zigbee-herdsman-converters/src/lib/reporting.ts:204:5) at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/devices/tuya.ts:3045:13) at Configure.configure (/app/lib/extension/configure.ts:117:13) at Configure.onMQTTMessage (/app/lib/extension/configure.ts:55:21))`

ajiau, 2023-11-28:
I've got the exact same problem.

jackhenderson101, 2023-11-29:
Also have the same problem. I found that the "failed to configure" went away when i switched to the dev branch, so maybe this was fixed recently.
Still have the other issues, including wrong scale for current and null energy.
I also don't think that the current is reported in milliamps, as that doesn't match with the other values.
For example, with a voltage of 232V and a power of 27W, the actual current should be 0.116A, but it is reported as 233 A.

ajiau, 2023-11-29:
Yup I have all those issues too. But current looks to be close to out by
1000.

{
    "child_lock": "UNLOCK",
    "current": 305,
    "indicator_mode": "off/on",
    "linkquality": 15,
    "power": 68,
    "power_outage_memory": "on",
    "state": "ON",
    "voltage": 232
}

ajiau, 2023-11-30:
If a dev can point me in the right direction to help, I'd be happy to.  I submit debug logs etc,?

ajiau, 2023-12-04:
Weirdly, this seems to be fixed in 1.34.0!

jimbdrafting, 2024-05-06:
This is still a problem for me... Not supported.
Error 2024-05-03 10:29:40Interview failed for '0xa4c138de1d4b980e with error 'Error: Interview failed because can not get node descriptor ('0xa4c138de1d4b980e')'
Error 2024-05-03 10:29:40Failed to interview 'Power meter socket Z5', device has not successfully been paired
Info 2024-05-03 10:29:53Interview for '0xa4c138de1d4b980e' started
Info 2024-05-03 10:29:53Starting interview of 'Power meter socket Z5'
Error 2024-05-03 10:32:11Interview failed for '0xa4c138de1d4b980e with error 'Error: Interview failed because can not get node descriptor ('0xa4c138de1d4b980e')'
Error 2024-05-03 10:32:11Failed to interview 'Power meter socket Z5', device has not successfully been paired

eerison, 2024-07-07:
Do you know if the basic functionality is working?

Turn on and off via z2m and fisical switch ?

closed 2025-01-10 by github-actions[bot]

ajiau, 2025-01-10:
unsubscribe