Koenkk/zigbee2mqtt #19902: OTA update broke support for Xiaomi SSM-U01 - UNSUPPORTED_ATTRIBUTE

Kugelfang666, 2023-11-28:
### What happened?

I‘m using several Xiaomi SSM-U01 ZigBee relays. 

Yesterday I updated my devices from DEC 30 2021 to 20231025 (screenshot attached)

For the updated devices I‘m now finding this error in the logs. 

Would be great to know how to resolve / mitigate this issue

[image]

### Zigbee2MQTT version

1.33.2

### Adapter firmware version

20230507

### Adapter

TubesZB Ethernet version

### Debug log

`Failed to configure 'Garage Deckenlampen', attempt 1 (Error: ConfigureReporting XXREDACTEDXX/1 genDeviceTempCfg([{"attribute":"currentTemperature","minimumReportInterval":300,"maximumReportInterval":3600,"reportableChange":1}], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE') at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:421:28) at Endpoint.configureReporting (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:762:22) at processTicksAndRejections (node:internal/process/task_queues:96:5) at Object.deviceTemperature (/app/node_modules/zigbee-herdsman-converters/src/lib/reporting.ts:106:5) at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/devices/xiaomi.ts:2740:13) at Configure.configure (/app/lib/extension/configure.ts:117:13) at Immediate.<anonymous> (/app/lib/extension/configure.ts:69:17))`

erinaceinae, 2023-11-28:
There might be a problem with the firmware. I updated it today, the same version and wall switch stopped working.

BlastyCZ, 2023-11-29:
I have the same problem with voltage and current but power is working correctly...  My wall switch is working fine.
What is your wiring diagram? I use this one
[image]

erinaceinae, 2023-11-29:
@BlastyCZ I have the same wiring. I tried manualy short S0 and S1 and nothing happen. 
Probably there are different HW versions, maybe firmware can detect the HW version by checking some of the GPIOs. Maybe this firmware didn't detect HW revision.

brhahlen, 2023-12-01:
I have the same issue here. Would be good to know how to revert to the previous version.

```
zigbee2mqtt  | Zigbee2MQTT:error 2023-12-01 15:02:39: Failed to configure 'switch_bathroom_fan', attempt 2 (Error: ConfigureReporting 0x54ef44100026e5de/1 genDeviceTempCfg([{"attribute":"currentTemperature","minimumReportInterval":300,"maximumReportInterval":3600,"reportableChange":1}], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')
zigbee2mqtt  |     at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:421:28)
zigbee2mqtt  |     at Endpoint.configureReporting (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:762:22)
zigbee2mqtt  |     at Object.deviceTemperature (/app/node_modules/zigbee-herdsman-converters/src/lib/reporting.ts:106:5)
zigbee2mqtt  |     at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/devices/xiaomi.ts:2884:13)
zigbee2mqtt  |     at Configure.configure (/app/lib/extension/configure.ts:117:13))
```

Otnow, 2023-12-02:
It's a pity that this firmware version turned out to be problematic.

[By analogy with another device](https://github.com/Koenkk/zigbee2mqtt/issues/13903#issuecomment-1244720072), I can suggest trying to roll back the firmware version to `0.0.0.0023` using modified firmware (_only the version number in the header has been changed to 99_), but at your own risk, because device may become bricked:

[20220121143603_lumi.switch.n0agl1_0.0.0.0023_20211230_0717A1_modded_to_v99.ota.zip](https://github.com/Koenkk/zigbee2mqtt/files/13532730/20220121143603_lumi.switch.n0agl1_0.0.0.0023_20211230_0717A1_modded_to_v99.ota.zip)

w-marco, 2023-12-06:
FWIW (i'm using ZHA though), for me this firmware version works on my variant of the Module. 

Voltage is reported as 23V instead of 230V (it was just at 0 with the version before forever) but power works now which always reported 0 before.

It didn't break anything and on/off also still works.

rjaros87, 2024-03-09:
Does anyone have 0.0.0.0021 ota firmware file? :)

jrhansen, 2024-07-28:
So... did anyone have the courage to install the newly found update (I don't know how to see the version)?

Kugelfang666, 2024-07-28:
After Rolling back I never considered looking into this again

jrhansen, 2024-08-23:
FWIW: I have now successfully updated nine devices previously running firmwares:

- Aug 8 2020
- Dec 30 2021
- 20231025
- Two units where the firmware version string is empty in Z2M (?). The units were purchased about 6 weeks ago, so they are probably pretty new.

closed 2025-01-10 by github-actions[bot]