Koenkk/zigbee2mqtt #19904: frytur not updating

drumstick77, 2023-11-28:
### What happened?

Since the last update, frytur update kept trying for days..everytime I restarted HA, it will then re attempt the update again, but never completed it.

### Zigbee2MQTT version

1.33.2

### Adapter firmware version

20211217

### Adapter

frytur

drumstick77, 2023-11-28:
[image]

Koenkk, 2023-11-28:
This will be fixed in the 1 December release.

closed 2023-11-28 by Koenkk

drumstick77, 2023-11-28:
Ah. Many thanks! I thought it's me only. Cheers

drumstick77, 2023-12-01:
Hi, has it been released yet? Sorry, I'm in UK, not sure the timezone of when this should be released today. Cheers

drumstick77, 2023-12-01:
Hi, received updates but it is still not updating fyrtur

drumstick77, 2023-12-01:
restarted HA, restarted host, restarted just the add-on and still getting this 
[image]

Koenkk, 2023-12-04:
Could you try to force-remove the device and re-pair it?