Koenkk/zigbee2mqtt #19910: Smoke detector "TS0205 _TZ3210_up3pngle" go to offline

jocamane, 2023-11-28:
### What happened?

The smoke detector "TS0205 _TZ3210_up3pngle" apparently pairs well but after a few hours (+- 24h) it goes offline.
If you press the test button it "wakes up".
Remaining battery in % also shows N/A%

Zigbee2MQTT version: 1.33.2 commit: unknown
Coordinator type: ConBee2/RaspBee2
Coordinator revision: 0x26780700
Frontend version: 0.6.142
Zigbee-herdsman-converters version: 15.106.0
Zigbee-herdsman version: 0.21.0

Core: 2023.11.3
Supervisor: 2023.11.3
Operating System: 11.1
Frontend: 20231030.2

If you need any more information, please let me know.

Thanks for the correction please.

### What did you expect to happen?

That the smoke detector did not go offline and that it had values on the battery

### How to reproduce it (minimal and precise)

the smoke detector is paired and everything works except the battery %. After about 24 hours it goes offline. During this time all other smoke sensors have "last seen" for a short time only this model this parameter is not changed

### Zigbee2MQTT version

1.33.2 

### Adapter firmware version

0x26780700

### Adapter

ConBee2

jocamane, 2023-11-28:
[image]

jocamane, 2023-11-28:
[image]

jocamane, 2023-11-30:
@Koenkk Can you help me ?

pixeldoc2000, 2023-12-04:
I have the same issue :-(

This probably by design to save power on the Device, could also be a firmware bug on the device?

https://www.zigbee2mqtt.io/devices/TS0205.html

pixeldoc2000, 2023-12-04:
Looking at the heardsman convert for TS0205, it looks like (very) basic support: https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/devices/tuya.ts
Also `ignore_basic_report` is defined, maybe thats why we don't get (status) update ?

pixeldoc2000, 2023-12-04:
My Device:
Zigbee Model: TS0205
Zigbee Manufacturer: _TZ3210_up3pngle

Looks like there is no "alias" for our "Model" for "_TZ3210_up3pngle"  ?

pixeldoc2000, 2023-12-04:
Sadly, the device does not seem to support / respond to bind of genPowerCfg for batteryPercentageRemaining or batteryVoltage

```
debug 2023-12-05 00:36:20 binding cluster 'genPowerCfg' from '0xa4c13867c545a189' to 'Coordinator'
error 2023-12-05 00:36:30 Failed to bind cluster 'genPowerCfg' from '0xa4c13867c545a189' to 'Coordinator' (Error: Bind 0xa4c13867c545a189/1 genPowerCfg from '0x00124b0021a8eb98/1' failed (AREQ - ZDO - bindRsp after 10000ms))
```

jocamane, 2023-12-05:
> ~Sadly, the device does not seem to support / respond to bind of genPowerCfg for batteryPercentageRemaining or batteryVoltage~
> 
> ```
> debug 2023-12-05 00:36:20 binding cluster 'genPowerCfg' from '0xa4c13867c545a189' to 'Coordinator'
> error 2023-12-05 00:36:30 Failed to bind cluster 'genPowerCfg' from '0xa4c13867c545a189' to 'Coordinator' (Error: Bind 0xa4c13867c545a189/1 genPowerCfg from '0x00124b0021a8eb98/1' failed (AREQ - ZDO - bindRsp after 10000ms))
> ```
> 
> Edit: Bind and batteryPercentageRemaining Report is working. The Devices sends it's battery state every 4h and stays online this way.

In my case the device after +-24 hours go to offline state

jocamane, 2023-12-05:
I repaired one of the devices. 
The other two I chose not to do.
I only removed the lid to cause the tamper to move and put it back in place. 
After that, it stopped being offline and went online. 

Now let's wait 24 hours to see the behavior of the 3 devices

jocamane, 2023-12-06:
[image]

It hasn't reached 24 hours yet but it looks like it will be offline again

pixeldoc2000, 2023-12-07:
It seems you have setup the external converter correctly, as it says Model: TS0205_new .

After setting up the external converter you need to repair the devices via zigbee2mqtt or pair the device again. Otherwise zigbee2mqtt may not bind the PowerCfg Cluster for the device.

It should look like this:
[image]

AFAIK the device should report the battery status (PowerCfg) at least every 4h .

pixeldoc2000, 2023-12-07:
@jocamane 
> I repaired one of the devices.

Which device did you repair and how?

pixeldoc2000, 2023-12-07:
> ![image](https://private-user-images.githubusercontent.com/32019937/288392162-9bf23ba4-7002-4bd1-b11d-cf2feac5d9b2.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3 [...]

Your LQI value of 255 is weird. Is every device right next to the coordinator?

jocamane, 2023-12-07:
> @jocamane It seems you have setup the external converter correctly, as it says Model: TS0205_new (= external converter).
> 
> After setting up the external converter, you need to pair the device again or try to repair the devices via zigbee2mqtt. Otherwise zigbee2mqtt may establish bind to PowerCfg Cluster and Reporting for the device.
> 
> It should look like this:
> 
> Bind: ![grafik](https://private-user-images.githubusercontent.com/376715/288694653-5e67c307-f8ff-467a-8000-70acb7f69ae5.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDE5NDA0NDEsIm5iZiI6MTcwMTk0MDE0MSwicGF0aCI6Ii8zNzY3MTUvMjg4Njk0NjUzLTVlNjdjMzA3LWY4ZmYtNDY3YS04MDAwLTcwYWNiN2Y2OWFlNS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBSVdOSllBWDRDU1ZFSDUzQSUyRjIwMjMxMjA3JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDIzMTIwN1QwOTA5MDFaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1hZjQ1NWIxMGQ5OWEzYzhmM2NmOTA0MzYzODhmYjgyOTcxNWE4MTcyZjExZTU1MGUyODZmNzljZmIxYzc0ODQwJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.zXoulAOqf3Gnk5CLUMe8iS5um33HFcroLAb2aIb9Tns)
> 
> Reporting: ![grafik](https://private-user-images.githubusercontent.com/376715/288698817-10034baa-4c4d-4c7c-b7a3-c33d8b52157e.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDE5NDA0NDEsIm5iZiI6MTcwMTk0MDE0MSwicGF0aCI6Ii8zNzY3MTUvMjg4Njk4ODE3LTEwMDM0YmFhLTRjNGQtNGM3Yy1iN2EzLWMzM2Q4YjUyMTU3ZS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBSVdOSllBWDRDU1ZFSDUzQSUyRjIwMjMxMjA3JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDIzMTIwN1QwOTA5MDFaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT03Yzc3ODNmOTMwMmEyNTNiMmRlZTIwMmJiYjQ3Mzg1NGYwN2FmNTAxZTQzODk4NzZhOWU4MGMyMjAwYzhjNGZjJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.ahGNTRpGm5garBSDYiHvC5s1dfoip8Hrj8Rlv65IStw) The device should report the battery status at least every 4h .
> 
> Change your log level to debug and check if there is any error for the device on interview while pairing / repair.

I repaired 1 device (Det Fumo Escritorio ZG). I deleted it from the z2m and re-paired it. Interestingly, it is one of those that is offline again. I didn't do anything with the other two, but I'm going to pair them.

Regarding bind and reporting, I have 2 identical to the ones you sent in the print and one of them has nothing.
Regarding LQI:
On the z2m home screen I have all devices with 255, but on the map they have different values. My coordinator is zigbeeII

This is what I repaired:
[image]
[image]

I didn't do anything about these:
[image]
[image]

[image]
[image]

The actual situation is:
[image]

jocamane, 2023-12-07:
[image]

pixeldoc2000, 2023-12-07:
@jocamane 
> I repaired 1 device (Det Fumo Escritorio ZG). I deleted it from the z2m and re-paired it. Interestingly, it is one of those that is offline again. I didn't do anything with the other two, but I'm going to pair them.

"Det Fumo Escritorio ZG" is connected via repeater to your coordinator. Maybe the repeater does not (fully) support the device. Maybe pair the device close to the coordinator. Some repeaters can cause all sorts of problems.
Also, restarting zigbee2mqtt may have trigger bind and reporting for (all) devices.

> Regarding bind and reporting, I have 2 identical to the ones you sent in the print and one of them has nothing.

Please repair the device close to the coordinator.

Some devices stay online and some don't, correct?

Never seen "wrong" LQI values in zigbee2mqtt device list. At least the Zigbee maps shows reals values. Maybe open a new issue for that problem.

jocamane, 2023-12-07:
@pixeldoc2000

> "Det Fumo Escritorio ZG" is connected via repeater to your coordinator. Maybe the repeater does not (fully) support the device. Maybe pair the device close to the coordinator. Some repeaters can [...]

The pair was made close to the coordinator but was connected to the repeater. I even put the "permit coordinator" only on z2m. Note that there are two sensors connected to the coordinator and one apparently works well and the other does not. tomorrow I will try repair again

Regarding restarting the z2m, I've already restarted the entire HA and it stayed the same.

> Some devices stay online and some don't, correct?

Yes, one of them goes online and the other two don't

> Never seen "wrong" LQI values in zigbee2mqtt device list. At least the Zigbee maps shows reals values. Maybe open a new issue for that problem.

I think that would be a good approach. If you need any help from me with logs, let me know

jocamane, 2023-12-08:
hi,
I did the following:
- I turned off the “LED Escadas ZG” device (where one of the smoke detectors was turning on)
- I delete the z2m's smoke detector from z2m
- Pair again by forcing the “coordenator”

In the end the map looked like this:

[image]

These messages appear in the core log whenever I pair one device, which have already appeared on other occasions when pairing other devices, so I don't pay much attention to them:

Logger: homeassistant.components.mqtt.mixins
Source: components/mqtt/mixins.py:553
Integration: MQTT (documentation, issues)
First occurred: 11:50:15 (10 occurrences)
Last logged: 12:04:11

Erroneous JSON:
____________________________________________________________________________
Logger: homeassistant.helpers.template
Source: helpers/template.py:2244
First occurred: 11:50:15 (10 occurrences)
Last logged: 12:04:11

Template variable error: 'value_json' is undefined when rendering '{{ value_json.smoke }}'
Template variable error: 'value_json' is undefined when rendering '{{ value_json.battery_low }}'
Template variable error: 'value_json' is undefined when rendering '{{ value_json.battery }}'
Template variable error: 'value_json' is undefined when rendering '{{ value_json.tamper }}'

__________________________________________________________________
Logger: homeassistant.helpers.template
Source: helpers/template.py:750
First occurred: 11:50:15 (7 occurrences)
Last logged: 12:04:11

Error parsing value: 'value_json' is undefined (value: , template: {{ value_json.smoke }})
Error parsing value: 'value_json' is undefined (value: , template: {{ value_json.battery_low }})
Error parsing value: 'value_json' is undefined (value: , template: {{ value_json.tamper }})

Follows the z2m debug log when pairing one of the devices:

[0xa4c138589ac24520.txt](https://github.com/Koenkk/zigbee2mqtt/files/13614044/0xa4c138589ac24520.txt)

Now Is there any way to know if it's ok without waiting 24 hours?

thanks

jocamane, 2023-12-08:
However, after this, I downgraded the conbeeII firmware because I read that the version I had could cause problems. (Warning: Conbee 2 firmware versions newer than 0x26580700 will result in an unstable network with devices dropping randomly, see [Issue 9554](https://github.com/Koenkk/zigbee2mqtt/issues/9554)) 

NOTE: After firmware downgrade, the LQI now appears well on the main screen
 
[image]

but strangely, the coordinator no longer has children connected to it and the 3 smoke sensors “TS0205_new” are not connected anywhere

[image]

Now wait....

jocamane, 2023-12-10:
[image]

Apparently it's ok, although the "last seen" time is much longer than other smoke sensors I have. I'll wait a little longer.

[image]

As far as the map is concerned, everything remains the same. the smoke detectors do not appear connected to anything. The next thing to do is to re-pair the devices now with the "old" version of the ZigBee firmware. Note that the devices were paired before I downgraded the firmware.

@pixeldoc2000 Do you have any other ideas for what I can do?

jocamane, 2023-12-11:
Apparently it's ok, the "last seen" time is 17 hours but ok

[image]

please apply the correction in the next z2m version

thanks

Chartreusito, 2023-12-13:
Just here to confirm the external converter is working fine on my end @pixeldoc2000.
This device model is advertised as "YG400A" on the one i bought. Do you want me to handle de picture PR ?

ThomasKlesse, 2023-12-15:
Yes, the external conveter works very well. Thank you. But don't forgett to reset - delete and teach again - your device.

pixeldoc2000, 2023-12-16:
@jocamane @Chartreusito @ThomasKlesse 
Thanks for your Feedback.

Created pull request with include fix in next Release: https://github.com/Koenkk/zigbee-herdsman-converters/pull/6724

jocamane, 2023-12-18:
> @jocamane @Chartreusito @ThomasKlesse Thanks for your Feedback.
> 
> Made a pull request that should fix the device: [Koenkk/zigbee-herdsman-converters#6724](https://github.com/Koenkk/zigbee-herdsman-converters/pull/6724)

Don't forget to change the picture, please

Chartreusito, 2023-12-18:
> > @jocamane @Chartreusito @ThomasKlesse Thanks for your Feedback.
> > Made a pull request that should fix the device: [Koenkk/zigbee-herdsman-converters#6724](https://github.com/Koenkk/zigbee-herdsman-converters/pull/6724)
> 
> Don't forget to change the picture, please

Picture PR is ready : https://github.com/Koenkk/zigbee2mqtt.io/pull/2406
Converter need to name the device accordingly for it to work.

pixeldoc2000, 2023-12-19:
@Chartreusito 
Fixed the name. I've added "Zigbee" to the model name because I noticed, the same model is sold with WIFI.

Smarticor, 2023-12-20:
Hi.
After adding external converter device was several times online
`info  2023-12-18 21:46:28: MQTT publish: topic 'zigbee2mqtt/smoke', payload '{"ac_status":false,"battery":63,"battery_defect":false,"battery_low":false,"last_seen":"2023-12-18T21:46:28+01:00","linkquality":176,"restore_reports":false,"smoke":false,"supervision_reports":false,"tamper":false,"test":false,"trouble":false}'
info  2023-12-19 04:45:11: MQTT publish: topic 'zigbee2mqtt/smoke', payload '{"ac_status":false,"battery":63,"battery_defect":false,"battery_low":false,"last_seen":"2023-12-19T04:45:11+01:00","linkquality":176,"restore_reports":false,"smoke":false,"supervision_reports":false,"tamper":false,"test":false,"trouble":false}'
`
But now it is offline again
[image]

pszewello, 2024-01-18:
I have the same issue... it joins, but can't set the bind or reporting (they timeout), battery % is always unknown, and device goes offline.

jocamane, 2024-01-20:
Hello. I just installed z2m version 1.35.1 and removed the external converter for this smoke detector.
Apparently everything is OK except the sensor image is not correct. At this moment it has the ZIGBEE symbol instead of the correct image. 
The correct image is same [TS0601_smoke_4] model above.

Was it possible to fix? Thanks

[image]

emilianogetino, 2024-01-23:
Does it already show the battery percentage?
It always shows it to me as:

Battery
Remaining battery in %, can take up to 24 hours before reported.
N/A%

Greetings, thank you.

attilabalazsy, 2024-01-25:
I've also updated to the lastest version as I'm facing the same issues as described above (no battery level indicator and continuous disconnect), but the latest version is not solving the battery level. I hope, that it at least solves the disconnect issue. Will see tomorrow.

george4ac, 2024-02-06:
I have updated to ver. 1.35.2 but smoke detectors are continuously disconnected. When tamper is activated smoke detector immediately send message. I tested detector by smoke also and it detect smoke and send message also. Know somebody solution for disconnecting and battery level?

attilabalazsy, 2024-02-06:
@george4ac I've updated to the latest version, but the battery was not solved, so I've applied the external converter workaround (see at the beginning of the thread) and this solved both the battery and connection issues. It is running smooth for 2 weeks.

fufurax, 2024-02-07:
> @george4ac I've updated to the latest version, but the battery was not solved, so I've applied the external converter workaround (see at the beginning of the thread) and this solved both the battery and connection issues. It is running smooth for 2 weeks.

I've the same issue. My zigbee devise goes to offline :(
I've installed latest version in docker container, delete and pair the smoke detector but nothing...same issue :(

So I've try to install the external converter : 
- copy paste the code in a file name "smoke.js"
- copy the file in a folder data at the root on my container (data\smoke.js)
- modify configuration.yaml to add the declaration of smoke.js 
```
external_converters:
  - smoke.js
```
- restart my container

but I've an error in the log folder: 
```
error 2024-02-07 19:00:12: Failed to load external converter file 'smoke.js' (ENOENT: no such file or directory, open '/app/data/smoke.js')
error 2024-02-07 19:00:12: Probably there is a syntax error in the file or the external converter is not compatible with the current Zigbee2MQTT version
```
Do you have an idea ?

Chartreusito, 2024-02-07:
@fufurax this looks like a config issue, i'd check file path and permissions. i'm using HA Add-on so i could not help you with  container config

fufurax, 2024-02-08:
> i'm using HA Add-on
?? 
It's the configuration.yaml is in Z2M not in HA.

founek2, 2024-02-17:
I am still having issues with disconnect even with external connector, which at least fixes the baterry percentage reporting. I am running zigbee2mqtt v1.35.1. Does anyone have some points, which would help with the disconnect? 

After pairing, it becomes "unavailable" after some time and never reconnects - only force disconnect and pairing again helps (but disconnects again).

[image]
[image]
> I apologise for the non-English language on the screenshots, but I am not sure how to switch the language in the UI.

jocamane, 2024-02-19:
here it is the same:
[image]
after repair go to offline after a few hours and the state is:

[image]
and
[image]

Core: 2024.1.6
Supervisor: 2024.01.1
Operating System: 11.4
Frontend: 20240104.0

Zigbee2MQTT version: [1.35.1] commit: [unknown]
Coordinator type:  ConBee2/RaspBee2
Coordinator revision:  0x26580700
Coordinator IEEE Address: 0x00212effff07afaa
Frontend version: 0.6.151
Zigbee-herdsman-converters version: 18.9.0
Zigbee-herdsman version: 0.30.0

what to do ?

fufurax, 2024-02-23:
Hi team, is there a workaround to fix the issue ?

vdb72, 2024-02-25:
@fufurax 
Have been trying a lot myself as per suggestions in this thread, but still going offline after 24h.  So I have no fix and might simply replace them with a bit more expensive zigbee smoke versions that I already own and do stay awake. (heiman hs1sa and xiaomi JTYJ-GD-01LM/BW)
  
Could have tried this myself by puting my house on fire ;-)... , but maybe someone knows...do they wake up / become active again when they detect smoke?

jocamane, 2024-02-25:
> @fufurax Have been trying a lot myself as per suggestions in this thread, but still going offline after 24h. So I have no fix and might simply replace them with a bit more expensive zigbee smoke [...]

YES. if there is smoke they activate

jocamane, 2024-02-25:
The solution is to put the external converter back in?

george4ac, 2024-02-29:
Try change this lines in TS0205.js:
  whiteLabel: [
        {vendor: 'Tesla Smart', model: 'TSL-SEN-SMOKE'},
        {vendor: 'Dongguan Daying Electornics Technology', model: 'YG400A'},
    ]

jocamane, 2024-03-02:
> Try change this lines in TS0205.js: whiteLabel: [ {vendor: 'Tesla Smart', model: 'TSL-SEN-SMOKE'}, {vendor: 'Dongguan Daying Electornics Technology', model: 'YG400A'}, ]

@george4ac and what exactly is this for?

george4ac, 2024-03-03:
I use this external converter to view the device as image YG400A.jpg

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    fingerprint: [{modelID: 'TS0205', manufacturerName: '_TZ3210_up3pngle'}],
    model: 'YG400A',
    vendor: 'TuYa',
    description: 'Smoke sensor',
    fromZigbee: [fz.ias_smoke_alarm_1, fz.battery, fz.ignore_basic_report],
    toZigbee: [],
    configure: async (device, coordinatorEndpoint, logger) => {
        try {
            const endpoint = device.getEndpoint(1);
            await reporting.bind(endpoint, coordinatorEndpoint, ['genPowerCfg']);
            await reporting.batteryPercentageRemaining(endpoint);
            //await reporting.batteryVoltage(endpoint);
        } catch (error) {/* Fails for some*/}
    },
    exposes: [e.smoke(), e.battery_low(), e.battery(), e.tamper()],
    whiteLabel: [
        {vendor: 'Tesla Smart', model: 'TSL-SEN-SMOKE'},
        {vendor: 'Dongguan Daying Electornics Technology', model: 'YG400A'},
    ],
};

module.exports = definition;
```

vdb72, 2024-03-03:
Both that I have are staying online it seems (first time 24h+) since Zigbee2MQTT 1.36.0-1 upgrade. 
Did the update and woke both up after the upgrade and they are still online after 24h. 
So far so good. Both show model = 'TS0205_smoke_2' where earlier I believe model was 'TS0205'

jocamane, 2024-03-03:
good news @vdb72 

will give news

jocamane, 2024-03-03:
I open my Smoke detector "TS0205 _TZ3210_up3pngle" and suprisse !!!

Model is: YG500A - A005

[image]

How do I make the detector seen as a YG500A - A005? it's possible ?

vdb72, 2024-03-04:
> Both that I have are staying online it seems (first time 24h+) since Zigbee2MQTT 1.36.0-1 upgrade. Did the update and woke both up after the upgrade and they are still online after 24h. So far so good. Both show model = 'TS0205_smoke_2' where earlier I believe model was 'TS0205'

'Last seen'  updated itself on both devices again so seems to stay online. 2 days in a row now. Considering this fixed (for now)

vdb72, 2024-03-04:
> I open my Smoke detector "TS0205 _TZ3210_up3pngle" and suprisse !!!
> 
> Model is: YG500A - A005
> 
> ![IMG_6062](https://private-user-images.githubusercontent.com/32019937/309546585-97f0e916-0d64-43a2-877e-54aeb6a6a5c4.JPEG?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDk1NDAzNjgsIm5iZiI6MTcwOTU0MDA2OCwicGF0aCI6Ii8zMjAxOTkzNy8zMDk1NDY1ODUtOTdmMGU5MTYtMGQ2NC00M2EyLTg3N2UtNTRhZWI2YTZhNWM0LkpQRUc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMzA0JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDMwNFQwODE0MjhaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0zM2M3ODI2MWE0Y2JjNGM4OGYyZjhhM2I3MmE3NTU1ZmNjNjY4YjVkZTMwNjg2NWZiMDE4YjNmYTc0MjBmZTI1JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.FKIwuA71BD8XbseJDULw_7TNEj_NVzymw6p8KW_jCps)
> 
> How do I make the detector seen as a YG500A - A005? it's possible ?

Doesn't the zigbee2mqtt update simply fix it for you?

fufurax, 2024-03-06:
> Both that I have are staying online it seems (first time 24h+) since Zigbee2MQTT 1.36.0-1 upgrade. Did the update and woke both up after the upgrade and they are still online after 24h. So far so good. Both show model = 'TS0205_smoke_2' where earlier I believe model was 'TS0205'

Hi, and no need to re-pair the device ?

vdb72, 2024-03-06:
> Hi, and no need to re-pair the device ?

No I did not do that. Did the upgrade and then just woke up the devices by pressing the button on the device (test function). That did the trick for me on both devices. Checked both devices just now and still active.

fufurax, 2024-03-06:
> No I did not do that. Did the upgrade and then just woke up the devices by pressing the button on the device (test function). That did the trick for me on both devices. Checked both devices just now [...]

Nice I do the same. I'll keep you posted ;)

fufurax, 2024-03-07:
Looks not working for me...My TS0205_smoke_2 is offline since 12 hours. currently in 1.36.0

fufurax, 2024-03-08:
new feedback. 
Il looks that the device comes Up now ;) 
see the activity of last seen in HA. 
It's Up for 4H and down for 14H. Is there a way to reduce the down time ?

[image]

vdb72, 2024-03-08:
Haven't looked at it in that view. Doesn't it look like that because the refresh intervals are just 1-2 times a day? 

This is how I look at the status of mine.... If it states 'online'  (which it now always does) I'm fine with it.
[image]

cyrkul, 2024-03-18:
For me device is detected and online after updates but when I tried testing smoke detection (with burning, not button) it does nothing. Am I the only one? Did anyone actually test that?

george4ac, 2024-03-18:
I have 2 and have tested them with smoldering cotton cloth. But they were online and smoke was detected. But they repeatedly go offline. I have lost confidence in them and I am going to replace them with HEIMAN HS1SA.

philsson, 2024-04-07:
> For me device is detected and online after updates but when I tried testing smoke detection (with burning, not button) it does nothing. Am I the only one? Did anyone actually test that?

So they do not report smoke over zigbee even when Active? I came to this thread to read as mine also go Offline. They get online as soon as I press the test button but I have not tested smoke, neither while active or inactive. Now I feel bothered as I have many of these. Was just assuming that they would connect back when detecting smoke.

[34 more replies]

closed 2025-08-12 by github-actions[bot]