Koenkk/zigbee2mqtt #19915: [New device support]: bulb

nicoto94, 2023-11-29:
### Link

https://boutique.deltadore.fr/products/easy-bulb-e27cw-ampoule-connectee-blanc-et-couleurs

### Database entry

{"id":26,"type":"Router","ieeeAddr":"0x040d84fffed03e63","nwkAddr":17844,"manufId":4762,"manufName":"Rademacher","powerSource":"Mains (single phase)","modelId":"RDM-35274001","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":269,"inClusterList":[0,3,4,5,6,8,768,4096,64642],"outClusterList":[25],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":2,"stackVersion":2,"hwVersion":1,"dateCode":"20211117","swBuildId":"2.09.03","zclVersion":3,"interviewCompleted":false,"meta":{},"lastSeen":1699377889182,"defaultSendRequestWhen":"immediate"}

### Comments

I tried to follow the support page but I was unsuccessful, I would like help to integrate these bulbs?

### Supported color modes

color temperature, RGB

### Color temperature range

{"colorTempPhysicalMin":153,"colorTempPhysicalMax":555}

Koenkk, 2023-12-02:
Could you check if the issue is fixed with the following external converter: https://gist.github.com/Koenkk/ec7a0f69f8b0e292b847c5654ba34d16

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

- what is the vendor of this device?

nicoto94, 2023-12-03:
Thanks for your feedback
the seller is DELTA DORE; I have provided you with a link of the product in the post

nicoto94, 2023-12-03:
Hello
So I tried your method

* create a .js file:
ext_converter.js
then I copied your line of code into this file:

`const  fz  =  require ( 'zigbee-herdsman-converters/converters/fromZigbee' ) ;
const  tz  =  require ( 'zigbee-herdsman-converters/converters/toZigbee' ) ;
const  expose  =  require ( 'zigbee-herdsman-converters/lib/exposes' ) ;
const  reporting  =  require ( 'zigbee-herdsman-converters/lib/reporting' ) ;
const  Legacy  =  require ( 'zigbee-herdsman-converters/lib/legacy' ) ;
const  extend  =  require ( 'zigbee-herdsman-converters/lib/extend' ) ;
const  ota  =  require ( 'zigbee-herdsman-converters/lib/ota' ) ;
const  tuya  =  require ( 'zigbee-herdsman-converters/lib/tuya' ) ;
const  utils  =  require ( 'zigbee-herdsman-converters/lib/utils' ) ;
const  globalStore  =  require ( 'zigbee-herdsman-converters/lib/store' ) ;
const  { light }  =  require ( 'zigbee-herdsman-converters/lib/modernExtend' ) ;
const  e  =  expose . préconfigurations ;
const  ea  =  expose . accéder ;

 définition  const =  {
    zigbeeModèle : [ 'RDM-35274001' ] ,
    modèle : 'RDM-35274001' ,
    vendeur : 'TODO' ,
    description : 'Ampoule RVB' ,
    extend : [ light ( { colorTemp : { range : [ 153 ,  555 ] } ,  color : true } ) ] ,
} ;

modules . exportations  =  définition ;`

Then I copied the code:
`external_converters:
   - ext_converter.js`
  in the _**configuration.yaml**_ file

Then I restarted z2m and it keeps stopping....

Did I make a mistake?

Here are the debug logs after restarting z2m

`> zigbee2mqtt@1.34.0 start
> node index.js
/opt/zigbee2mqtt/node_modules/js-yaml/lib/loader.js:1722
throw new YAMLException('expected a single document in the stream, but found more');
^
YAMLException: expected a single document in the stream, but found more
at Object.load (/opt/zigbee2mqtt/node_modules/js-yaml/lib/loader.js:1722:9)
at Object.read (/opt/zigbee2mqtt/lib/util/yaml.ts:7:21)
at read (/opt/zigbee2mqtt/lib/util/settings.ts:333:20)
at getInternalSettings (/opt/zigbee2mqtt/lib/util/settings.ts:438:21)
at Object.reRead (/opt/zigbee2mqtt/lib/util/settings.ts:721:5)
at start (/opt/zigbee2mqtt/index.js:91:14)`

> Pourriez-vous vérifier si le problème est résolu avec le convertisseur externe suivant : https://gist.github.com/Koenkk/ec7a0f69f8b0e292b847c5654ba34d16
> 
> * enregistrez-le en tant que fichier à côté `configuration.yaml`de`ext_converter.js`
> * ajoutez-le à `configuration.yaml`:
> 
> ```yaml
> external_converters:
>   - ext_converter.js
> ```
> 
> * démarrez z2m, vérifiez si le problème est résolu
> * quel est le vendeur de cet appareil ?

> Pourriez-vous vérifier si le problème est résolu avec le convertisseur externe suivant : https://gist.github.com/Koenkk/ec7a0f69f8b0e292b847c5654ba34d16
> 
> * enregistrez-le en tant que fichier à côté `configuration.yaml`de`ext_converter.js`
> * ajoute-le à `configuration.yaml`:
> 
> ```yaml
> external_converters:
>   - ext_converter.js
> ```
> 
> * démarrez le z2m, vérifiez si le problème est résolu
> * quel est le vendeur de cet appareil ?

nicoto94, 2023-12-03:
I tried to use a zigbeelinker plugin which is on Jeedom and which creates a .js file as well as a converter in the configuration.yaml file
When I have this code in the .js file I do not have Z2m crash:
`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['RDM-35274001'],
    model: 'RDM-35274001',
    vendor: 'Changez par le nom du fabricant',
    description: 'Tapez la description du module ici',
    fromZigbee: [],
    toZigbee: [],
    exposes: [e.battery()],
};

module.exports = definition;`

but when I copy the code you sent me in the link; z2m keeps stopping....

`
const  fz  =  require ( 'zigbee-herdsman-converters/converters/fromZigbee' ) ;
const  tz  =  require ( 'zigbee-herdsman-converters/converters/toZigbee' ) ;
const  expose  =  require ( 'zigbee-herdsman-converters/lib/exposes' ) ;
const  reporting  =  require ( 'zigbee-herdsman-converters/lib/reporting' ) ;
const  Legacy  =  require ( 'zigbee-herdsman-converters/lib/legacy' ) ;
const  extend  =  require ( 'zigbee-herdsman-converters/lib/extend' ) ;
const  ota  =  require ( 'zigbee-herdsman-converters/lib/ota' ) ;
const  tuya  =  require ( 'zigbee-herdsman-converters/lib/tuya' ) ;
const  utils  =  require ( 'zigbee-herdsman-converters/lib/utils' ) ;
const  globalStore  =  require ( 'zigbee-herdsman-converters/lib/store' ) ;
const  { light }  =  require ( 'zigbee-herdsman-converters/lib/modernExtend' ) ;
const  e  =  expose . préconfigurations ;
const  ea  =  expose . accéder ;

 définition  const =  {
    zigbeeModèle : [ 'RDM-35274001' ] ,
    modèle : 'RDM-35274001' ,
    vendeur : 'TODO' ,
    description : 'Ampoule RVB' ,
    extend : [ light ( { colorTemp : { range : [ 153 ,  555 ] } ,  color : true } ) ] ,
} ;

modules . exportations  =  définition ;`

Koenkk, 2023-12-03:
Are you using z2m 1.34.0?

Koenkk, 2023-12-04:
What error shows up in the z2m log when startup failure?

nicoto94, 2023-12-07:
`MQTT publish: topic 'zigbee2mqtt/Répéteur Tuya Zigbee local technique EDF', payload '{"device":{"applicationVersion":64,"dateCode":"","friendlyName":"Répéteur Tuya Zigbee local technique EDF","hardwareVersion":1,"ieeeAddr":"0x943469fffe547eb0","manufacturerID":4098,"manufacturerName":"_TZ3000_nlsszmzl","model":"TS0207_repeater","networkAddress":16792,"powerSource":"Mains (single phase)","stackVersion":0,"type":"Router","zclVersion":3},"linkquality":160}'
Debug 2023-12-07 21:48:04Received Zigbee message from 'Zlinky', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"rmsCurrent":3}' from endpoint 1 with groupID 0
Info 2023-12-07 21:48:05MQTT publish: topic 'zigbee2mqtt/Zlinky', payload '{"active_enerfy_out_d01":16604,"active_enerfy_out_d02":17379,"active_power":1888,"active_power_max":7380,"active_power_ph_b":2374,"apparent_power":2040,"available_power":9,"average_rms_voltage_meas_period":227,"current_date":"H231207214030","current_index_tarif":2,"current_price":"HEURE PLEINE","current_summ_delivered":41045,"current_summ_received":1634,"current_tarif":"H PLEINE/CREUSE","current_tier1_summ_delivered":19901,"current_tier2_summ_delivered":21144,"device":{"applicationVersion":12,"dateCode":"20230214","friendlyName":"Zlinky","hardwareVersion":1,"ieeeAddr":"0x00158d000638e304","manufacturerID":4151,"manufacturerName":"LiXee","model":"ZLinky_TIC","networkAddress":49563,"powerSource":"Mains (single phase)","softwareBuildID":"4000-0013","stackVersion":2,"type":"Router","zclVersion":3},"drawn_v_a_max_n1":6320,"injected_active_load_n":0,"injected_active_load_n1":0,"injected_v_a":0,"injected_v_a_max_n":920,"injected_v_a_max_n1":210,"linkquality":100,"message1":"PAS DE MESSAGE","message2":"","meter_serial_number":"021961305060","power_threshold":9,"reactive_power":3781,"reactive_power_ph_b":-22188,"reactive_power_ph_c":-5450,"relais":0,"rms_current":3,"rms_voltage":227,"site_id":"12483791554880","software_revision":2,"status_register":"003AC501","total_reactive_power":10910,"update":{"installed_version":13,"latest_version":13,"state":"idle"},"warn_d_p_s":0}'
Debug 2023-12-07 21:48:08Received Zigbee message from 'Thermomètre Local WC', type 'attributeReport', cluster 'msTemperatureMeasurement', data '{"measuredValue":791}' from endpoint 1 with groupID 0
Info 2023-12-07 21:48:08MQTT publish: topic 'zigbee2mqtt/Thermomètre Local WC', payload '{"battery":28,"device":{"applicationVersion":0,"dateCode":"20231109","friendlyName":"Thermomètre Local WC","hardwareVersion":14,"ieeeAddr":"0xa4c13846a38c73f7","manufacturerID":4417,"manufacturerName":"Xiaomi","model":"LYWSD03MMC","networkAddress":21414,"powerSource":"Battery","softwareBuildID":"3001-1006","stackVersion":2,"type":"EndDevice","zclVersion":3},"humidity":75.05,"linkquality":160,"temperature":7.91,"update":{"installed_version":268840961,"latest_version":268840961,"state":"idle"},"voltage":2400}'
Debug 2023-12-07 21:48:08Received Zigbee message from 'Thermomètre Local WC', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":268840961,"imageType":515,"manufacturerCode":4417}' from endpoint 1 with groupID 0
Debug 2023-12-07 21:48:08Device 'Thermomètre Local WC' requested OTA
Debug 2023-12-07 21:48:08Received Zigbee message from 'Prise 2', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":22,"rmsCurrent":204,"rmsVoltage":226}' from endpoint 1 with groupID 0
Info 2023-12-07 21:48:08MQTT publish: topic 'zigbee2mqtt/Prise 2', payload '{"child_lock":"UNLOCK","current":0.2,"device":{"applicationVersion":192,"dateCode":"","friendlyName":"Prise 2","hardwareVersion":1,"ieeeAddr":"0xa4c1388f19938e31","manufacturerID":4417,"manufacturerName":"_TZ3000_2putqrmw","model":"A1Z","networkAddress":26330,"powerSource":"Mains (single phase)","stackVersion":0,"type":"Router","zclVersion":3},"energy":10.78,"indicator_mode":"off/on","linkquality":196,"power":22,"power_outage_memory":"restore","state":"ON","update":{"installed_version":192,"latest_version":192,"state":"idle"},"voltage":226}'
Debug 2023-12-07 21:48:08Received Zigbee message from 'Prise 2', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered":[0,1078]}' from endpoint 1 with groupID 0
Info 2023-12-07 21:48:08MQTT publish: topic 'zigbee2mqtt/Prise 2', payload '{"child_lock":"UNLOCK","current":0.2,"device":{"applicationVersion":192,"dateCode":"","friendlyName":"Prise 2","hardwareVersion":1,"ieeeAddr":"0xa4c1388f19938e31","manufacturerID":4417,"manufacturerName":"_TZ3000_2putqrmw","model":"A1Z","networkAddress":26330,"powerSource":"Mains (single phase)","stackVersion":0,"type":"Router","zclVersion":3},"energy":10.78,"indicator_mode":"off/on","linkquality":188,"power":22,"power_outage_memory":"restore","state":"ON","update":{"installed_version":192,"latest_version":192,"state":"idle"},"voltage":226}'
Debug 2023-12-07 21:48:09Received Zigbee message from 'Thermomètre chambre fille', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":268840961,"imageType":515,"manufacturerCode":4417}' from endpoint 1 with groupID 0
Debug 2023-12-07 21:48:09Device 'Thermomètre chambre fille' requested OTA
Debug 2023-12-07 21:48:09Received Zigbee message from 'Zlinky', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"rmsVoltage":229}' from endpoint 1 with groupID 0
[... cut ...] _b":-22188,"reactive_power_ph_c":-5450,"relais":0,"rms_current":3,"rms_voltage":229,"site_id":"12483791554880","software_revision":2,"status_register":"003AC501","total_reactive_power":10910,"update":{"installed_version":13,"latest_version":13,"state":"idle"},"warn_d_p_s":0}'
Debug 2023-12-07 21:48:38Received Zigbee message from 'Thermomètre chambre fille', type 'attributeReport', cluster 'msRelativeHumidity', data '{"measuredValue":6105}' from endpoint 1 with groupID 0
Info 2023-12-07 21:48:38`

nicoto94, 2023-12-07:
zigbee2mqtt crashes afterwards, I no longer have any access; I am obliged to delete the .js file

nicoto94, 2023-12-08:
> Quelle erreur apparaît dans le journal z2m lorsque le démarrage échoue ?

salut,
Désolé la page des codes avaient été traduite en Francais et donc j ai copié le mauvais code... J 'arrive a controler maintenant a lumière : - - - On/Off , 
- la luminosité et la température des blancs 

- mais je n ai pas les couleurs RGB ?peux tu me donner un code avec les couleurs RGB ?

En te remerciant par avance

Bonne soirée

nicoto94, 2023-12-10:
> Are you using z2m 1.34.0?

Hi,
Sorry the codes page had been translated into French and so I copied the wrong code... I can now control the light: On/Off, the brightness and the white temperature

**but I can't adjust the RGB colors? Can you add the code for adjusting the RGB colors?**

Thanking you in advance

Good evening

nicoto94, 2023-12-20:
Help

> Are you using z2m 1.34.0?

Hi,
Sorry the codes page had been translated into French and so I copied the wrong code... I can now control the light: On/Off, the brightness and the white temperature

but I can't adjust the RGB colors? Can you add the code for adjusting the RGB colors?

Thanking you in advance

Good evening

closed 2024-07-18 by github-actions[bot]