Koenkk/zigbee2mqtt #19916: OTA update fails with error: "Update failed (Device didn't respond to OTA request)"

c0mplex1, 2023-11-29:
### What happened?

I'm trying to push a firmware update to a Salus SR600 model.
“Checking for new updates” was successful and new firmware was detected.
When you click the OTA "Update device firmware" button, the error message "Update failed (device did not respond to OTA request)" appears.
The Salus SR600 is powered via mains power, restarting the device by disconnecting the power for a few seconds does not have the desired result.

### What did you expect to happen?

I expected the OTA firmware update process to begin.

### How to reproduce it (minimal and precise)

Click OTA "Update device firmware" button, wait 60 seconds for error

### Zigbee2MQTT version

1.33.2

### Adapter firmware version

20230507

### Adapter

Slaesh's CC2652RB stick

### Debug log

info 2023-11-29 14:58:09 Updating 'CV-ketel' to latest firmware
info 2023-11-29 14:58:09 MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Updating 'CV-ketel' to latest firmware","meta":{"device":"CV-ketel","status":"update_in_progress"},"type":"ota_update"}'
info 2023-11-29 14:58:09 MQTT publish: topic 'zigbee2mqtt/CV-ketel', payload '{"last_seen":"2023-11-29T14:58:36+01:00","linkquality":174,"power_on_behavior":null,"state":"ON","update":{"installed_version":538380576,"latest_version":538510357,"state":"available"},"update_available":true}'
info 2023-11-29 14:59:09 MQTT publish: topic 'zigbee2mqtt/CV-ketel', payload '{"last_seen":"2023-11-29T14:58:36+01:00","linkquality":174,"power_on_behavior":null,"state":"ON","update":{"installed_version":538380576,"latest_version":538510357,"state":"available"},"update_available":true}'
info 2023-11-29 14:59:09 MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Update of 'CV-ketel' failed (Device didn't respond to OTA request)","meta":{"device":"CV-ketel","status":"update_failed"},"type":"ota_update"}'
info 2023-11-29 14:59:09 MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/update', payload '{"data":{"id":"CV-ketel"},"error":"Update of 'CV-ketel' failed (Device didn't respond to OTA request)","status":"error","transaction":"vdw7p-1"}'
error 2023-11-29 14:59:09 Update of 'CV-ketel' failed (Device didn't respond to OTA request)

muzzy124, 2024-01-12:
same problem with legrand 067773 and 067774
Z2M: 1.35.1-dev commit: [ef84ff8](https://github.com/Koenkk/zigbee2mqtt/commit/ef84ff8)
EZSP v7 coordinator (hacked tuya gateway)
HA 2023.12.0

532910, 2024-04-04:
same issue with ikea E1745 motion sensor

Loic691, 2024-04-09:
Hi
Z2M 1.36.1
Legrand device 067774 (wireless double hangs)  with 24 firmware
A firmware is ready in Z2M but update don't working. The update don't start...

532910, 2024-04-09:
https://github.com/home-assistant/core/issues/110103#issuecomment-2041526310

kdkavanagh, 2024-05-04:
Same issue with all my inovelli blue switches

Loic691, 2024-07-02:
Hi all, hi @Nerivec
Since upgrade Z2M to 1.39.0 (using Ember driver on Sonoff Dongle  E), I no longer have the "device dodn't respond to OTA").
But I have now the following message : 
`zh:ember: ~x~> [ZCL to=2225] Failed to send request with status=MESSAGE_TOO_LONG.`

Note I had upgraded sucessfully 412171,  067771 Legrand device this morning.

I already have a problem for : 
- 067773 wireless simple switch (object of this post and fews line upstairs)
- 067774 double gangs wireless => update go to 100% but the device isn't upgraded
- 064873 home away wireless

Nerivec, 2024-07-03:
Are these devices with "message too long" farther away from the coordinator?
It could happen if the devices are nested too deeply (as in "device > router > router > ... > coordinator"). Since ZigBee needs some overhead to get messages to the destination, that can eat away at the possible payload size (which OTA tends to push close to the limit).

muzzy124, 2024-07-03:
067773 with 3c firmware are still getting Update failed (device did not respond to OTA request)
z2m 1.39.0-dev commit: [addc238]

Mincka, 2025-01-02:
Not stale

Nihvel, 2025-01-06:
Got the same error "Update failed (Device didn't respond to OTA request)" and fixed as follows:

1. Force remove the device from z2m and remove the batteries
2. Stop z2m
3. Add the 'ext_converter.js', because my device is a Ikea Styrbar and I followed the instructions [here ](https://github.com/Koenkk/zigbee2mqtt/issues/18515#issuecomment-1692857032)
4. Start z2m
5. Add the batteries and let it join the network
6. As soon as the device joins, with the z2m message printing "successful" and without any delay, click the "check for new updates" button in z2m OTA page

The OTA update definitely works, but requires some tinkering. Start the update with fully charged batteries. It can take hours to complete.

lluisd, 2025-01-12:
same problem here from time ago

c0mplex1, 2025-01-12:
It's been 200 days since I reported the issue, but the Salus SR600 firmware upgrade issue "Update failed (device did not respond to OTA request)" is still not resolved.
Please fix it.
Thank you.

DominikFilipiak, 2025-01-25:
I had a lot of problems updating my Sonoff TRVZB through z2m. I tried with different channels and setting some z2m options, but no luck. 

Turns out, my z2m network was kind of overloaded by HA-side hysteresis automations adjusting radiator valve degrees. Every 30 seconds there were messages flying around from every radiator.

**When I turned automations generating the traffic off, the update worked like a charm** (well, it still took about an hours, but it eventually worked).

In z2m, just open logs and see how many messages are you getting. If there's a lot, try to disable their sources (such as mine automations) during the update. Good luck!

defanator, 2025-03-08:
@fabian19vs thanks for sharing this - I was able to update IKEA's Rodret E2201 using that advice - https://github.com/Koenkk/zigbee2mqtt/issues/6729#issuecomment-2706721185

OliviezGit, 2025-03-15:
hi any news about that issue? i'm facing the same issue....

and not all Switches detect.. an update very stange...

[image]

[image]

[image]

Nihvel, 2025-03-18:
> hi any news about that issue? I'm facing the same issue....
> 
> and not all Switches detect an update... very strange...
> 
> [image]
> 
> [image]
> 
> [image]

There's no magic. I literally followed the steps I wrote and I get a 100% success rate when updating devices. With or without batteries.

[image]

Mincka, 2025-03-18:
Ok, but do you have Legrand devices?

Because I tried the method you described but it never worked for Legrand switches.

soif, 2025-05-02:
Hi guys,

I’ve run into frustrating OTA update issues with Zigbee2MQTT, like crashes, "Device didn't respond to OTA request" errors, or updates failing after a while. 

To help, I've created a free open-source **[script](https://github.com/soif/z2m_helpers)** (`z2m_ota.sh`) that automatically retries OTA updates until they succeed. It doesn’t fix the core issue, but it saves hours of manually re-clicking "update firmware." 

Give it a try and let me know how it works for you! 😊

Loic691, 2025-05-04:
Hi with last version of Z2M 2.3.0 I was finally able to update home away 064873.
I have to test 2 or 3 times before update starts...

eekamouze, 2025-05-09:
I have a Legrand Centralized ventilation switch (https://www.zigbee2mqtt.io/devices/067766.html) that always failed to be updated (same error) until z2m 2.2.1. For a reason that I ignore, update eventually went all the way on. Now with z2m 2.3.0, I have a warning log saying that "Firmware is newer than latest available firmware" 🤷🏼‍♂️

OliviezGit, 2025-05-09:
hi,
your Legrand Centralized ventilation switch (067766) is already at the above version that Z2M knows...
Z2M are not directly connect to legrand firmware repository. https://developer.legrand.com/production-firmware-download/

if you don't want to wait to upgrade the firmware version you have to do that manually:  
-local ota file: [Local OTA index and firmware files](https://www.zigbee2mqtt.io/guide/usage/ota_updates.html#local-ota-index-and-firmware-files)
-OR propose Z2m update: https://github.com/koenkk/zigbee-ota#adding-new-and-updating-existing-ota-files

nicozocco, 2025-05-24:
> > hi any news about that issue? I'm facing the same issue....
> > and not all Switches detect an update... very strange...
> > [image]
> > [image]
> > [image]
> 
> There's no magic. I literally followed the steps I wrote and I get a 100% success rate when updating devices. With or without batteries.
> 
> [image]

I don't have the Identify button un HA for this device :S I've tried following those steps but I only managed to get Z2M to detect there's an update available, haven't been able to update it as once i click to update it never does

closed 2025-08-01 by github-actions[bot]