Koenkk/zigbee2mqtt #19922: [New device support]: TuYa MTG235-ZB-RL

Ferugf, 2023-11-29:
### Link

https://pl.aliexpress.com/item/1005005633355282.html?spm=a2g0o.order_list.order_list_main.27.67515e5bpE050Q&gatewayAdapt=glo2pol

### Database entry

{"id":17,"type":"Router","ieeeAddr":"0x3425b4fffee69b62","nwkAddr":63151,"manufId":4098,"manufName":"_TZE204_clrdrnya","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"\u0000\u0000\u0000\u0000e\u0000\u0000\u0000\u0000\u0012\u0000\u0000\u0000\u0000\u0012","65506":31,"65508":0,"65534":0,"modelId":"TS0601","manufacturerName":"_TZE204_clrdrnya","powerSource":1,"zclVersion":3,"appVersion":70,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":70,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-1440156621},"lastSeen":1701274301826,"defaultSendRequestWhen":"immediate"}

### Comments

 ZigBee 24G ceiling -  24G MmWave radar human presence motion sensor, similar to MTG275-ZB-RL
    Router
Model Zigbee   TS0601
Producent Zigbee   _TZE204_clrdrnya

closed 2024-06-27 by github-actions[bot]

srpablillo, 2024-08-11:
I am also interested in this same device. I can integrate to ZHA but no info is shown. 

I am running HASSOS on a RP5, last soft version available right now.

[image]

austin7777, 2024-11-14:
I have one too!